linuxboot/heads #1156: dGPU boards failed gpg operations

walliams, 2022-04-15:
v0.2.0-1192-g3201cd4 boards for w530-dgpu-k2000m-hotp-maximized and t430-dgpu-hotp-maximized (configs and vbios scripts merged from @eganonoa's https://github.com/eganonoa/heads/tree/t430-dgpu) have common issues with gpg operations.

OEM Factory Reset works to completion on both boards, but the following operations fail immediately after final confirmation prompt:
- Add GPG key to running BIOS and reflash
- Add GPG key to standalone BIOS image and flash
- Replace GPG key(s) in the current ROM and reflash

1st and 3rd option read the flash chip 3 times and then prompt "This will reflash your BIOS with the updated version. Do you want to proceed". upon hitting Enter, it flashes the lines of code from the previous screen and then immediately skips back to Main Menu.
2nd option prompts "this will replace your old ROM with /... Do you want to proceed?" and then upon hitting Enter, it flashes a couple lines about the gpg key for a split second. what I can make out by taking a video and freeze framing it is:
`gpg: Total number processed: 1
gpg:                   w/o User IDs: 1`
then, like the other 2 options, it jumps to the Main Menu

the following option appears to get further but ultimately fails:
-Generate GPG keys manually on a USB security token

in this case i either generate a new key or just type admin the quit, and it asks if i would like to save the gpg key to a usb, then to add the gpg key to the running bios and sign files in /boot. upon confirming, it reads the flash 3 times, initializes the internal programmer, shows a progress bar as though it is reflashing, completes, and then prompts me to update checksums. it then reports "Failed to update checksums / sign default config" then when i hit enter again it immediately says "Checksums have been updated and /boot files signed. Press Enter to reboot" as if it thinks it succeeded... however, this doesn't actually seem to be the point of failure (checksums), because when i reboot and select "List GPG keys in your keyring" it returns empty.

finally, i think that i have actually had this issue or similar issues with gpg operations on all dgpu boards i have built, so this is not unique to any recent commits. the only way that i have ever been able to get a gpg key added to the bios is with an OEM Factory Reset. and i would love it if someone else with one of these dgpu boards could try to replicate the issue so i know i'm not soing something wrong.

tlaurion, 2022-04-15:
It seems that tour public key was not merged into the firmware?

A generated public key, exported to USB dongle can only work if matching the private key on the smartcard.

If the factory reset scripts generates a key that is then fused into firmware and is recognised as working, it works.

Factory resetting the USB security dongle wipes previous private subkeys. 

Injecting a public key that is not matching the signing private key in the USB security dongle would explain the behavior you are having.

Otherwise I have no explanation @walliams .

walliams, 2022-04-15:
thing is, i am able to get it to the point where it passes the hotp verification but still won't sign the checksums... so i am merging the right key into the firmware, creating a new secret successfully, but it refuses to update checksums.
is it possible it is a privileges thing? is it safe to manually wipe kexec files from /boot with `rm -rf kexec*` and then try updating checksums again? i considered doing this, but didn't want to fiddle with my /boot. i also tried to do a backup of the kexec files before wiping, but i was unable to mount a usb drive. `mkdir /tmp/usb` then `mount /dev/sdb1 /tmp/usb` fails. which is also confusing, because that worked for me on a previous heads build...

tlaurion, 2022-04-15:
Two concepts are mixed here.

HOTP sealing can be done to any compatible HOTP USB Security dongle as long as its GPG Admin PIN is known.

GPG function requires having a public key matching private key on the USB Security dongle.

Doing a factory reset will regenerate keypairs for which the public key is fused to the firmware, and exported to USB thumb drive. But that public key is valid only until the next OEM factory reset, which will invalidate previous private keys on the USB Security dongle.

Each time you run the factory reset, you invalidate past private keys on the USB Security dongle.

My hypothesis here is that the public key you try to inject in ROM doesn't match any private key anymore in your USB Security dongle, which seems to match your success story with OEM factory resetting your USB Security dongle.

Makes sense? If yes please rename issue title and close. 

Unfortunately, there is no way to restore private keys under Heads. When you factory reset, you invalidate past keypairs. I think this is where there is a misunderstanding.

walliams, 2022-04-15:
thankyou for clarification of what's going on with gpg keys, but i still think your hypothesis is incorrect.
let me back up a bit. so my goal is to do this:

1) OEM Factory Reset one laptop
2) Export the new key to USB
3) Merge this key to the firmware of another laptop, sign /boot files, generate a new secret

is this process possible-- to have 2 laptops verified from the same HOTP Security Dongle? or am i still misunderstanding?
i am certain i have been trying to merge the most recently generated key with the firmware on the second laptop, but it is never able to update checksums.

tlaurion, 2022-04-15:
You can't use the same USB Security dongle to use HOTP for remote attestation. 

This is a one laptop/HOTP sealing. 

For the GPG signing, the signing is made on the USB dongle, and then verified with the public key fused in ROM.

The reason why I still believe in my hypothesis (private and public key not matching) comes from the error you posted initially.

You can verify current public key by showing it under options->gpg options-> show public key

walliams, 2022-04-15:
ok my apologies. i guess i have been attempting to do something that is impossible. i misunderstood the purpose of the GPG options available. if you could would you please elaborate on what purpose these operations serve? is it, for example, if you flashed a new rom with no gpg and then want to add the original key back in?

tlaurion, 2022-04-15:
HOTP/TOTP are firmware attestation features. TOTP is not requiring USB Security dongle, while HOTP binds one to Heads (HOTP variants of boards). This permits visual remote attestation of firmware measurements, including the measurement of the public key fused into the firmware.

A GPG Security dongle is required under Heads. A valid keypair is required, the public key matching the private key needs to be injected in the ROM and reflashed.

GPG is used to detach sign the checksum file of /boot content. Then the detach signature is used to verify that user authenticated the last changes that happened under /boot.

Other questions on documents station, needed clarifications should happen under heads-wiki for what is unclear on https://osresearch.net

walliams, 2022-04-15:
@tlaurion thanks for the breakdown.

a little followup for clarity:

i just built t430-dgpu-maximized (w/o hotp) and then using the gpg key generated from my w530-dgpu-K2000m-hotp-maximized board in the latest OEM Factory Reset, i was able to merge the gpg key into the t430 board (w/o hotp) and successfully updated checksums.

so it seems that my previous issue was that the checksum updates were blocked by a mismatched hotp secret on the key... that seems to be the only difference here.

anyway, i am satisfied with this result and thanks again for the guiding hand.

now that i have successfully built and flashed the t430-dgpu board i will follow the guide in the wiki to fork the git and PR a merge to bring those configs upstream to master.

tlaurion, 2022-04-15:
> gpg: no default secret key: No public key
> gpg: signing failed: No public key
> gpg: no default secret key: No public key
> gpg: signing failed: No public key
> gpg: no default secret key: No public key
> gpg: signing failed: No public key
> /boot: Unable to sign kexec hashes

Then there is another possibility.
You flashed adding a key, and the error that mattered in that block was that there was no default secret key. So here, that would have meant that multiple public keys were fused into the ROM, where Heads expects to have only one.

Would be nice if having two public keys was retested to confirm the behavior, in which case the title of this issue being renamed. The error not linked to a specific board, but an unsupported use case, that is correct. In which case, code code be changed to warn that having two user keys would cause error and prompting the user to replace public key (not add). You might have found a bug, by adding a public key where one was already present.

walliams, 2022-04-17:
if i am interpreting the results of my testing correctly, it seems this issue applies to not only multiple public keys on the ROM, but also multiple instances of the same key written into the ROM -- here was my process:

1) OEM Factory Reset on my w530-hotp. Key exported to usb.
2) on t430: Add GPG to standalone BIOS image and flash. update checksums. success. boots.
3) repeat 2), but this time, checksums fail to update, and kexec files prevent a Default Boot.

so, this occured to me: the GPG is being fused into the _actual .rom file_, not a temporary copy.
i confirmed this by running diff on the .rom in my build-vm and the one on the USB, and they now differ. (maybe this is understood, but i did not expect this)

i propose that before flashing, the bios image be copied and renamed on the external storage device -- appended with '-gpg' for example -- so that the GPG key is never merged with the original rom. 
this way, a user can be sure that only 1 key is ever added to a rom, but also have the option of flashing other computers with the gpg injected rom.

and to your point, if a user selects "Add GPG to Running BIOS..." that already has a key in the keyring, or if a user attempts to "Add GPG to Standalone BIOS Image... " that is appended with '-gpg', then implement your suggestion -- prompt user to replace instead of add.

i think that would cover all bases as far as duplicate keys?

tlaurion, 2022-04-17:
@walliams adding the same key two times would not cause a bad behavior. Basically, Heads injects the keychain in ROM and GPG would neglect adding the same key.

I will try to test my hypothesis when back in front of computer in the next days.

tlaurion, 2022-04-17:
gpg --list-keys from recovery shell would show what is happening. cbfs-init script being responsible to export the files from ROM after having measured them, the files are then available in ramfs as part of / filesystem exposed to Heads.

usb-init from recovery shell, followed by gpg --card-status and gpg --list-keys will tell what happens.

flash-gui.sh contains the logic of backup of firmware, key injection and  flashing back to SPI.

walliams, 2022-04-17:
> @walliams adding the same key two times would not cause a bad behavior. Basically, Heads injects the keychain in ROM and GPG would neglect adding the same key.
> 
> I will try to test my hypothesis when back in front of computer in the next days.

yes, sorry i made a fatal mistake in my testing, did it again more intentionally after i made that hypothesis and proved myself wrong so i deleted that whole comment lol. i think you are on the right track.

jtmoree-github-com, 2022-05-11:
> is this process possible-- to have 2 laptops verified from the same HOTP Security Dongle? or am i still misunderstanding?
i am certain i have been trying to merge the most recently generated key with the firmware on the second laptop, but it is never able to update checksums.

@walliams  fyi.  I use the same usb security dongle with two laptops but only 1 has coreboot.  The process should be repeatable on multiple coreboot setups.  I created the public/private keys on a separate system and transferred to the security dongle before loading the public key into coreboot.    (of course I deleted the private key from the first system).

tlaurion, 2022-05-11:
@jtmoree-github-com where the same USB security dongle can be used to sign/verify /boot content across multiple devices, HOTP function of the USB security dongle is bound to a single Heads computer.

You can generate keypair externally and copy to card your subkeys to multiple USB security dongles, and use each privisioned USB dongles on a imdividual Heads computer, that is true.

USB security dongles supported by Heads only have one HOTP slot used for remote attestation.

This is a recurring question @jans23 @kylerankins : can that limitation be leveraged on the firmware level of those USB security dongles in the future, so that a single USB security dongle could be used to attest integrity of multiple Heads computers?

closed 2022-08-03 by tlaurion