linuxboot/heads #1158: x230 Coreboot error 1

mikeoxsmal, 2022-04-18:
2022-04-18 15:14:44-07:00 HASHES    build/x230-flash/tools.cpio
2022-04-18 15:14:44-07:00 CPIO      build/x230-flash/heads.cpio
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = "en_US",
	LC_ALL = (unset),
	LANG = "C MAKE=/home/michel/heads/build/make-4.2.1/make /home/michel/heads/build/make-4.2.1/make /home/michel/heads/build/x230-flash/heads.cpio"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
2022-04-18 15:14:44-07:00 UNCHANGED build/x230-flash/heads.cpio
dec5a954c805bff68f3d55769a4ad942cedf16502cbb1d5e3897d64272cc6ab3  /home/michel/heads/build/x230-flash/heads.cpio
2022-04-18 15:14:44-07:00 HASHES    build/x230-flash/heads.cpio
2022-04-18 15:14:44-07:00 MAKE coreboot
tail /home/michel/heads/build/log/coreboot.log
-----
/home/michel/heads/build/make-4.2.1/make -C "/home/michel/heads/build/coreboot-4.13/x230-flash" -C /home/michel/heads/build/coreboot-4.13 obj=/home/michel/heads/build/coreboot-4.13/x230-flash CROSS=/home/michel/heads/crossgcc/bin/i386-linux-musl- IASL=/home/michel/heads/build/coreboot-4.13/util/crossgcc/xgcc/bin/iasl DOTCONFIG=/home/michel/heads/build/coreboot-4.13/x230-flash/.config BUILD_TIMELESS=1 CFLAGS_x86_32=-fdebug-prefix-map=/home/michel/heads=heads -gno-record-gcc-switches -Wno-error=packed-not-aligned CFLAGS_x86_64=-fdebug-prefix-map=/home/michel/heads=heads -gno-record-gcc-switches -Wno-error=packed-not-aligned -j2 --max-load 16
make[1]: Entering directory '/home/michel/heads/build/coreboot-4.13'
    GEN        generated/assembly.inc
    CREATE     x230-flash/mainboard/lenovo/x230/cbfs-file.vPkvp4.out (from /home/michel/heads/build/coreboot-4.13/x230-flash/.config)
#
# configuration written to /home/michel/heads/build/coreboot-4.13/x230-flash/.config
#
    MAKE       /home/michel/heads/build/coreboot-4.13/vboot_lib/libvboot_host.a
vboot SHA256 built with tight loops (slower, smaller code size)
    CC            cgpt/cgpt_add.o
gcc: error: unrecognized command line option '-Wimplicit-fallthrough'
make[2]: *** [Makefile:1100: /home/michel/heads/build/coreboot-4.13/x230-flash/util/vboot_lib/cgpt/cgpt_add.o] Error 1
make[1]: *** [util/cbfstool/Makefile.inc:126: /home/michel/heads/build/coreboot-4.13/x230-flash/util/vboot_lib/libvboot_host.a] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/michel/heads/build/coreboot-4.13'
Makefile:504: recipe for target '/home/michel/heads/build/coreboot-4.13/x230-flash/.build' failed
make: *** [/home/michel/heads/build/coreboot-4.13/x230-flash/.build] Error 1

Can somone help me or tell me what im doing wrong
im using Ubuntu 16.04

tlaurion, 2022-04-19:
@mikeoxsmal hello there.
Ubuntu 16.04 ils a bit old but what is confusing here is that coreboot doesn't seem to be built with 4.13 cross compiling environment. On first build, musl-cross-make is build to build everything but coreboot, while per coreboot version cross compiling environment is built and used to  build coreboot. This doesn't seem to be the vmcase here, where "unrecognized command line option '-Wimplicit-fallthrough" should not happen.

Could you upload your whole logs/coreboot.* files?

mikeoxsmal, 2022-04-19:
[coreboot.log](https://github.com/osresearch/heads/files/8509215/coreboot.log)

tlaurion, 2022-04-19:
Ubuntu 20.04 and debian-11 are know to work, but the logs you gave unfortunate don't contain the console given logs on the cross compiling environement that should have been built

tlaurion, 2022-04-19:
You should as well build for x230-maximized or x230-hotp-maximized as x230 and x230-fladh are legacy boards.

tlaurion, 2022-04-19:
Full builds and logs are produced over CircleCI

Last commit builds as of now are here https://app.circleci.com/pipelines/github/osresearch/heads/405/workflows/18ef207d-afc6-4bd3-ab2a-7afd8d5066a1

tlaurion, 2022-04-19:
Prep steps
https://app.circleci.com/pipelines/github/osresearch/heads/405/workflows/18ef207d-afc6-4bd3-ab2a-7afd8d5066a1/jobs/3127/parallel-runs/0/steps/0-101

https://app.circleci.com/pipelines/github/osresearch/heads/405/workflows/18ef207d-afc6-4bd3-ab2a-7afd8d5066a1/jobs/3127/parallel-runs/0/steps/0-110

Building x230-maximized
https://app.circleci.com/pipelines/github/osresearch/heads/405/workflows/18ef207d-afc6-4bd3-ab2a-7afd8d5066a1/jobs/3139/parallel-runs/0/steps/0-103

closed 2026-04-20 by tlaurion