linuxboot/heads #1159: Does dom0 kernel 5.17.4+ work on anyone's T430 or X230 machines?

icequbes1, 2022-05-02:
In summary, starting from Qubes R4.1 dom0 kernel 5.17.4, a T430 (with embedded GPU) fails to boot by resulting in an unresponsive machine 1-3 seconds after the "kexec" output is shown from Heads.

I'm cross-referencing qubesos/qubes-issues#7480 here as the issue started after using the latest linux kernel. Debugging in that issue led to some conclusions that Xen is booting, the kernel is booting, but graphics issues might be causing a kernel panic which I can't see on-screen. Adding `nomodeset` to the kernel arguments appears to allow further boot such that the rootfs init is being executed, but the screen remains on the kexec output lines from Heads.

In the event this is something dependent on Heads, I'm raising the issue here to see if dom0 kernel 5.17.4 causes boot failures for those who are on Qubes R4.1.

tlaurion, 2022-05-02:
@icequbes1 : Which rom commit? Version of the board configs please?

icequbes1, 2022-05-02:
@tlaurion rom commit is latest on master - [fb5cfd5](https://github.com/osresearch/heads/commit/fb5cfd5cc2125f54a5e85447b81c1b970c416b0a).

`make BOARD=t430`

icequbes1, 2022-05-07:
I take it with these machines, most people are not using kernel-latest. The most recent kernel at the moment, 5.10.112, has no issue.

icequbes1, 2022-05-13:
Resolved in kernel-latest-5.17.7-1, QubesOS/qubes-issues#7480. Closing.

closed 2022-05-13 by icequbes1

icequbes1, 2022-10-11:
Based on the changelog of 5.17.4 [1], there is this commit against i915 which seems fitting:

drm/i915: Sunset igpu legacy mmap support based on GRAPHICS_VER_FULL

    The intent of the version check in the mmap ioctl was to maintain
    support for existing platforms (i.e., ADL/RPL and earlier), but drop
    support on all future igpu platforms.  As we've seen on the dgpu side,
    the hardware teams are using a more fine-grained numbering system for IP
    version numbers these days, so it's possible the version number
    associated with our next igpu could be some form of "12.xx" rather than
    13 or higher.  Comparing against the full ver.release number will ensure
    the intent of the check is maintained no matter what numbering the
    hardware teams settle on.

I do have an iGPU and the T430 might be considered legacy? I'll try to unpatch this and rebuild the kernel to verify.

So far, what I've learned is that i915 loading with modesetting enabled = freeze. Blacklisting i915 or setting i915.modeset=0 doesn't freeze, but no graphics after kexec.

[1]: https://cdn.kernel.org/pub/linux/kernel/v5.x/ChangeLog-5.17.4 commit 7959ca85c

tlaurion, 2022-10-12:
@icequbes1 i'm confused with this issue.

The x230 and t430 are the most supported Qubes laptops since they are certified. 
This comes with Qubes  being testing kernel updates on real hardware, where on some really corner cases, there is some kernel that reaches stable repositories without having been properly tested.

But that is unrelated to Heads? Unless for some obscure reasons, something should be added for the t430 to boot properly, which was not reported back into Heads but present in Nitrokey Heads repository?

Heads on the t430 initializes displays through coreboot, as opposed to the x230 which I am testing.
Then, when using fbwhiptail (default in t430 maximized), the framebuffer is initialized through Heads kernel (4.14 as of now, I know quite old but without any benefit for Heads other then adding support for other stuff like filesystems etc).

Then when booting OS, settings defined under grub configuration (/boot/grub2/grub.cfg) are being parsed and passed to kexec call. This is the OS kernel here, so Qubes dom0.

From my understanding of the following:
 

> Based on the changelog of 5.17.4 [1], there is this commit against i915 which seems fitting: drm/i915: Sunset igpu legacy mmap support based on GRAPHICS_VER_FULL     The intent of the version check [...]

I understand that you still have some issue booting your T430:

> So far, what I've learned is that i915 loading with modesetting enabled = freeze. Blacklisting i915 or setting i915.modeset=0 doesn't freeze, but no graphics after kexec.

The only thing I can do is to ask other board owners from #692 if they report the same issue  @Thrilleratplay @alexmaloteaux @lsafd @bwachter(iGPU) @shamen123 @eganonoa(iGPU) @nitrosimon @jans23 @icequbes1 (iGPU) @weyounsix (t430-dgpu) ?

If there is no answer here, I would recommend asking on Qubes forum, where a lot of other t430 owners, including Sven there, is active and using it as a daily driver.