linuxboot/heads #1170: Nitrokey flashes red despite GUI indicating HOTP:Success (W530-dgpu-k1000m-maximized)

Sirwhofromwhere, 2022-06-11:
First of all thanks for reading & helping.

After a recent install of Heads with Qubes 4.1, when I boot the Thinkpad with the nitrokey plugged in the nitrokey blinks green twice before anything is on the screen, then I get the bootsplash that I posted here
[image]
e

As soon as it goes to the last part of the bootsplash I get a series of 10 red flashes. In the boot menu I have the TOTP & It says HOTP:  Success

If I boot without the nitrokey, the video attached shows what happens after I put it back in & hit 'retry' on the prompt screen which background is yellow.

https://user-images.githubusercontent.com/104425324/173207264-07898455-ca20-4a9c-bb61-386beb7ac6c6.mp4

Perhaps this should be a separate issue, but potentially related is that I was unable to seal the disk unlock key in the TPM. When I got to seal it, I get this issue:

Do you wish to add disk encryption to the TPM (y/N): yEncrypted LVM group? (e.g qubes_dom0 or blank):
I left this blank and hit enter
Block Devices (blkld):
/dev/sda2 UUID="numbers"
/dev/sda1 UUID="numbers"
Encrypted devices? (e.g /dev/sda2 or blank): /dev/sda2It then flashes super quickly this series of messages, and kicks me back out to the chose default boot menuRunning kexec-save-key with parans: -s -p /boot /dev/sda2
Failed to get UUID for device /dev/sda2
Failed to save the disk key
Failed to save defaults

This happens whether I select /dev/sda2 or sda1.

Here is a video of this occurring:

https://user-images.githubusercontent.com/104425324/173207490-5712ca1a-5413-4961-99e6-f09a19e943de.mp4

Any help is greatly appreciated.

walliams, 2022-06-12:
have you set the hwclock?
i noticed on the initial splash you see the gpg was made in the future... i have had weird errors without setting the hwclock
in recovery shell:
`date -s YYYY-MM-DD hh:mm:ss` # Use UTC
`hwclock -w`

Sirwhofromwhere, 2022-06-12:
Thank you very much for your response @walliams.

Unfortunately I booted up the recovery shell & entered it as you specified (with UTC filled in) and also in the other suggested formats, and every time I hit enter it gives me the list of options for the date command without actually saving that date.

Any thoughts?

Once again, thanks.

walliams, 2022-06-12:
try just the date and then just the time:
`date -s YYYY-MM-DD`
`date -s hh:mm:ss`
`hwclock -w` # saves system date to hw

tlaurion, 2022-06-12:
@Sirwhofromwhere time is wrong, as pointed by @walliams .

You can go to recovery shell as he instructed and do that manually, or plug a network cable and call network-init-recovery, which will randomize Mac and attempt to sync time from NTP from your router (provided DNS server) and if that fails will reach NTP server on the internet.

I would have to reproduce this, but first picture shared shows error saying your public key is generated 308 days in the future and video showing your clock is in 2010.
The behavior is weird, since HOTP doesn't rely on time.

For the other issue I'm confused since blkid is showing youbhave 2 partitions found. But then complains that UUID fornsda2 cannot be found.... Please fix time and report back?

Sirwhofromwhere, 2022-06-12:
@walliams That worked, the date is not correct 2022-06-12 05:32:50

Unfortunately on reboot it still gave me the same red flashing with "HOTP Success" on main menu, even with the clock being properly set.

RE: the second issue of disk encryption in TPM @tlaurion 

After fixing the clock it still has this issue. When I type y to disk encryption, then leave the LVM section blank, then type in my disk /dev/sda2 it kicks me back to the boot options menu 

Should I try to clear & reown heads?

tlaurion, 2022-06-12:
@Sirwhofromwhere first screenshot shows error as @walliams had insight, saying signature was made 308 days in the future.

Your video shows time and date in 2010. Setting clock would be first step as pointed out.

The errors going fast and back to whiptail is a bug, they should be readable. Warnings and errors should at least pause for a second so the messages are readable.

As for your /dev/sda2 not being readable

tlaurion, 2022-06-12:
@Sirwhofromwhere OEM factory reset/Re-Owneership is the fastest option.

Test would be to use  regenerate digest and sign /boot option, then go back to options show boot option, then select first option, make default and say y to create a second Luis header key and set TPM disk unlock key passphrase.

But if you still get UUID not found, I'm a bit clueless and will need to push a PR that at least pauses for a second when a warn/error message comes from the console. Otherwise every of those happening between GUI is just not visible for the user.

tlaurion, 2022-06-12:
If you have no use for current gpg keypair, I would go the OEM factory/re-ownership  path, which would re-own TPM, factory reset USB security dongle, update firmware with generated public key, sign /boot digest and reboot. At that stage with good time, you would have to reseal HOTP/TOTP and then selecting default boot should guide you to setup a new default boot option with disk unlock key setup.

Sirwhofromwhere, 2022-06-12:
@tlaurion This is bizare. I completed the whole re-ownership process, and when I go to generate a new HOTP secret, the security token lights up red once it says it has successfully put the secret on the nitrokey.

So the same issue is occurring and its occurring from that very first step.

tlaurion, 2022-06-12:
Red once is normal (operation is red)

Legend:
Intermittent red (nonstop): HOTP fails
Steady red: operation
Intermittent green(short term): HOTP success

tlaurion, 2022-06-12:
When you generated gpg keypairs, dongle was steady red while the on card operation was happening.

HOTP operation takes only a second.

On reboot, on verification, it should flash green.

Please report back

Sirwhofromwhere, 2022-06-12:
@tlaurion Sorry I wasn't clear. I meant it keeps flashing red after it says it has successfully put the secret on the nitrokey.

Poor phrasing by me.

tlaurion, 2022-06-12:
And on reboot HOTP is OK on screen but dongle keeps flashing red....

tlaurion, 2022-06-12:
Seems like a good case for Nitrokey support ticket here. 

Picture of System Info?

Sirwhofromwhere, 2022-06-12:
Restting gpg key has a solid green light on the card that lasted the whole resetting gpg key process

On reownership I chose the following options.

Would you like to change the current LUKS disk recovery key passphrase: n
Would you like to re-encrypt: n

Would you like to set one password for...: n
Would you like to set individual passwords for...: y

Set the tpm ownership password, gpg admin and user key.

The no to both custom information for the gnpupg key and exporting the public key.

Then it checks for the dongle, flashes green then solid green for on-token operation.

Could this indicate tampering with the nitrokey?

tlaurion, 2022-06-12:
Or commit Id built/name of ROM downloaded/flashed?

Nothing changed in the code used for HOTP for a really long while...

Sirwhofromwhere, 2022-06-12:
I downloaded from the CircleCL w530-dgpu-k1000m board.

I verified the hash on it. I tried to build it myself but unfortunately I have limitations about how I expose my laptop to networks that led me to use the more expedited route due to threat model.

tlaurion, 2022-06-12:
I confirm that for a little moment they had a bug

https://github.com/Nitrokey/nitrokey-pro-firmware/issues/88

Seems like your dongle will always have inverse flashing pattern. Was not aware until searching.

You can contact them for replacement or live with it.

Sirwhofromwhere, 2022-06-12:
@tlaurion So to verify this I should change /boot files & see if I get green flashing ?

tlaurion, 2022-06-12:
Signing yes

Sirwhofromwhere, 2022-06-12:
@tlaurion Last question, when it was resetting the GPG key for reownership, the light on the Nitrokey was hard green.

Does that mean that ordinarily it is supposed to be hard red?

And also, I get flashing red 10 times when I boot when it says success, does this mean that it is meant to be 10 times green flash? I thought it was supposed to just be hard green, so in my case wouldn't that be hard red?

tlaurion, 2022-06-12:
Because

> Nitrokey/nitrokey-pro-firmware#88

Your key
Legend:
Intermittent green (nonstop forever): HOTP fails
Steady green: operation ongoing
Intermittent red(short term then stops): HOTP success

Sirwhofromwhere, 2022-06-12:
Perfect. @tlaurion Thank you greatly for your help.

tlaurion, 2022-06-12:
> @tlaurion Last question, when it was resetting the GPG key for reownership, the light on the Nitrokey was hard green.
> 
> Does that mean that ordinarily it is supposed to be hard red

Again yes. Because Nitrokey/nitrokey-pro-firmware#88

You should contact nitrokey.

tlaurion, 2022-06-12:
Or see if they will acknowledge the issue of dongles that should not have been shipped and why their led patterns is not documented anymore on their website

@szszszsz @jans23

tlaurion, 2022-06-12:
@Sirwhofromwhere https://github.com/Nitrokey/nitrokey-pro-firmware/releases/tag/v0.14

tlaurion, 2022-06-13:
Next time, if anything weird like this happens. The way to get Nitrokey/Librem Key flashed firmware version is through recovery shell with the command:
`htop_verification info`

Cross-linking from nitrokey-pro, opening issue there.

szszszsz, 2022-06-13:
> As soon as it goes to the last part of the bootsplash I get a series of 10 red flashes. In the boot menu I have the TOTP & It says HOTP: Success

Hello,
1. For a certain hardware revision indeed the firmware v0.13 (as seen in the very first screenshot) had the LEDs switched. I confirm this was fixed in the firmware release v0.14, which is the latest one logically wise (v0.15 just handle more hardware revisions).
2. The correct behavior is 20 blinks green for the correct HOTP, or "infinite" red blinking (LED behavior for the HOTP verification takes priority over other LED behavior).
3. I will pass further the suggestion to check the documentation for the LED patterns for the HOTP verification.

Listing again connected tickets/releases for completeness:
- https://github.com/Nitrokey/nitrokey-pro-firmware/issues/88
- https://github.com/Nitrokey/nitrokey-pro-firmware/commit/1d867fcb09e87ce581f9f0833e1222c2a961f804
- https://github.com/Nitrokey/nitrokey-pro-firmware/releases/tag/v0.14
- https://github.com/Nitrokey/nitrokey-pro-firmware/blob/5164178f0d47f8564186b6d71d90a5f0920cb5d8/src/keyboard/report_protocol.c#L463-L471

jans23, 2022-06-14:
@Sirwhofromwhere Did you execute a OEM factory reset?

Sirwhofromwhere, 2022-06-14:
> @Sirwhofromwhere Did you execute a OEM factory reset?

Yes, I did. Multiple times.

szszszsz, 2022-06-14:
@Sirwhofromwhere Sorry, that's my mistake. The actual blink count is 10, while the other half is simply off state.
Either way I recommend to run firmware update to v0.14 to make it fixed (among other improvements). The tool for that is still tested and lacks complete setup documentation, but it should work already on Linux with libnitrokey installed. If you are interested please take a look at this development guide:
- https://github.com/Nitrokey/nitrokey-pro-firmware/blob/master/DEVELOPMENT.md#firmware-update

I hope it will be rephrased in the coming weeks, and published in the proper place at [docs.nitrokey.com](https://docs.nitrokey.com/pro/linux/index.html)

Sirwhofromwhere, 2022-06-15:
@szszszsz I'm stuck here with Debian-11 I don't have GLIBC_2.23 which is required, and apparently to install it would create a host of other issues re: dependencies. However, on Fedora-34 (I can upgrade to 35) I can't find the libhidapi-dev & libusb which are required for libnitrokey.

What should I do here?

If I should be creating a separate issue for this on nitrokey someone let me know, as this is my first time using Github.

Thanks Szczepan.

szszszsz, 2022-06-15:
@Sirwhofromwhere Feel free to write to https://support.nitrokey.com/ (preferred, more users) or support@nitrokey.com for the further instructions, so we would not pollute this ticket. Alternatively you can make a new ticket on pynitrokey: https://github.com/Nitrokey/pynitrokey/issues , as this concerns the actual setup problems.
I expect we will have a proper guide in the coming week or so.

Regarding Fedora, the packages for build are:
```
$ sudo dnf install hidapi-devel libusb1-devel
```
Also it should be fine to use the Fedora's `libnitrokey` package (version 3.5), without building it yourself.

Either way let's move with further the discussion to somewhere else. Please link here the new place if you decide to use Github or Nitrokey Support forum.

Sirwhofromwhere, 2022-06-15:
Issue resolved. I updated the firmware to v0.14 and I am now getting the correct LED's.

I was unable to do this on my Qubes installation however, and had to create a Fedora 36 workstation live USB, downloading Libnitrokey, pynitrokey to use nitropy commands as listed in the above document from szszsz.

This was a very fast and easy process now that I was on the live USB.

Thank you to @walliams @tlaurion and @szszszsz for taking the time to help me find a resolution to this issue. Apologies for typing too much irrelevance in to this ticket.