linuxboot/heads #1179: Terrible performance & screen freezing on w530-dgpu-K1000m-hotp-maximized board with Qubes 4.1

Sirwhofromwhere, 2022-06-22:
I'm running a w530-dgpu-K1000m-hotp-maximized board with Qubes 4.1 with latest kernel, updated etc, 16gb RAM.

I'm experiencing terrible performance in VM's and total screen freezing requiring hard reboot, some examples are as follows:

- Booted Qubes and walked away from laptop, came back to find it frozen.
- Experiencing freezing during both initialization and shutdown, not every time, but it happens.
- Freezing during installation.
- If I open a VM and try to watch a youtube video, the tab crashes, and if I persist, I get sudden VM shutdown. If I try to watch a youtube video and open another VM, I get screen freezing.
- If I try to watch a VLC video in a VM after a short period the screen will freeze

These are some examples of the issues, I can elaborate further and get more specific if needed.

I've tried to install Qubes on a different SSD, it froze during installation. I've had issues where the RAM doesn't seem to be dynamically balancing between Qubes, and the default 4gb MAX ram allocation would result in freezing if I opened four of those VM's. Updated Dom0, kernel etc with no effect.

Some have speculated it could have to do with Nvidia drivers, and @tlaurion has mentioned the following:

> The way I see it here to properly support and diagnose issue would be to compare performance with nvramtool enablement, and if that improves the situation, have coreboot build nvramtool and even have the config activated in prepared ROM? 

Can someone please advise me on how to enable nvramtool?

Any logs you need, diagnostic efforts I should make, or further elaboration, please let me know.

Thank you all for any help you can provide. @eganonoa

tlaurion, 2022-06-22:
@Sirwhofromwhere the w530 comes in a lot of flavors, currwntly with dgpu enablement support (two dgpu) or without (hotp and without).

If memory is not an issue, my next test would be to go to the w530 variant not adding support for dgpu. As of now, you are not using dgpu (which on w530, from my understanding but I could be wrong, is used with external displays which you are not using nor testing for the moment. Qubes does not enable external display unless told so by enabling display extension or mirroring).

By upgrading to w530-maximzed, you could roll out freezing and bad perf being caused on dgpu since the firmware would not be exposing the dgpu existence.

Sirwhofromwhere, 2022-06-23:
@tlaurion Great suggestions, I'll begin this now and get back to you.

I've just put in 12GB of different RAM, and had the same freezing (even faster, instant after entering desktop). 3x4gb. I'll try memtest as well to absolutely rule it out, but it doesn't look like that's the issue.

> I understand that you are using board config without hotp

Can I just clarify what you mean by this? I am using HOTP board with nitrokey.

So I should build & flash the w530-maximized-hotp board that doesn't have dGPU through the Options -> Flash the firmware with a new ROM, retain settings?

Thanks, and please give me an XMR so I don't have to go to the trouble of converting it for Open Initiative. In the future i'll just donate bitcoin direct to that, but if you have an XMR it's a relatively small $ figure so that'd be easiest. I don't care if you buy coffee with it :'D

tlaurion, 2022-06-23:
> I'm running a w530-dgpu-K1000m-hotp-maximized

OK

> I'll try memtest as well to absolutely rule it out

Excellent, yes

> So I should build & flash the w530-maximized-hotp board that doesn't have dGPU through the Options -> Flash the firmware with a new ROM, retain settings?

Or download, yes.

> Thanks, and please give me an XMR so I don't have to go to the trouble of converting it for Open Initiative. In the future i'll just donate bitcoin

Your convenience would be my inconvenience for reasons explained on the chat. Thanks for your understanding: Insurgo Initiative with its open book is the way to support this project.

Sirwhofromwhere, 2022-06-25:
@tlaurion It appears that I have a single faulty RAM slot in the internal RAM section.

I ran 9 passes of memtest on both RAM sticks in my outer RAM slots with no errors, I then ran a test with both the same sticks in the internal, with immediate errors mounting up. I ran a test with one stick of RAM in the top, and subsequently the bottom slot, and both had errors. So it appears it was the internal faulty RAM slots that was causing this grief.

I have no idea what to do now about that specifically, but I'm grateful for your support, as I had tried to get some kind of RAM test on the laptop but it was a pain in the behind.

If you have any idea about what can be done about faulty RAM slots let me know, but it's clear this isn't a HEADS nor Qubes issue.

This issue can be closed.

closed 2022-06-25 by Sirwhofromwhere