linuxboot/heads #1186: W530 t530 t520 @wiki

3hhh, 2022-07-02:
I noticed that W530 t530 t520 apparently [are supported](https://github.com/osresearch/heads/pull/1074) by heads. :-)

However the [wiki indicates that they are not](https://github.com/osresearch/heads-wiki/blob/master/Installing-and-Configuring/Prerequisites.md).
I guess it needs updating?

tlaurion, 2022-07-03:
@3hhh yes to all to above.

Documentation is scattered in PR created by @eganonoa and would need someone to take the lead to create guides related to disassembling and flashing.

A discussion is happening under https://forum.qubes-os.org/t/request-for-advice-to-install-coreboot-heads-on-thinkpad-w530/12301/9

Will crosslink here, hopefully for collaboration in creating the docs

tlaurion, 2022-07-03:
Documentation issues should happen under heads-wiki though. Issues related to Heads code should happen here.

tlaurion, 2022-07-03:
Issue already opened under https://github.com/osresearch/heads-wiki/issues/83

closed 2022-07-03 by tlaurion