linuxboot/heads #1187: Severe performance issues with 32GB RAM, but works fine with 16GB. W530

Sirwhofromwhere, 2022-07-05:
**What I run**

Latest HEADS w530-hotp-k1000m-maximized on a W530 i7-3840QM with updated Qubes.

**The issue**

Internal motherboard RAM slots causing serious performance issues to where 32GB of RAM is critically slow, yet 16GB of ram in the external RAM slots works fine.

**Troubleshooting done**

Tested both 2x4GB and now the full 4x8GB RAM in the laptop with memtest64, running 8+ passes on each without error. 

To note, I created a previous issue here that ultimately turned out to be due to using my 2x8GB RAM in the internal slots during installation and use. When I ran memtest on those specific RAM modules (2x8GB SP 1600) I would get immediate and voluminous errors, whereas running them in the external RAM slots would return none over many passes.

I've just now acquired another identical 2x8GB SP RAM and attempted to load my installation that works fine with 16GB using the full 32GB. I get slower boot times, and upon loading the performance has been critically impacted. Slower everything. No crashing yet observed however/

When I run a memtest on the **new** RAM modules (2x8GB) in the internal motherboard RAM slots, I get no issues on many memtest passes.

What I'd like to know are some reasonable troubleshooting methods I can make to understand why 32GB of working RAM would be causing such issues, and also how HEADS might be affecting the RAM. 

 **Thank you greatly for any and all assistance.**

tlaurion, 2022-07-05:
@Sirwhofromwhere what are your

> get slower boot times and critically impacted performance ?

Do you get errors from dom0 with `sudo dmesg` ?

@walliams @eganonoa (owners of w530) do you see lower performances with 32gb ram?

tlaurion, 2022-07-05:
Qubes has this opened issue https://github.com/QubesOS/qubes-issues/issues/7404

tlaurion, 2022-07-06:
The next step, if I understand well, might be to test a good external slot memory to internal slot. And isolate which internal slot is problematic.

Reading the above I'm still not clear which ones of the internal slot would be faulty or if one memory stick is at cause?

Sirwhofromwhere, 2022-07-06:
@tlaurion I've now run three different sets of RAM through the internal slots with immediate failures in the slot. One 4GB RAM stick, one 8GB SP, and another separate batch 8GB SP.

Seems like it must be the slot that is at fault, as those memory sticks work fine in the external slots. Very bizarre, but nothing I can do about it as I think getting the memory slot replaced might be prohibitively expensive, as will buying a new motherboard.

I tried both slots before individually with one stick of the 8GB RAM and both failed, just one failed slower. I'll try again to see if I can isolate a slot, but I believe both are at fault.

Thanks for your efforts in helping me.

Sirwhofromwhere, 2022-07-07:
Closing due to isolating issue to the internal RAM slot or other motherboard fault.

closed 2022-07-07 by Sirwhofromwhere