linuxboot/heads #1189: Create 8mb flash images for t430/w530 (to bypass complete disassembly requirement)

tlaurion, 2022-07-14:
This way no complète disassembly would ne needed.

Since the 8mb flash exposed is actually the one containing ifd GBE and me, the equivalent of x230-flash could be created to flash theaximized full 12mb image internally.

Fell upon https://ch1p.io/coreboot-t530-one-chip/ while searching for disassembly instructions, to realize that if this was automated there would be no nee d for total disassembly.

tlaurion, 2022-07-14:
Naming proposition: w530-flash-bottom t530-flash-bottom?

zifxify, 2022-07-15:
I did the following to bypass complete disassembly for flashing the w530.

**Downgrade BIOS to 2.58 or an earlier version** for W530 (I downgraded to 2.50 if I remembered correctly)
You can do it as described on the following link (https://github.com/gch1p/thinkpad-bios-software-flashing-guide#downgrading-bios)

EC downgrade is not needed.

**Read the 8MB chip, I'm using the ch341a**

`flashrom -p ch341a_spi -c "MX25L6406E/MX25L6408E" -r w530_8mb`
`flashrom -p ch341a_spi -c "MX25L6406E/MX25L6408E" -r w530-2_8mb`

**Check if the same**

`diff w530_8mb w530-2_8mb`

**Unlock firmware descriptor and ME region**

`./ifdtool -u w530_8mb`

**Flash the 8MB chip with unlocked descriptor**

`flashrom -p ch341a_spi -c "MX25L6406E/MX25L6408E" -w w530_8mb.new`

**Install and boot 1vyrain.** See https://github.com/n4ru/1vyrain#installing

**1. Press enter key to attempt BIOS exploit.**
**2. Press the power button to wake it up.**
**3. When asked: "Please enter a choice" press CTRL-C**
**4. Mount a USB drive with a 12MB w530 heads rom and save the original bios** (for restoring back to stock if needed later.)
`eg: mount /dev/sd* /mnt `
`cd /mnt`
`flashrom -p internal:laptop=force_I_want_a_brick -r 12mb-original.rom`
`flashrom -p internal:laptop=force_I_want_a_brick -w w530*.rom`
**5. Umount USB drive and reboot system**

tlaurion, 2023-12-18:
So, got my w530 dgpu K1000 today. Disasembled keyboard+palmrest, and saw 8bm spi chip exposed only. 
Tried my generic programmer clip: too large. Had to find my pamona and rewire.
Updated https://github.com/linuxboot/heads-wiki/issues/120#issuecomment-1861846082

Now, lets make a w530-flasher_exposed_under_palmrest for fun and profit.

tlaurion, 2023-12-20:
#1558 won't work.

Field experience is that if crafting a 8mb self-contained firmware image that neuters ME, reduces it size and modify IFD that reduces ME region to minimal and maximizes BIOS region, we basically get a 8MB flash image that becomes a legacy board, not a maximized board that has 12mb of combined SPI space.

The problem here is this:
- To be able to boot the w530/t530 on the BIOS contianed in the 8mb chip alone, the IFD cannot span the BIOS region across the two combined SPI chips into opaque combined SPI space. If the IFD specifies the whole 12MB availabe space for BIOS region, when booting the computer, it boots on the proprietary BIOS with its implemented BIOS lock protections which prevents writing to the BIOS region unless 1vyrain exploit chain is ran. No win
- When modifying IFD to limit BIOS region to 8mb region alone, Heads flasher boots as expected. But when attempting to flash 12mb maximized image, flashrom fails with "read_flash: failed to read (0x001000..0xbffff)" which is flashrom unable to read or write to the 4mb region at the end of the 12mb region. Flashrom cannot read or write to the 4mb chip, even though IFD, ME, GBE and BIOS region are unlocked per IFD config. I might be missing something but I haven't been successful getting that BIOS lock bypass from IFD/coreboot config changes alone.

So next steps as of now:
- returning to backup of 8mb
- using windows coming with refurbished laptop to run https://github.com/n4ru/IVprep 
- creating backup of 8mb, ifdtool -u it, flash back
- use ivyrain to run exploit and then ctrl-c, flashing maximized 12mb image internally.
- See how long that takes and maybe reconsider disassembling whole laptop

Alternative:
- Try to figure out how to do a gadget that would pin/clip/be stable enough to use exposed cn100 pad as shown under https://github.com/linuxboot/heads-wiki/issues/83#issuecomment-1173130373
- Try to figure out report to coreboot author of https://doc.coreboot.org/mainboard/lenovo/w530.html the "easy trick" to pin CS to next chip (I don't understand. Might be the easiest but unsure how to properly do that)

tlaurion, 2024-01-09:
Can't be done https://github.com/linuxboot/heads/pull/1558#issuecomment-1883694085

tlaurion, 2024-01-09:
Note that 8mb only could be done (legacy images). But since we try to get away of legacy boards.... That would need to have volonteers to struggle with me with smaller firmware footprint and jump into #590 revival.

closed 2024-01-09 by tlaurion

zifxify, 2024-01-22:
@tlaurion you don't need full windows to flash back to previous version. (to much hassle)
It's a long time ago but if I remembered correctly I only did the following. Extracted the .FL1,FL2(ec) with geteltorito.
Created a bootable FREEDOS stick (with rufus (on windows :) ) I think.

I will check if I can find my stick and the  right commands I used. 
Something like below:

`dosflash.exe /sd /file .FL1`
-> the command below is actually not necessary
`dosflash.exe /sd /ipf ec /file .FL2`