linuxboot/heads #1193: Error 1 Error 2

githubuseravailable, 2022-07-21:
I have below error while building heads: make BOARD=x230-maximized
But not sure about the cause. i have searched the issue list, and couldn't find solution. 
Please kindly help. 

arch/arm64/Makefile:27: ld does not support --fix-cortex-a53-843419; kernel may be susceptible to erratum
arch/arm64/Makefile:48: Detected assembler with broken .inst; disassembly will be unreliable
  CHK     include/config/kernel.release
Cannot use CONFIG_CC_STACKPROTECTOR_STRONG: -fstack-protector-strong not supported by compiler
make[3]: *** [/home/pi/heads/build/linux-4.14.62/Makefile:1110: prepare-compiler-check] Error 1
make[3]: *** Waiting for unfinished jobs....
  UPD     include/config/kernel.release
make[2]: *** [Makefile:146: sub-make] Error 2
make[1]: *** [Makefile:24: __sub-make] Error 2
make[1]: Leaving directory '/home/pi/heads/build/linux-4.14.62/linux-x230-maximized'
make: *** [Makefile:390: /home/pi/heads/build/linux-4.14.62/linux-x230-maximized/.build] Error 1

tlaurion, 2022-07-22:
@githubuseravailable: Suggestion is to build with debian-11, which is what is used over CircleCI as a docker image.

githubuseravailable, 2022-07-23:
i built with Debian 11, the error above was produced in Debian 11. 

PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

Distributor ID:	Debian
Description:	Debian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye

tlaurion, 2022-07-25:
@githubuseravailable You are attempting to cross compile from a rpi? (user PI and cortex warning being hints?)

If so, you are hitting a situation I unfortunately cannot really help with as of now.

githubuseravailable, 2022-07-30:
@tlaurion thanks, let me try

githubuseravailable, 2022-09-11:
hi @tlaurion , please kindly help me with this , 
I read this page https://osresearch.net/Configuring-Keys/
and could not understand this part: 
"Ensure your USB security dongle and the USB drive with your key are still inserted."

i have Yubikey and empty USB drive inserted, but then heads cannot find the key, 
so does it mean that, i need to export the public key from Yubikey, 
then store it into USB drive ?

tlaurion, 2022-09-11:
@githubuseravailable your yubikey is already provisioned? You have a copy of your public key? 

That public key would need to be put on USB drive, and imported under Heads following referred instructions.

Otherwise this is a new yubikey?

In all cases, the name of the issue and what is asked here has nothing to do one with another.

Please close this issue and open another issue. This seems to be documentation related and should be opened under heads-wiki instead.

githubuseravailable, 2022-09-11:
@tlaurion yes, it's new yubikey, assuming it has not been provisioned yet, also i'm totally new with yubikey. 
okay, i'm going to close this issue.

but, i couldn't find any place to post issue in the heads-wiki https://github.com/osresearch/heads/wiki
is this the correct link ?

tlaurion, 2022-09-11:
@githubuseravailable https://github.com/osresearch/head-wiki

githubuseravailable, 2022-09-11:
@tlaurion thanks

closed 2022-09-11 by githubuseravailable