linuxboot/heads #1197: Set hybrid igpu dgpu by default on boards including dgpu blobs?

tlaurion, 2022-08-03:
t430/t530/w530 opened relevant (add others)

The problem/feature is that currently deployed coreboot configs are setting  both STATIC_OPTION_TABLE and USE_OPTION_TABLE.

USE_OPTION_TABLE instructs coreboot to read CMOS settings from file if not existing in memory, while STATIC_OPTION_TABLE instructs coreboot to overwrite settings in memory with the content of cbfs content on each boot, whatever is set in memory, not letting the user change settings but by flashing changes back into rom.

See https://github.com/coreboot/coreboot/blob/master/src/Kconfig

By removing STATIC_OPTION_TABLE, we could add either nvramtool or another GUI tool under heads rom that can manipulate directly CMOS settings, which would be kept in CMOS memory, and survive firmware upgrades.

The ROM would come with defaults from coreboot, populate cmos if not existing, but  otherwise customizations would live until CMOS battery is disconnected.

The danger of removing STATIC_OPTION_TABLE and providing nvramtool is that if the user sets a setting that causes problem, CMOS battery would need disconnection to rollback to what is stored into compiled ROM cmos_layout.txt

tlaurion, 2022-08-03:
Another option would be to create hybrid boards, or embed a different cmos_layout file into coreboot configurations for boards having dgpus

https://github.com/coreboot/coreboot/blob/master/src/Kconfig#L734

I think it would make sense to activate hybrid graphics automatically for boards which integrate dgpu blobs.

Otherwise I'm not sure why we would want to include such blobs if end user is not intending to activate dgpu. Or why igpu would be deactivated. Point being that my understanding for such boards is that hybrid mode is desired, so not sure why it would be anything else then in hybrid mode?

tlaurion, 2022-08-03:
x220 (xx20): @techge @Thrilleratplay @BlackMaria
t420 (xx20): @alexmaloteaux @natterangell (iGPU) @akfhasodh
t430 (xx30): @Thrilleratplay @alexmaloteaux @nickfrostatx @lsafd @bwachter @shamen123 @eganonoa(iGPU) @nitrosimon @jans23 @walliams (dGPU) @icequbes1 (iGPU)
t430s and additional information (xx30): @Siproqu
t520 (xx30): @eganonoa
t530 (xx30): @eganonoa
w530 (xx30): @eganonoa @zifxify @walliams (w530-dgpu-K2000m)

weyounsix, 2022-08-04:
i own w530-dgpu-K2000m, t430-dgpu, and i would like to see this. would it be possible to call nvramtool during make so that the setting is baked into the rom by default?

natterangell, 2022-08-06:
I have iGPU only, but nvramtool included would be great!

tlaurion, 2022-08-17:
@natterangell @weyounsix 2 answers, two different opinions, no other participation.

If we refer to OP, choices are
- remove static option table.
  - Have nvramtool under heads being able to change settings in CMOS. And have user remove battery to revert to default. Won't survive firmware upgrade.
- Keep static option table
  - have nvramutil under heads, have nvramutil modify the ROM directly. Won't survive upgrade.
  - have nvramutil modify ROM post build. Invalidates the ROM hashes.

Not aware of other options. I would tend to prefer the removal of static option table under dgpu boards and have nvramtool under heads, where users can change settings in CMOS and simply disconnect the battery if they mess up, which would get them back to compiled defaults.

natterangell, 2022-08-19:
Seems removing static option table makes sense.

tlaurion, 2022-08-19:
@natterangell @weyounsix modified my previous post. Please choose path.

bwachter, 2022-08-25:
I'm not touching anything with a nvidia chip, so all my thinkpads are iGPU only.

tlaurion, 2022-08-25:
@bwachter then if you use igpu only you should not compile with a dgpu enabled board?

natterangell, 2022-08-30:
Removing static option table seems prudent.

tlaurion, 2022-09-15:
> i own w530-dgpu-K2000m, t430-dgpu, and i would like to see this. would it be possible to call nvramtool during make so that the setting is baked into the rom by default?

I have created a PoC which included a hack around building ifdtool from coreboot to be cross-compiled, so that it could be included into packaged rom.

This required
- coreboot module file to copy coreboot util (here ifdtool) to be copied into its own directory, so that it could be cross-compile: https://github.com/tlaurion/heads/commit/a4b5381e65f0f2da5fbbe742696ad6b15f1b9bc2#diff-18936189b28399cf48703d0c1ec1df33e57c559de2a12f4438be00e6813bdb68R53
- Add an environment variable into board configurations to be able to ask for that util to be compiled and packaged under initrd, here NVUTIL: https://github.com/tlaurion/heads/commit/a4b5381e65f0f2da5fbbe742696ad6b15f1b9bc2#diff-fb98d6f892bb15e235b3e7ad3689cc1602832d9fdcb50d1fbe447286973043c9R25
- Then when declared for a board, Makefile added magic actually compiles and packages additional ifdtool: https://github.com/tlaurion/heads/commit/a4b5381e65f0f2da5fbbe742696ad6b15f1b9bc2#diff-76ed074a9305c04054cdebb9e9aad2d818052b07091de1f20cad0bbac34ffb52R464-R470

A similar logic could be added for nvramutil to be packaged under dGPU boards.
Then, users could call nvramutil from under Heads and modify cmos values, which would persist in CMOS, if STATIC_OPTION_TABLE is removed under coreboot configuratio nfor a board.

Someone of yous are interesting into proposing a PR filling your need following that guidance?
I see @weyounsix being interested to test. Someone up to the task of filling their own needs?