linuxboot/heads #1198: Have all modules point to either github/debian archives for tarballs

tlaurion, 2022-08-03:
Heads had several build breakages in the past because repositories go either down, have certificate problems etc.

Pointing to github releases or Debian archives would mitigate this issue for reproducibility in the future.
Neither of those should have either cert/DNS issues that go unnoticed.

Those may require checksums update, though under modules, as opposed to bouncing packages from mirror to mirrors like we do from time to time. But building an old Heads commit after that change should be possible as long as the recipes from CircleCI are respected to have the same docker image (OS version) and installed packages.

tlaurion, 2023-07-11:
Becomes relevant again, see #1439

JonathonHall-Purism, 2024-01-02:
Maybe we could add locations as secondary mirrors instead of switching everything over to another source?

If we have another place (or more than one) to store archives, we could upload whatever version we are using and have the build system try those sources as secondary mirrors.  Mirrors could be down at any given time (builds only break if there are no reachable sources for a given archive).  Digests don't have to change right away so it has a lower initial impact.

Even if we're not perfect about uploading everything we use, we usually have a copy of the relevant archive when the primary source goes down (and build system checks the digest), we could upload it reactively.

Tricky part there is finding place(s) to put the archives I think.  As a secondary mirror, I might be able to do something on cloud.puri.sm, or maybe source.puri.sm could handle it (does not sound significantly different from anything already done for PureOS).

tlaurion, 2024-01-02:
Hey @githubisnonfree (Leah), we discussed over libreboot channel a while back the possibility of hosting, and mirroring, tarballs used by Heads. Does the offer still stands?

The easiest way to do this as of today is to simply build Heads locally:
- one server variant (kgpe-d16)
- one hotp variant (x230-hotp-maximized) 

And doing so, cairo/pixman/whiptail/fbwhiptail and all other dependencies would be downloaded locally with their checksums validated.

The result could then be uploaded to what is then mirrored to/from others. And then Heads could find a way either statically/better, dynamically, to point/parse to libreboot mirror list to get new mirror list in case prior sources are failing to download.

Thoughts?

---

Sidenote, does libreboot plan to rebridge to Matrix? Otherwise, how can we foster better collaboration?
Matrix supports OFTC, Soonet, W3C, gitter.im until librechat<->Matrix resolve their withstanding issues. What would be the way to bridge the gap between our communities?

PR #1576 (2024-01-09): Makefile: Support mirrors for dependency source packages
Try to download dependency source packages from mirrors if the primary source fails or the archive has changed.

Move the download and verify logic to bin/fetch_source_archive.sh.  The mirror list is here, currently only
https://storage.puri.sm/heads-packages/, but others can be added.  The mirror list is randomized to load each mirror equally.

The verify logic is moved to this script too so it can fail over to a mirror (or another mirror) if a mismatched archive is served, not just for a failure.  Makefile no longer needs to verify sep [... cut]
closed 2024-01-09 by tlaurion