linuxboot/heads #1203: Have qemu boards support writeable pflash and internal flashing

tlaurion, 2022-08-24:
Seems like both qemu and flashrom will need to be patched. Meanwhile, ijecting key is made from seperate qemu board build statement, and "reflashing" is rebooting qemu/kvm with newer built rom.

------
Current limitations stated under OP under #1188, where the following traces are what needs to be resolved: 

> @JonathonHall-Purism : would like to know what is the issue with nvram you were talking about earlier, and also what avenues possible with pflash since from what I've read from libvirt XML and qemu doc, the flash is not writable. Any input you have there would be useful in other open issues

Looks like the size limit is just a matter of changing this default for max_fw_size: https://gitlab.com/qemu-project/qemu/-/blob/master/hw/i386/pc.c#L1833

Re: writable flash, it seems to exist but the only discussion I can find is about people using it incorrectly: https://bugs.launchpad.net/qemu/+bug/1818367/comments/4

>The <loader> element, with @type='pflash', no other attributes, and then
>no <nvram> sibling element either, happens to be valid, but it is an
>extremely niche use case. It is used when you have a unified, writeable,
>OVMF.fd file that contains both the firmware executable and the live
>variable store.

This does sound like what we want, I have no idea right now how this works with plain qemu, whether flashrom supports it, or what it would take to add.

_Originally posted by @JonathonHall-Purism in https://github.com/osresearch/heads/issues/1188#issuecomment-1218004671_

tlaurion, 2022-09-30:
Leaving traces of pertinent mailing list sub threads

- spi chips implementation https://www.mail-archive.com/qemu-devel@nongnu.org/msg58003.html

tlaurion, 2024-05-24:
Track https://github.com/Dasharo/dasharo-issues/issues/828

tlaurion, 2024-05-24:
Specifically
- https://review.coreboot.org/c/coreboot/+/82555
- https://review.coreboot.org/c/coreboot/+/82639

As referred by https://github.com/Dasharo/dasharo-issues/issues/828

tlaurion, 2024-11-21:
Question asked under Dasharo OSFV matrix channel at https://matrix.to/#/!MwWOJhMJzlxIdAQxae:matrix.3mdeb.com/$-G0q4GBkLoQwCk7d1geSf-U74XuqTNwtSb6HT072K-s?via=matrix.org&via=nitro.chat&via=matrix.3mdeb.com

tlaurion, 2024-11-21:
Of course, changes to qemu would need to be merged inside of nix, and then flake.lock under Heads using this new built qemu, included under nix based created docker images under Heads.

mkopec, 2024-12-10:
This would definitely unblock some tests for us.

We planned to use qemu for automating part of Dasharo tests but we found that many features (OEM factory reset and GPG most notably) depend on being able to write to the flash. Without this ability a lot of our testing has to be done manually on hardware instead

pietrushnic, 2024-12-10:
I achieved writes in QEMU and Dasharo (coreboot+SeaBIOS) with sortbootorder as a secondary payload. Someone would have to build that with Heads for QEMU, but I need to see how it would work out of the box if coreboot contains @krystian-hebel driver.

How I tested it? I used:

* https://github.com/Dasharo/open-source-firmware-validation/pull/574
* Dasharo (coreboot+SeaBIOS) from this action: https://github.com/Dasharo/dasharo-pq/actions/runs/12128880813

tlaurion, 2024-12-12:
> if coreboot contains @krystian-hebel driver.

Eli5?

pietrushnic, 2024-12-12:
https://review.coreboot.org/c/coreboot/+/82555

mkopec, 2024-12-12:
That patch enables Qemu flash writes in coreboot, we would need someone to port this code to flashrom, that's what Heads uses to write to flash

krystian-hebel, 2024-12-12:
I'm not sure if pflash is something that can be easily added to flashrom, there is no way of probing for it, other than trying to write, which may fail miserably for any real flash device. Would it be good enough to add a dedicated app/script (I've been testing it with `busybox devmem`) just for QEMU instead?

tlaurion, 2024-12-12:
> I'm not sure if pflash is something that can be easily added to flashrom, there is no way of probing for it, other than trying to write, which may fail miserably for any real flash device. Would it [...]

Since https://github.com/linuxboot/heads/pull/1769, Heads is not hardcoding use of flashrom/flashprog, but instead requires each board to describe what is used to flash internally, now flashprog because of --progress output (and because @i-c-o-n responded and fixes everything that was needed for Heads tht flashrom took years to).

The ideal, of course, would be to have internal flashing of qemu pflash from flashprog/flashrom with same minimal arguments so that Heads do not have to implement another set of workarounds just for qemu, while keeping the same output that is existing for all other boards for automated testing.

If we take a look at a random #1769 board config change:
- inclusion of flashprog module binaries inside initrd: https://github.com/linuxboot/heads/pull/1769/files#diff-4bc64e2c24813aa57928559e3b09022cfdc65249bc09b4b43b48c0835065c6dfR31
- internal flasher command line (`NOTE: without -r/-w/-v arguments, without filename argument)` https://github.com/linuxboot/heads/pull/1769/files#diff-4bc64e2c24813aa57928559e3b09022cfdc65249bc09b4b43b48c0835065c6dfR77

So as log as the internal flasher used permits similar flashrom/flashprog arugments (-r/-w/-v  + filename), then Heads will prepend any internal flashing call with the content of `CONFIG_FLASH_OPTIONS="flashprog --progress --programmer internal"`

Note that flash.sh script currently uses BOARD name to exclude qemu boards as of today and print a warning on console pointing to qemu.md docs.

The ideal, in my opinion, would be the addition or a qemu programmer to flashrom/flashprog, ie `flashprog --progress --programmer qemu` which would know what to do, and exclude probing from internal programmer. Thoughts?

i-c-o-n, 2024-12-14:
Wrt. flashprog, adding another "programmer" would be possible, but not sure if worth it. I see a few options with the `internal` programmer as well:
1. Run `qemu` with some specific flash ID. Just tested `-global driver=cfi.pflash01,property=id0,value=0xab -global driver=cfi.pflash01,property=id1,value=0xcd` and could at least see the `0xcd` in a log from `flashprog -p internal:laptop=this_is_not_a_laptop`. This would require to add some QEMU-compatible, bigger parallel flash chips to the database (virtually nothing above 512KiB was used for BIOS's, so such chips were never added). Not sure if it's fun to dig into parallel flash in 2024, though.
2. Add CFI support to flashprog. CFI seems to be for automatic probing of flash parameters and similar to SFDP for SPI flash. Probably more fun than 1., but it's a 25 year old standard, so no use beyond QEMU probably.
3. Actually emulate a "modern" controller: https://github.com/9elements/qemu/commits/qemu_ich9_spi I never had a close look and don't know why it didn't go upstream. But the idea seems right and we could test code paths that are much closer to those used on current hardware. IMO, getting this upstream would benefit the ecosystem the most and is what I would pursue. What I don't know: does the SPI flash emulation in QEMU have all the wanted features? i.e. anything that the `pflash` driver has and is missing?

tlaurion, 2024-12-22:
@pietrushnic @mkopec @krystian-hebel thoughts on prior comment?

tlaurion, 2026-01-17:
Seems like basic requirements in coreboot are for q35

```
CONFIG_BOARD_EMULATION_QEMU_X86_Q35=y
CONFIG_BOOT_DEVICE_SUPPORTS_WRITES=y
CONFIG_SPI_FLASH=y
```

And then qemu command line changes

 ```
-M q35,max-fw-size=16M \
  -drive if=pflash,format=raw,file=coreboot.rom,unit=0 \
  -m 2048 \
  -enable-kvm \
  -kernel /path/to/heads/build/bzImage \
  -initrd /path/to/heads/build/initrd
```

flashrom/flashprog internal programmer should work, if all qemu related exceptions in code and inject_gpg is removed to make things universal. Key injection and flashtool/cbfs should cooperate as intended, with exception that q35 doesn't have an ifd.

tlaurion, 2026-08-31:
### Fact correction on current SoA

Stock qemu still has **no ICH9 SPI controller** on the x86/Q35 machines. `-drive if=pflash` attaches the ROM as a memory-mapped boot device (`cfi.pflash01`), but there is no functional SPI controller behind the SPIBAR, so `flashprog -p internal` (which drives the ICH9 SPI controller at LPC RCRB +0x3800) fails with `FCERR=1` / "Transaction error!" / "Running OPCODE 0x9f failed" — the trace in the OP. `if=mtd` itself is standard on the ARM/Aspeed machines; what was never done (and still isn't) is wiring it into the Q35/ICH9 side qemu upstream.

**The qemu-side requirement** — what exposes HW SPI to flashprog/flashrom:

- Compare: https://github.com/qemu/qemu/compare/master...9elements:qemu:upstream-qemu_ich9_spi
- 9elements' patchset adds `hw/isa/spi_ich9.c` + ICH9 SPI emulation behind the SPIBAR; `-drive if=mtd,format=raw,file=...` then attaches a real serial flash over the SPI controller, which is what `flashprog -p internal` consumes. Same patchset is what allows a coreboot build for ICH9 to use a UEFI variable store / write back.

**Why it's stuck in a fork.** The series, `[PATCH 0/3] hw/i386: Add MTD controller on ICH9` (Oct 2023), got a request-for-changes from Michael S. Tsirkin:

> "the feature is guest-visible, so it must have a flag to disable it, tied to existing machine types"

…i.e. migration/device-compat gating behind the per-machine-type flags, and the series was never resubmitted to address it: https://lists.gnu.org/archive/html/qemu-devel/2023-10/msg05779.html

**What needs to happen next.** Pick a lane, in increasing order of effort/upstream value:

1. **Validate the fork end-to-end.** Use the 9elements build (8.1.50) in CI first to prove `flashprog -p internal` reads/writes flash in the QEMU guest — cheap, and confirms whether any *further* work is even warranted before investing upstream.
2. **Carry the patch on top of current heads's nix's used qemu.** Apply the (small, ~3-file) series onto a current qemu rather than downgrading to 8.1.50, so the docker image stays current. Only residual risk is whether the series applies cleanly to the newer base.
3. **Get it upstream properly.** Adapt the series to satisfy the machine-type-gated-disable-flag request and push it again. This is the only path that removes the need for a fork permanently.

**Persistence angle.** Today almost nothing in the qemu/coreboot world seems to write back to the ROM to simulate persistence — "reflash" is really "reboot with a freshly built ROM." But once the SPI controller patch is in place, `flashprog -p internal --write` should write through to the attached image, which would give heads a genuine persisted-flash workbench in CI and permit removal of all qemu gating under Heads (change firmware → reboot → the change is still there), matching real-hardware behavior.