linuxboot/heads #1204: x330 support

toothlesslizard, 2022-08-24:
Hi. First impressions.

3612Q - detected
FHD/2k - poor performance HiDPI on Q 4.1.1 latest kernel ( xfce 4.14 )
Also no support in Heads.
AX210 - bad support (fedora/debian), too new for linux 
NVME M2 2230 mod doesn’t work out of box need enable in coreboot config.
New motherboard - new spi chip (not listed in Heads, so can’t trust /boot with new gpg key)

tlaurion, 2022-08-24:
https://github.com/xy-tech/x330-bios ?

toothlesslizard, 2022-08-24:
@tlaurion yes, based on tianocore. I'll try to fix it myself at night

tlaurion, 2022-08-24:
@toothlesslizard : If I understand upstream work correctly (interesting mod, but haven't read everything of it), coreboot support for this board is basically a x230+edp, where the vbt there should (but to be verified by adventurous board owner):
https://github.com/xy-tech/x330-bios/blob/main/coreboot/dotconfig#L101

So basically, it seems you could use #967 as a base.
PR welcome, as usual!

tlaurion, 2022-08-24:
Not sure why they include the vgabios (maybe tianocore)
https://github.com/xy-tech/x330-bios/blob/main/coreboot/dotconfig#L150

tlaurion, 2022-08-24:
Yep, confirm that they seem to use upstream patch applied to #967 :)
https://github.com/xy-tech/x330-bios/blob/main/coreboot/dotconfig#L191

toothlesslizard, 2022-08-24:
spi 

-c "MX25L3205D/MX25L3208D" - 4MB
-c "MX25L6405D"  - 8MB

tlaurion, 2022-08-24:
> -c "MX25L3208D"
https://osresearch.net/x230-maximized-flashing/ 

> -c "MX25L3205D/MX25L3208D" - 4MB
Not sure why it was changed, interested.

All of those should be supported externally from latest flashrom version available from your OS (linux) repositories.
It would be to verify to know if opaque (combined flash internal flashing) will be possible when inside of Heads though, from currently used flashrom version compiled to be part of heads binaries at commit https://github.com/osresearch/heads/blob/master/modules/flashrom#L5

tlaurion, 2022-08-24:
> -c "MX25L3205D/MX25L3208D" - 4MB

Seems supported since 8 years ago: https://github.com/flashrom/flashrom/blame/d4091395546fe9d2452f74cd1a9c4c9e0fc7fb4d/flashchips.h#L507

You'll have to test and see.