linuxboot/heads #1207: Support Canokey under Qemu+tpm - fastpath?

tlaurion, 2022-08-28:
https://qemu.readthedocs.io/en/latest/system/devices/canokey.html#canokey

tlaurion, 2022-08-28:
Idea here would be to have a fully hardware independent testing platform, where no hotp implementation (no hardware keys) would be needed, while permitting to extend Heads features for testers and contributers, and test variations between yubikey/Nitrokey/Librem Key and Canokey implementation, since we already had issues in the past with default being different and having different behaviors

Related to: #1076 without hardware

daringer, 2022-08-29:
Looks like a very promising idea, although I have no experience with it and cannot tell if it comes w/o or with further challenges...

tlaurion, 2023-06-05:
Small update from https://docs.canokeys.org/userguide/openpgp/#supported-algorithm
no RSA4096 nor RSA3076. Meh.

ZenithalHourlyRate, 2023-06-05:
> no RSA4096 nor RSA3076. Meh.

The doc is only for the actual hardware. The virtual card actually supports RSA3072 now and generating private keys on card is possible (with small code modification) against the MbedTLS crypto backend. The hardware does not have the new firmware installed and its crypto backend is not capable of generating RSA private key of such length so in the doc we do not say it supports that.

tlaurion, 2023-06-07:
> no RSA4096 nor RSA3076. Meh.
>
> The doc is only for the actual hardware. The virtual card actually supports RSA3072 now and generating private keys on card is possible (with small code modification) [...]

Thank you.

Once we move to nix layer, we will add a qemu-coreboot-tpm1/2 testing board supporting canokey virtual card. Maybe sooner then that.

closed 2024-05-13 by tlaurion