linuxboot/heads #1215: W530-HOTP-maximized does not allow navigation of the heads menu, and Qubes password decryption fails

Tailsorheads, 2022-09-20:
## Please identify some basic details to help process the report

On an updated Qubes 4.1 with W530-hotp-maximized board, everything has worked fine, and then randomly my password, which is correct, begins to fail to decrypt. Then when I go to scroll through the menu to boot options I find I can't move in the menu at all. The only thing I can do is hit enter, which of course loads the default boot options which fails to decrypt, or hit esc, which recycles back to the same menu.

No other key registers a reaction.

### A. Provide Hardware Details

**1. What board are you using (see list of boards [here](https://github.com/eganonoa/heads/tree/master/boards))?**

W530-hotp-maximized

**2. Does your computer have a dGPU or is it iGPU-only?**
It has a dGPU but I am not running the dGPU board. And it was working fine for a long time prior.

**3. Who installed Heads on this computer?**

- [1 ] Self-installed

**4. What PGP key is being used?**

- [1 ] Nitrokey Pro 2

**5. Are you using the PGP key to provide HOTP verification?**
- [ 1] Yes

### B. Identify how the board was flashed

**1. Is this problem related to updating heads or flashing it for the first time?**
- [1 ] First-time flash
- [ ] Updating heads 

**3. How was Heads initially flashed**
- [1 ] External flashing

**4. Was the board flashed with a maximized or non-maximized/legacy rom?**
- [1 ] Maximized

**5. If Heads was externally flashed, was IFD unlocked?**
- [1 ] Yes

### C. Identify the rom related to this bug report

**1. Did you download or build the rom at issue in this bug report?**
- [1 ] I downloaded it

**2. If you downloaded your rom, where did you get it from?**
- [1 ] Heads CircleCi

*Please provide the release number or otherwise identify the rom downloaded*

**3. If you built your rom, which repository:branch did you use?**
- [ 1] Heads:Master

**4. What version of coreboot did you use in building?**
- [1 ] I don't know

## Please describe the problem

**Describe the bug**
Heads boots and verifies the HOTP fine, but when I default boot Qubes fails to decrypt with what I am sure is the right password. If I try to move up or down in the HEADS boot menu, nothing happens. Only esc recycles the menu and enter boots default.

**To Reproduce**
No idea. Just turn on the laptop.

**Expected behavior**
Qubes decrypts as expected, and I am able to navigate using the up down key through the menu options.

tlaurion, 2022-09-20:
@Tailsorheads 

> 1. Is this problem related to updating heads or flashing it for the first time?
> 
> [1 ] First-time flash

Seems like the keyboard connection is not steady. I would recommend reconnecting

Tailsorheads, 2022-09-20:
@tlaurion Then why would my luks be failing to decrypt when I am 100% entering the right password?

Also, when I'm in the Qubes graphic to enter the password, the up key works as expected (cycles between gui view and CLI), backspace works etc.

Obviously all the keys work if I can type out my password and perform that function without error.

I just reseated the keyboard cable in my laptop and the same issue occurs.

Something else must be wrong here, something with HEADS.

Tailsorheads, 2022-09-21:
@tlaurion Apologies for not noticing your first reply, my attention was drawn only to the second one.

I can enter a recovery shell by spamming 'r' at boot.

I can type everything I want there.

It is indeed the in-built keyboard on my W530.

Tailsorheads, 2022-09-21:
@tlaurion I'm embarrassed to report it is a faulty keyboard. Only 3 of the keys do not work, the down arrow, the b & the n, which affected my password and cycling the menu.

Sorry for wasting your time. I just assumed because it appeared as though I could type out all of my characters of the password, that it must've been a fault with HEADS.

Really should've figured that one out myself, again sorry & thanks for what you do.

closed 2022-09-21 by Tailsorheads