linuxboot/heads #1217: x230-maximized failed at building gcc: ada header init?

sebek0F9, 2022-09-22:
I dont think its a bug, maybe I'm missing something. Here's the last parts of the code before the error. I'm trying to build
x230-maximized, for the lenovo. building on an aarch64 machine, debian 11.

near the end this is shown:
**../../gcc-8.3.0/gcc/ada/init.c:575:18: error: missing binary operator before token "("**

can someone guide me?

g++ -fno-PIE -c   -O2  -march=armv8-a -mtune=cortex-a72  -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../../gcc-8.3.0/gcc -I../../gcc-8.3.0/gcc/. -I../../gcc-8.3.0/gcc/../include -I../../gcc-8.3.0/gcc/../libcpp/include -I/home/*****/apps/lenovo/heads/build/x86/coreboot-4.13/util/crossgcc/xgcc/include -I/home/*****/apps/lenovo/heads/build/x86/coreboot-4.13/util/crossgcc/xgcc/include -I/home/*****/apps/lenovo/heads/build/x86/coreboot-4.13/util/crossgcc/xgcc/include  -I../../gcc-8.3.0/gcc/../libdecnumber -I../../gcc-8.3.0/gcc/../libdecnumber/dpd -I../libdecnumber -I../../gcc-8.3.0/gcc/../libbacktrace   -o pretty-print.o -MT pretty-print.o -MMD -MP -MF ./.deps/pretty-print.TPo ../../gcc-8.3.0/gcc/pretty-print.c
g++ -fno-PIE -c  -DLOCALEDIR=\"/home/*****/apps/lenovo/heads/build/x86/coreboot-4.13/util/crossgcc/xgcc/share/locale\" -O2  -march=armv8-a -mtune=cortex-a72  -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../../gcc-8.3.0/gcc -I../../gcc-8.3.0/gcc/. -I../../gcc-8.3.0/gcc/../include -I../../gcc-8.3.0/gcc/../libcpp/include -I/home/*****/apps/lenovo/heads/build/x86/coreboot-4.13/util/crossgcc/xgcc/include -I/home/*****/apps/lenovo/heads/build/x86/coreboot-4.13/util/crossgcc/xgcc/include -I/home/*****/apps/lenovo/heads/build/x86/coreboot-4.13/util/crossgcc/xgcc/include  -I../../gcc-8.3.0/gcc/../libdecnumber -I../../gcc-8.3.0/gcc/../libdecnumber/dpd -I../libdecnumber -I../../gcc-8.3.0/gcc/../libbacktrace   -o intl.o -MT intl.o -MMD -MP -MF ./.deps/intl.TPo ../../gcc-8.3.0/gcc/intl.c
g++ -fno-PIE -c   -O2  -march=armv8-a -mtune=cortex-a72  -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../../gcc-8.3.0/gcc -I../../gcc-8.3.0/gcc/. -I../../gcc-8.3.0/gcc/../include -I../../gcc-8.3.0/gcc/../libcpp/include -I/home/*****/apps/lenovo/heads/build/x86/coreboot-4.13/util/crossgcc/xgcc/include -I/home/*****/apps/lenovo/heads/build/x86/coreboot-4.13/util/crossgcc/xgcc/include -I/home/*****/apps/lenovo/heads/build/x86/coreboot-4.13/util/crossgcc/xgcc/include  -I../../gcc-8.3.0/gcc/../libdecnumber -I../../gcc-8.3.0/gcc/../libdecnumber/dpd -I../libdecnumber -I../../gcc-8.3.0/gcc/../libbacktrace   -o sbitmap.o -MT sbitmap.o -MMD -MP -MF ./.deps/sbitmap.TPo ../../gcc-8.3.0/gcc/sbitmap.c
g++ -fno-PIE -c   -O2  -march=armv8-a -mtune=cortex-a72  -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../../gcc-8.3.0/gcc -I../../gcc-8.3.0/gcc/. -I../../gcc-8.3.0/gcc/../include -I../../gcc-8.3.0/gcc/../libcpp/include -I/home/*****/apps/lenovo/heads/build/x86/coreboot-4.13/util/crossgcc/xgcc/include -I/home/*****/apps/lenovo/heads/build/x86/coreboot-4.13/util/crossgcc/xgcc/include -I/home/*****/apps/lenovo/heads/build/x86/coreboot-4.13/util/crossgcc/xgcc/include  -I../../gcc-8.3.0/gcc/../libdecnumber -I../../gcc-8.3.0/gcc/../libdecnumber/dpd -I../libdecnumber -I../../gcc-8.3.0/gcc/../libbacktrace   -o input.o -MT input.o -MMD -MP -MF ./.deps/input.TPo ../../gcc-8.3.0/gcc/input.c
g++ -fno-PIE -c  -DBASEVER="\"8.3.0\"" -DDATESTAMP="\"\"" -DREVISION="\"\"" -DDEVPHASE="\"\"" -DPKGVERSION="\"(coreboot toolchain v ) \"" -DBUGURL="\"<https://gcc.gnu.org/bugs/>\"" -O2  -march=armv8-a -mtune=cortex-a72  -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../../gcc-8.3.0/gcc -I../../gcc-8.3.0/gcc/. -I../../gcc-8.3.0/gcc/../include -I../../gcc-8.3.0/gcc/../libcpp/include -I/home/*****/apps/lenovo/heads/build/x86/coreboot-4.13/util/crossgcc/xgcc/include -I/home/*****/apps/lenovo/heads/build/x86/coreboot-4.13/util/crossgcc/xgcc/include -I/home/*****/apps/lenovo/heads/build/x86/coreboot-4.13/util/crossgcc/xgcc/include  -I../../gcc-8.3.0/gcc/../libdecnumber -I../../gcc-8.3.0/gcc/../libdecnumber/dpd -I../libdecnumber -I../../gcc-8.3.0/gcc/../libbacktrace   -o version.o -MT version.o -MMD -MP -MF ./.deps/version.TPo ../../gcc-8.3.0/gcc/version.c
g++ -fno-PIE -c   -O2  -march=armv8-a -mtune=cortex-a72  -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../../gcc-8.3.0/gcc -I../../gcc-8.3.0/gcc/. -I../../gcc-8.3.0/gcc/../include -I../../gcc-8.3.0/gcc/../libcpp/include -I/home/*****/apps/lenovo/heads/build/x86/coreboot-4.13/util/crossgcc/xgcc/include -I/home/*****/apps/lenovo/heads/build/x86/coreboot-4.13/util/crossgcc/xgcc/include -I/home/*****/apps/lenovo/heads/build/x86/coreboot-4.13/util/crossgcc/xgcc/include  -I../../gcc-8.3.0/gcc/../libdecnumber -I../../gcc-8.3.0/gcc/../libdecnumber/dpd -I../libdecnumber -I../../gcc-8.3.0/gcc/../libbacktrace   -o ggc-none.o -MT ggc-none.o -MMD -MP -MF ./.deps/ggc-none.TPo ../../gcc-8.3.0/gcc/ggc-none.c
g++ -fno-PIE -c   -O2  -march=armv8-a -mtune=cortex-a72  -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../../gcc-8.3.0/gcc -I../../gcc-8.3.0/gcc/. -I../../gcc-8.3.0/gcc/../include -I../../gcc-8.3.0/gcc/../libcpp/include -I/home/*****/apps/lenovo/heads/build/x86/coreboot-4.13/util/crossgcc/xgcc/include -I/home/*****/apps/lenovo/heads/build/x86/coreboot-4.13/util/crossgcc/xgcc/include -I/home/*****/apps/lenovo/heads/build/x86/coreboot-4.13/util/crossgcc/xgcc/include  -I../../gcc-8.3.0/gcc/../libdecnumber -I../../gcc-8.3.0/gcc/../libdecnumber/dpd -I../libdecnumber -I../../gcc-8.3.0/gcc/../libbacktrace   -o memory-block.o -MT memory-block.o -MMD -MP -MF ./.deps/memory-block.TPo ../../gcc-8.3.0/gcc/memory-block.cc
g++ -fno-PIE -c   -O2  -march=armv8-a -mtune=cortex-a72  -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../../gcc-8.3.0/gcc -I../../gcc-8.3.0/gcc/. -I../../gcc-8.3.0/gcc/../include -I../../gcc-8.3.0/gcc/../libcpp/include -I/home/*****/apps/lenovo/heads/build/x86/coreboot-4.13/util/crossgcc/xgcc/include -I/home/*****/apps/lenovo/heads/build/x86/coreboot-4.13/util/crossgcc/xgcc/include -I/home/*****/apps/lenovo/heads/build/x86/coreboot-4.13/util/crossgcc/xgcc/include  -I../../gcc-8.3.0/gcc/../libdecnumber -I../../gcc-8.3.0/gcc/../libdecnumber/dpd -I../libdecnumber -I../../gcc-8.3.0/gcc/../libbacktrace   -o selftest-diagnostic.o -MT selftest-diagnostic.o -MMD -MP -MF ./.deps/selftest-diagnostic.TPo ../../gcc-8.3.0/gcc/selftest-diagnostic.c
g++ -fno-PIE -c   -O2  -march=armv8-a -mtune=cortex-a72  -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedant [... cut ...]
                 from ../../gcc-8.3.0/gcc/system.h:298,
                 from ../../gcc-8.3.0/gcc/ada/init.c:65:
../../gcc-8.3.0/gcc/ada/init.c:575:18: error: missing binary operator before token "("
  575 | # if 16 * 1024 < MINSIGSTKSZ
      |                  ^~~~~~~~~~~
make[1]: *** [Makefile:1110: ada/init.o] Error 1
make[1]: *** Waiting for unfinished jobs....
rm gcc.pod
make[1]: Leaving directory '/home/****/apps/lenovo/heads/build/x86/coreboot-4.13/util/crossgcc/build-i386-elf-GCC/gcc'
make: *** [Makefile:4244: all-gcc] Error 2

sebek0F9, 2022-09-22:
[Uploading build.log.bak…]()

sebek0F9, 2022-09-22:
heres the log:
[build2.log](https://github.com/osresearch/heads/files/9623571/build2.log)

tlaurion, 2022-09-22:
gcc-gnat installed on host OS?

I have not tested cross compilation from other arch, and as of now building is tested only from debian-11

sebek0F9, 2022-09-22:
I have gnat Installed on host Debian-11.

so I should try it on an x86_64 machine maybe? Or an i386 machine?

sebek0F9, 2022-09-22:
Ah, ok. I’ll give it a try .
Thank you.

closed 2022-09-22 by sebek0F9

sebek0F9, 2022-09-25:
just wanted to say it went passed that error i had, im trying a build out on circleci. 
im new to all this stuff, but especially, new at using ci stuff. i wasnt sure how to set the make option to specify BOARD=x230-maximized in the circleci UI, so i just edited the yml file in my fork and commented out what i thought wasn't necessary for x230.

thanks.

tlaurion, 2022-09-25:
@sebek0F9 confused on your past comment since those boards are generating ROMs by default.

sebek0F9, 2022-09-25:
when i added the heads project (my fork) to my circleci, and i started the build, it was building for all boards. (or i think it was?)

tlaurion, 2022-09-25:
@sebek0F9 yes. CircleCI builds are building all boards. This is why I got confused since x230-maximized is built by default?

I am confused by the nature of your comment. 

By starting a build on individual CircleCI instance, you can verify reproducibility. Otherwise you are building for your specific commits on your branches.

Can you speicfy what you are trying to accomplish here?

sebek0F9, 2022-09-25:
i want to flash the two bios chips on my x230 properly, with the ME cleaned, to prepare it for qubesos. i need to build the firmware images/roms of heads to start.

tlaurion, 2022-09-25:
@sebek0F9 you can download latest artifacts as specified in download instructions at https://app.circleci.com/pipelines/github/osresearch/heads for latest commit builds for chosen platform 

As of today, latest commit build, as for your own CircleCI instance for latest commit, should be something similar to 
https://app.circleci.com/pipelines/github/osresearch/heads/453/workflows/c30e5296-a092-4e26-afa1-5682aedd15aa/jobs/4702/artifacts

tlaurion, 2022-09-25:
Again, and as explained in other tickets, CircleCI builds will be reproducible since build paths will be the same (/root/project) as opposed to local builds. 

So if you build with CircleCI you should get get the same hashes. Otherwise, as long as reproducible issues and build system is replaced it will not be reproducible.

tlaurion, 2022-09-25:
@sebek0F9 if you want to check reproducibility of builds, clone repo and make your CiclreCi build reproduce outcome, and check hashes.txt for your platform

sebek0F9, 2022-09-30:
I understand now.
@tlaurion Thank you so much . I've said before, I really appreciate the help and information you give openly without restriction and promply. It's hard to find that.

sebek0F9, 2022-09-30:
the top rom doesn't download from here? mozilla firefox-esr, i just see a grid of black dots/diamonds when i click the link. tries to redirect me to the AWS. 
all the other links worked . strange...

edit: using wget from the terminal and it worked. i don't think i should be doing that...
ah, right clicking the link and "save as" worked.

https://app.circleci.com/pipelines/github/osresearch/heads/446/workflows/53d5397c-7eeb-41f4-a441-a2ac4dc212c7/jobs/4580

tlaurion, 2022-09-30:
> the top rom doesn't download from here? mozilla firefox-esr, i just see a grid of black dots/diamonds when i click the link. tries to redirect me to the AWS.
> all the other links worked . strange... [...]

@sebek0F9 yes, as documented under Downloading section of the rendered wiki
https://osresearch.net/Downloading#how-to-download