linuxboot/heads #1220: Heads' System Information is x86 centric

tlaurion, 2022-09-26:
`![2022-09-26-150328](https://user-images.githubusercontent.com/827570/192359540-d5b0304b-6c6b-41d3-bdce-c433084bb6b7.png)

- for x86, CPU version can easily be obtained grepping "cpu model" from /proc/cpuinfo.
On Power, it seems that this is `model` only. 

- FW_VER cannot be extracted from dmesg grepping DMI/BIOS.
Both can have the firmware version extracted through coreboot console output with the following:
```
cbmem -1 | grep Heads | awk -F " " {'print $1'} | awk -F "--" {'print $NF'} 
| uniq
```
Outputs: `Heads-v5.0.1-212-g585e581d`

macpijan, 2022-09-27:
@SergiiDmytruk Can `/proc/cpuinfo` content depend on what we put in the firmware?

Do you have any more suggestions on getting the CPU info here?

SergiiDmytruk, 2022-09-27:
> Can `/proc/cpuinfo` content depend on what we put in the firmware?

It probably can to some extend (e.g., if it comes from device tree), but `/proc` contents can also be arch-specific.

krystian-hebel, 2022-09-27:
CPU model can't be done through `/proc/cpuinfo`, at least without patches to Linux:
- x86 https://github.com/torvalds/linux/blob/master/arch/x86/kernel/cpu/proc.c#L72
- PPC https://github.com/torvalds/linux/blob/master/arch/powerpc/kernel/setup-common.c#L228

Also, what would be printed if there were two different CPU models installed? :thinking: 

As for FW_VER, there is some kind of information in device tree, it should be obtainable through sysfs. However, it is read from PNOR version partition and isn't reported correctly for coreboot, because possible entries are [hardcoded in Skiboot](https://github.com/open-power/skiboot/blob/master/core/flash-firmware-versions.c#L25) (and few other components, but these shouldn't be too much of a problem for coreboot). We haven't decided on how to approach this without breaking anything else yet.

tlaurion, 2022-09-27:
@krystian-hebel 

> CPU model can't be done through /proc/cpuinfo, at least without patches to Linux:
>
>     x86 https://github.com/torvalds/linux/blob/master/arch/x86/kernel/cpu/proc.c#L72
>     PPC [...]

Question is if the following would be satisfying as a base for further parsing:

```
cat /proc/cpuinfo | grep -e "cpu" -e "revision" | sort | uniq
cpu		: POWER9, altivec supported
revision	: 2.1 (pvr 004e 1201)
```
If two different CPUs, we would have 2 different revisions strings. Not sure who would put two different revisions of CPUs either? Something that happens a lot in the wild?

> As for FW_VER, there is some kind of information in device tree, it should be obtainable through sysfs. However, it is read from PNOR version partition and isn't reported correctly for coreboot, [...]

From Heads perspective,
`cbmem -1 | grep Heads | awk -F " " {'print $1'} | awk -F "--" {'print $NF'} | uniq` reports Heads correct versioning:  `Heads-v5.0.1-212-g585e581d` (without coreboot standard's `CBET4000` reported from DMI).  That is, release+commit#+g+commitId.

@SergiiDmytruk : Good enough? (FWUPD/LVFS related question/comparison).

That would change how FW_VER is extracted under Heads at https://github.com/osresearch/heads/blob/f2e45c4396456679284eeda9a26196ee7a3a14d1/initrd/init#L116-L119

Which is then used as is in `System Information` https://github.com/osresearch/heads/blob/f2e45c4396456679284eeda9a26196ee7a3a14d1/initrd/bin/gui-init#L514

SergiiDmytruk, 2022-09-29:
> Good enough? (FWUPD/LVFS related question/comparison).

You mean for versioning of firmware for LVFS? It might need a bit of reformatting to match one of [supported formats](https://fwupd.org/lvfs/docs/metainfo/version), maybe to make it look like `5.0.1.212`.