linuxboot/heads #1221: Patch for Coreboot

natterangell, 2022-10-03:
Hi,

I apologize as this is not technically an issue with Heads, but I'm requesting guidance here as it might be relevant for other T420 users too.

There is an [unmerged patch](https://review.coreboot.org/c/coreboot/+/39043) on Coreboot Gerrit that fixes an issue with wwan cards on the T420. In short, the patch enables power on wwan if slot has SMBus connected. Without this, the card will pull the lines low, causing DRAM init to fail. 

This patch allows me to use a Sierra Wireless MC7304 LTE capable card, instead of my ancient Ericsson F3507g, which is rather nice since older 2G/3G cards are more or less irrelevant these days.

 I suppose the current Makefile verifies hashes for coreboot(?). So, how should I manually edit the coreboot source files when building Heads?

tlaurion, 2022-10-04:
@natterangell you would have adapt that patch and apply it from patches/coreboot-4.13/wwan-smbus.patch or similar, which is applied on top of validated coreboot 4.13 tarball sha256sum hash.

Note that patches are applied only when the build dir doesn't exist, so you have to remove coreboot directory under build/x86/coreboot for that patch to be applied.

You can test application of your patch on top of coreboot tarball extraction in another directory first. Package is under packages/arch/ 

This patch doesn't seem complete and implementation was questioned seeing comments.

Tag me if you need more info.

tlaurion, 2022-10-04:
@natterangell [2dad98d](https://github.com/tlaurion/heads/commit/2dad98d123dfbdd47eddfec314d129784fa5c0ca) just adds the upstream patch to be applied against coreboot 4.13 without further modification.

Doing so in a branch on your github, followed under your CircleCI account would result in a build starting to build all roms and see if things break at a build level.

This is happening with a blunt addition of upstream patch in a branch over my github for Heads project at:
https://app.circleci.com/pipelines/github/tlaurion/heads/1209/workflows/f7310d77-131b-440c-96ac-df2903c3f11a/jobs/10749

If the commit goes green, it means the build passed without breaking anything. Otherwise, the build having failed will show where the error happened under "Making Board" last lines of such board. Basically, in this first step, failing will probably be the result of a patch not being directly applicable, showing what lines of the patch needs to be reworked for it to be successfully applied.

tlaurion, 2022-10-04:
@natterangell : we see it failing in the first coreboot 4.13 built board at https://app.circleci.com/pipelines/github/tlaurion/heads/1209/workflows/f7310d77-131b-440c-96ac-df2903c3f11a/jobs/10751?invite=true#step-103-55
```

Applying patch file : patches/coreboot-4.13/0081-wwan-smbus.patch 
patching file src/ec/lenovo/h8/Makefile.inc
patching file src/ec/lenovo/h8/wwan.c
Hunk #1 succeeded at 24 with fuzz 2 (offset -13 lines).
patching file src/mainboard/lenovo/t420/early_init.c
Hunk #1 succeeded at 1 with fuzz 1 (offset -15 lines).
Hunk #2 succeeded at 61 (offset -15 lines).
patching file src/mainboard/lenovo/t430/early_init.c
Hunk #1 succeeded at 2 (offset -13 lines).
Hunk #2 succeeded at 57 (offset -13 lines).
patching file src/mainboard/lenovo/t520/early_init.c
Hunk #1 succeeded at 3 (offset -15 lines).
Hunk #2 succeeded at 56 (offset -15 lines).
patching file src/mainboard/lenovo/t530/early_init.c
Hunk #1 succeeded at 3 (offset -15 lines).
Hunk #2 succeeded at 39 (offset -15 lines).
patching file src/mainboard/lenovo/x220/early_init.c
Hunk #1 FAILED at 18.
Hunk #2 succeeded at 57 (offset -17 lines).
1 out of 2 hunks FAILED -- saving rejects to file src/mainboard/lenovo/x220/early_init.c.rej
can't find file to patch at input line 175
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|diff --git a/src/mainboard/lenovo/x230/early_init.c b/src/mainboard/lenovo/x230/early_init.c
|index 70240a7..54513fe 100644
|--- a/src/mainboard/lenovo/x230/early_init.c
|+++ b/src/mainboard/lenovo/x230/early_init.c
--------------------------
File to patch: EOF
Skip this patch? [y] EOF
Skipping patch.
2 out of 2 hunks ignored
make: *** [Makefile:410: /root/project/build/x86/coreboot-4.13/.canary] Error 1
```

So the patch needs to be reworked to be applicable on top of 4.13.

natterangell, 2022-10-04:
Wow, thank you for that! 😊

yes, I’m aware the patch has issues, but it might be useful for my machine still.

It seems to fail for the x220, and succeed patching early_init.c for the t420. I’m not even sure this issue applies to the x220, it certainly doesn’t for the x230.

I can do a manual test as outlined in your first reply.

tlaurion, 2022-10-04:
@natterangell On coreboot matters, its always the same thing. When things are worked on downstream, it doesn't make things merged upstream. Collaboration should happen upstream here. Unfortunately, Heads is still based on 4.13, so there are patches that will fail against 4.17+ (coreboot 4.18 to be released being master on coreboot as of now).

Whatever your outcome here, adding that being successful upstream (and removing x220 and x230 downstream in a branch of yours) and adding feedback under https://review.coreboot.org/c/coreboot/+/39043 would help everybody. It seems that x230 doesn't have early_init under 4.13 (which could be removed in patch) while resolving issue with x220. You could then report success upstream for your board, which would ping the developers and make this patch merged upstream. That would result into that wwan SMBus patch being merged under 4.18 or sonner then later, and have Heads not need any patch when we do a version bump, sooner then later :)

tlaurion, 2022-10-04:
Massively edited https://github.com/osresearch/heads/issues/1221#issuecomment-1267139461 so people can replicate and test CircleCI builds fast against OP's question : "Patching Coreboot before compiling".

Outside of commenting this ticket, doing the process to actually create a discussable outcome took me literally 30 seconds with everything already being setuped on my side. And for everybody to be able to see the result of a failed applied patch to entice collaboration for people having the same feature goal to be able to work locally on a fix and discuss on possible solutions, experiement and propose a working solution that don't break anything (on a build level). The next step of this is to check from the commit what platform could break with actual proposed changes, and be able to to some regression testing prior of the change to land in master after a pull request is poroposed, reviewed and finally merged after proper regression testing per board owners for changes that could affect them.

natterangell, 2022-10-04:
Thanks again! I’ll be sure to report back upstream once I’ve tested.

tlaurion, 2022-10-04:
@natterangell cloning/branching Heads deservers a new wiki entree/bonifying under https://osresearch.net/Development

Command line is always better for that.... Agreed there is a learning curve associated to that though.

This could be used as a draft to be added there.

1. Create a fork under Github at https://github.com/osresearch/heads/fork so that changes can be localized and branches submitted on personal fork 
2. Have github know your ssh public key so you can upload back your changes to github in your fork https://medium.datadriveninvestor.com/how-to-clone-a-github-repository-using-ssh-for-windows-linux-macos-78ad9a3959e?gi=a48768b328cd
3. Have GitHub know your pulic key so that you can sign your commits (this is important so that commits can be trusted and have accountability) https://www.freecodecamp.org/news/git-ssh-how-to/
4. Clone Heads (osresearch/master : will be origin) : `git clone https://github.com/osresearch/heads`
5. Make sure latest version of osresearch/heads `git fetch origin`
6. Add your fork's remote to be useable with ssh: git remote add GITHUB_USERNAME git@github.com:GITHUB_USERNAME/heads.git
7. git fetch GITHUB_USERNAME 
8. I always make sure I'm at the right latest commit against master prior of changing branch : git checkout GITHUB_NAME/master
9. git reset --hard
10. Change to a branch you want to track your changes: git checkout -b NAME_OF_CHANGESET
11. Do limited changes to accomplish a goal (a commit should contain changes linked to accomplish one single thing at a time, and a branch should contain the smallest number of commits needed to accomplish that goal. Think iterative, your branch might be asked to be reworked to seperate your changes across multiple pull requests (different goals) if too many commits are present. A good pull request (your branch pushed against master) should contain maximum 4 commits unless really justifiable.
12. git add file1 file2 file2
13. git commit / git commit -m "goal: adding function xyz"
14. continue doing your changes
15. Add your changed and do commits as above
16. When done, git push GITHUB_USERNAME

On step 16 above, when it is asked of you to modify your pull request, you might have to dig a little bit more the internals of git.
- git commit -amend will permit you to modify your last commit
- Interactive rebasing will permit you to go back in your commit history to squash changes, amend commits etc: https://thoughtbot.com/blog/git-interactive-rebase-squash-amend-rewriting-history

Otherwise, there is plenty of GUI frontends to git out there that you might want to investigate if command line git is giving you nausea. Unfortunetaly, I do not use them so I cannot really recommend any of them. I would go against using the web based VisualStudio frontend proposed by github, since even though it might seem attractive since proposed by Github, amending/rebasing and pushing changes results in a lot of back and forth and many pull requests being opened/closed in the past. Doing changes locally and pusshing them is definitely more interesting, both for your learning and collaborating to other projects and for maintainership of this perticular project.

CircleCI:
1. Open account on CircleCI
2. Subscribe on your github project.
3. That's it. When you will push commits to your fork, CircleCI will build automatically and will inform your branch of success/failed commits. that should probably be another entree.

Will edit this post later on following feedback, and will use that entree here to open another issue in wiki documentation.

tlaurion, 2023-01-27:
@natterangell can this be closed?

natterangell, 2023-01-27:
Yes! Thanks!

tlaurion, 2023-01-27:
@natterangell maybe you want to rename the issue?

Or give some more input on if the patch was merged upstream? In which version, etc?

natterangell, 2023-01-27:
I haven't gotten around to suggesting a patch upstream yet, I'm a little short on time these days. I hope to get to it before the end of February.

tlaurion, 2023-10-22:
Documentation issue opened on heads-wiki. Closing here, feel free to reopen if again relevant.

closed 2023-10-22 by tlaurion