linuxboot/heads #1223: Success heads build, but generate 3 roms with name "dirty"

githubuseravailable, 2022-10-14:
Board: x230-maximized
Build rom by using heads master branch.
Using script in /heads/blobs/xx30/extract.sh, to generate me.bin, gbe.bin, ifd.bin, from a combined dump rom. 
The dump rom was extracted from a backup rom taken from x230 device.
Then, sudo make BOARD=x230-maximized
The build is success, but it generate 3 roms with name:
- romName-dirty.rom
- romName-dirty-bottom.rom
- romName-dirty-top.rom

does the word "dirty" in the built roms, indicate build failure ? 
since, it doesn't generate "dirty" roms, while using the original bins from heads

tlaurion, 2022-10-14:
Dirty here is from git perspective.

'git status' will report on the file changed, which should be gbe.bin, since it was extracted from your backup

githubuseravailable, 2022-10-15:
i see, so the build was success,  
okay, thanks

closed 2022-10-15 by githubuseravailable