linuxboot/heads #1224: Rebuild w530-dgpu-K1000m-hotp-maximized rom in CircleCi

jnscmns, 2022-10-17:
Could you please rebuild w530-dgpu-K1000m-hotp-maximized in CircleCi. There are no artifacts (more than 30 days passed).
Thanks in advance

tlaurion, 2022-10-17:
Happening now under https://app.circleci.com/pipelines/github/osresearch/heads/446/workflows/2ccdf262-3d1a-486d-b173-a39229ca9000 (or through latest commit yellow light that should turn into green after everything is built).

This is a complete build from scratch, no from cache. So this should be done within the next 2 hours. 
Note that w530-dgpu-K1000m-hotp-maximized will be built prior the whole pipeline being successful, so maybe in an hour from now.

closed 2022-10-17 by tlaurion

tlaurion, 2022-10-17:
Make board section of latest board build https://app.circleci.com/pipelines/github/osresearch/heads/446/workflows/2ccdf262-3d1a-486d-b173-a39229ca9000/jobs/4754

https://app.circleci.com/pipelines/github/osresearch/heads/446/workflows/2ccdf262-3d1a-486d-b173-a39229ca9000/jobs/4754/parallel-runs/0/steps/0-103 

```
2022-10-17 15:08:05-04:00 INSTALL   build/x86/coreboot-4.13/w530-dgpu-K1000m-hotp-maximized/coreboot.rom => build/x86/w530-dgpu-K1000m-hotp-maximized/heads-w530-dgpu-K1000m-hotp-maximized-v0.2.0-1263-gf2e45c4.rom
if cmp --quiet "/root/project/build/x86/coreboot-4.13/w530-dgpu-K1000m-hotp-maximized/coreboot.rom" "/root/project/build/x86/w530-dgpu-K1000m-hotp-maximized/heads-w530-dgpu-K1000m-hotp-maximized-v0.2.0-1263-gf2e45c4.rom" ; then echo "`date --rfc-3339=seconds` UNCHANGED build/x86/coreboot-4.13/w530-dgpu-K1000m-hotp-maximized/coreboot.rom" ; fi ; cp -a "/root/project/build/x86/coreboot-4.13/w530-dgpu-K1000m-hotp-maximized/coreboot.rom" "/root/project/build/x86/w530-dgpu-K1000m-hotp-maximized/heads-w530-dgpu-K1000m-hotp-maximized-v0.2.0-1263-gf2e45c4.rom" ; 
fb418d892d39bdac0458464cea1a7735454794b7c5aa5afeadec1e6d3742a5b4  build/x86/w530-dgpu-K1000m-hotp-maximized/heads-w530-dgpu-K1000m-hotp-maximized-v0.2.0-1263-gf2e45c4.rom
2022-10-17 15:08:05-04:00 DD 8MB build/x86/w530-dgpu-K1000m-hotp-maximized/heads-w530-dgpu-K1000m-hotp-maximized-v0.2.0-1263-gf2e45c4-bottom.rom
dd of=/root/project/build/x86/w530-dgpu-K1000m-hotp-maximized/heads-w530-dgpu-K1000m-hotp-maximized-v0.2.0-1263-gf2e45c4-bottom.rom if=/root/project/build/x86/w530-dgpu-K1000m-hotp-maximized/heads-w530-dgpu-K1000m-hotp-maximized-v0.2.0-1263-gf2e45c4.rom bs=65536 count=128 skip=0 status=none
71d3ccb49d09afab4a0f43ce8c97d902ebf251b4e21b340cdfab02e652f13d7d  /root/project/build/x86/w530-dgpu-K1000m-hotp-maximized/heads-w530-dgpu-K1000m-hotp-maximized-v0.2.0-1263-gf2e45c4-bottom.rom
2022-10-17 15:08:05-04:00 DD 4MB build/x86/w530-dgpu-K1000m-hotp-maximized/heads-w530-dgpu-K1000m-hotp-maximized-v0.2.0-1263-gf2e45c4-top.rom
dd of=/root/project/build/x86/w530-dgpu-K1000m-hotp-maximized/heads-w530-dgpu-K1000m-hotp-maximized-v0.2.0-1263-gf2e45c4-top.rom if=/root/project/build/x86/w530-dgpu-K1000m-hotp-maximized/heads-w530-dgpu-K1000m-hotp-maximized-v0.2.0-1263-gf2e45c4.rom bs=65536 count=64 skip=128 status=none
64ba01ae0f9af7d84d9671f9bf09d05ed0236111a7544728b52ff0ef9eccea2a  /root/project/build/x86/w530-dgpu-K1000m-hotp-maximized/heads-w530-dgpu-K1000m-hotp-maximized-v0.2.0-1263-gf2e45c4-top.rom
fb418d892d39bdac0458464cea1a7735454794b7c5aa5afeadec1e6d3742a5b4  /root/project/build/x86/w530-dgpu-K1000m-hotp-maximized/heads-w530-dgpu-K1000m-hotp-maximized-v0.2.0-1263-gf2e45c4.rom
```

So reproducible builds are not a thing even on same environment anymore. diffoscope and more work needs to be done to expect a reproducible environement to produce a reproducible build. Seems like time or something else also leaks in the builds now.