linuxboot/heads #1229: Question: Getting rid of unneeded kernel options on Librem 14

Unb0rn, 2022-10-21:
**Is your feature request related to a problem? Please describe.**
I'm trying to add BTRFS (will affect #1202) and exFAT filesystems support with the corresponding tools which means we'll need some space on flash to contain those. While digging the partial config (and comparing it to the full one after loading it) I found several options I'm not sure we want to have.

**Describe the solution you'd like**
So, those options found fall in several categories:

Debug/Verbosity:
```
CONFIG_DYNAMIC_DEBUG=y
CONFIG_DEBUG_INFO=y
CONFIG_DEBUG_INFO_DWARF4=y
CONFIG_GDB_SCRIPTS=y
CONFIG_DEBUG_FS=y
CONFIG_STACKTRACE=y
CONFIG_KALLSYMS=y
CONFIG_UNUSED_SYMBOLS=y
CONFIG_TRIM_UNUSED_KSYMS=n - probably should allow this one
CONFIG_PNP_DEBUG_MESSAGES=y
CONFIG_ATA_VERBOSE_ERROR=y
CONFIG_DEBUG_MISC=y
```
Do we really need any of those in non-debug production setup?

Ancient/obsolete syscalls/features:
```
CONFIG_X86_MPPARSE=y - do we have any target without acpi?
CONFIG_X86_VSYSCALL_EMULATION=y
CONFIG_X86_IOPL_IOPERM=y
CONFIG_LEGACY_VSYSCALL_NONE=n - probably should allow this one
CONFIG_ZONE_DMA=y - are there any devices on any target using this?
CONFIG_ISA_DMA_API=y - isn't it ancient?
```
Do we have anything using these?

Unused features:
```
CONFIG_CPU_ISOLATION=y
CONFIG_X86_5LEVEL=y - isn't it only for new systems with hue memory?
CONFIG_PCI_IOV=y - are  there any targets with pci vfs needed during boot?
CONFIG_RSEQ=y
CONFIG_NR_CPUS=64 - probably should limit it to like 16
CONFIG_RD_ZSTD=y - we don't use zstd initramfs anyway
```

Support for systems it will never be run on:
```
CONFIG_CPU_SUP_HYGON
CONFIG_CPU_SUP_ZHAOXIN
XZ_DEC_POWERPC
XZ_DEC_IA64
XZ_DEC_ARMTHUMB
XZ_DEC_SPARC
```

I'm also not sure about these three:
```
CONFIG_PRINTK=y - will make the kernel mute, but do we care?
CONFIG_BUG=y
CONFIG_MULTIUSER=y - the question is are we using multiuser system or everything runs as root now?
```

**Describe alternatives you've considered**
Applied -Os patch from @tlaurion mentioned in #590 

**Additional context**
Would be great to have any input about these parameters. If there is a chance any of them will make the system unusable it would be nice to know...
Also, maybe it's a good idea to enable discussions in project for questions like this? Issues don't really feel appropriate for it)

tlaurion, 2022-10-21:
@Unb0rn really interesting points @JonathonHall-Purism might want to look at.

We had brief discussion in regard of my attempts in reducing kernel footprint and have some kind of guidelines for minimal requirements.

Some joint efforts would be nice. I did https://github.com/osresearch/heads/pull/1184 on that matter (xx20/xx30 boards related) but there is a lot of options that are unneeded. Outside of drivers that are specific for specific boards, the same options should be deactivated/activated.

Unb0rn, 2022-10-22:
Thank you for your #1184 work, I think it helped a lot
Made several tests, it looks like both BTRFS and exFAT are able to fit the image now (without the corresponding tools yet), I think turning off KSYMS saves a lot of space
Still have several options to try, but I'm not sure if its's a good idea to turn them off:
```
CONFIG_X86_VSYSCALL_EMULATION
CONFIG_X86_IOPL_IOPERM
CONFIG_LEGACY_VSYSCALL_NONE
CONFIG_ZONE_DMA
CONFIG_ISA_DMA_API

CONFIG_RSEQ

CONFIG_PRINTK=y
CONFIG_BUG=y
CONFIG_MULTIUSER=y
```
It would be nice to know any opinions on them. I'm unsure if we still have some devices using CONFIG_ZONE_DMA and CONFIG_ISA_DMA_API as well as if it's okay to leave the kernel mute.

I also discovered two more options:
```
CONFIG_X86_REROUTE_FOR_BROKEN_BOOT_IRQS -do we need it?
CONFIG_SYN_COOKIES=y - feels weird in almost network-disabled kernel. Some strange dependency?
```
For now, I attached both full and partial configs. Ready to create a PR if @JonathonHall-Purism thinks its okay to use this config

[linux-librem_common.zip](https://github.com/osresearch/heads/files/9844159/linux-librem_common.zip)

tlaurion, 2022-10-22:
@Unb0rn i would suggest creating a pr which will create board Roma automatically to ne tester ans will ease kernel config review

On a side note, I would advise moving away from savedefconfig format from past discussions, to ease kernel version bumps in the future as well.

So please do that PR giving full .config replacing librem share configuration across all boards.

 It will permit direct testing and collaboration on provided config file, and collateral impacts on other boards Linux configurations.

Unb0rn, 2022-10-23:
Created the PR. Left the old defconfig until the time we're ready to merge - I can't be 100% sure that it works fine on other Librem machines...

tlaurion, 2022-10-24:
@Unb0rn replying here instead of in PR since this is the discussion that should lead to a final clean PR.

I used your branch to produce savedefconfig so that boards can be built automatically for testing/comments. Waiting for you to pick up the changes so they can be reviewed/tested by Purism which own the boards under https://github.com/tlaurion/heads/tree/Unb0rn_savedefconfig_changes which, since I follow my Heads repo as a CircleCI project under CircleCI, builds roms for each of my branches/commits into my CircleCI instance instead of osresearch's. Note that since you placed the kenrle config outside of what is currently used inside of the board configurations, the CircleCI builds under your PR have not built your kernel config and are building master's shared linux librem config. In my branch, the linux config has overwritten master and is building roms for all librem boards under https://app.circleci.com/pipelines/github/tlaurion/heads?branch=Unb0rn_savedefconfig_changes 

Will also follow discussion and apply changes in cleanup/kernel bersion bump for other boards.  I would also advise into tagging JonathonHall-Purism (adding @ before the name) when you want review from Purism. 

----

On your questions:

A really nice online resource to understand kernel options and dependencies for a kernel main version is https://www.kernelconfig.io

Analysis below is theoretical: nothing better then having regression testing to see impacts of changes (hypothesis/verification):
> CONFIG_X86_IOPL_IOPERM - legacy. Is there something in heads still using it?

I do not think its used. Would have to do regression tests with flashrom operations 
> CONFIG_ZONE_DMA - are there any devices on librem machines still using it?

I do not think its useful either?

> CONFIG_ISA_DMA_API - isn't it ancient?

ISA?!

> CONFIG_RSEQ

> CONFIG_X86_REROUTE_FOR_BROKEN_BOOT_IRQS - do we need it? I think its only needed for some pretty old specific HW

Same

> CONFIG_SYN_COOKIES=y - the kernel is almost network-less. Is it some kind of dependency?

Well, networking backend cannot be completely removed (https://github.com/osresearch/heads/issues/564#issuecomment-507165095). I am not aware that Librem have Ethernet device, but might be wrong. Idea there is that dropbear, busybox (wget and other network components) would require networking, and if such networking is enabled, then networking should at least include basic security config as well. I think it should be safe to keep it if networking is enabled in kernel.

> CONFIG_PRINTK=y - will make the kernel mute, but do we care?

We care. Debugging output (extreme) vs no log (other extreme) are a total continuum. We want to know if hardware is misbehaving and for that dmesg is a friend we cannot lose. From docs, this is important to differenciate source of output in dmesg in multithread environements. This happens under Heads when generating gpg keys for example. Not sure we would want to loose possible tracing of users debugging traces.

> CONFIG_BUG=y

Not sure if needed. I once had to create a x230 debugging config, needed to enable this along many other debugging options for dynamic debugging but that is corner case which might require documentation section instead of bloating kernel.

> CONFIG_MULTIUSER=y - the question is are we using multiuser system or everything runs as root now?

Definitely not needed.

tlaurion, 2022-10-24:
@Unb0rn I added two linux target helpers under https://github.com/tlaurion/heads/commit/dbe3fd3d9ee5604f7322b8cfd14b3dcc8ae8d300

I think it will be helpful in our quests of comparing different boards tweaked configurations between each other and quicken the development cycle in the future.

In my branch, doing 
```
user@heads-tests:~/heads$ make BOARD=librem_15v4 linux.generateoldconfig 
mkdir -p "/home/user/heads/build/x86/linux-5.10.5/linux-librem_common" \
&& cp "/home/user/heads/config/linux-librem_common.config" "/home/user/heads/build/x86/linux-5.10.5/linux-librem_common/.config" \
&& make -C "/home/user/heads/build/x86/linux-5.10.5" \
	O="/home/user/heads/build/x86/linux-5.10.5/linux-librem_common" \
	olddefconfig \
&& echo "" \
&& echo "You can now edit /home/user/heads/build/x86/linux-5.10.5/linux-librem_common/.config" \
&& echo "Either:" \
&& echo " Manually through text editor" \
&& echo " Through make BOARD=XYZ linux.menuconfig" \
&& echo "" \
&& echo "To save chances in git tree for review, type:" \
&& echo "make BOARD=XYZ linux.saveconfig" 
make[1]: Entering directory '/home/user/heads/build/x86/linux-5.10.5'
make[2]: Entering directory '/home/user/heads/build/x86/linux-5.10.5/linux-librem_common'
  GEN     Makefile
#
# configuration written to .config
#
make[2]: Leaving directory '/home/user/heads/build/x86/linux-5.10.5/linux-librem_common'
make[1]: Leaving directory '/home/user/heads/build/x86/linux-5.10.5'

You can now edit /home/user/heads/build/x86/linux-5.10.5/linux-librem_common/.config
Either:
 Manually through text editor
 Through make BOARD=XYZ linux.menuconfig

To save chances in git tree for review, type:
make BOARD=XYZ linux.saveconfig
```

Creates /home/user/heads/build/x86/linux-5.10.5/linux-librem_common/.config from defconfig as explanations outputted in developer's terminal.

Digging directly under /home/user/heads/build/x86/linux-5.10.5/linux-librem_common/.config shows us a lot of other things that could be tweaked, as said under #1184 including (strict example) AMD CPU options:
```
user@heads-tests:~/heads$ grep AMD /home/user/heads/build/x86/linux-5.10.5/linux-librem_common/.config | grep -v "^#"
CONFIG_CPU_SUP_AMD=y
CONFIG_AMD_NB=y
CONFIG_HW_RANDOM_AMD=m
```
Editing manually the oldefconfig file permits whole sections that can either be deactivated (OPTION=n) and passed to make BOARD=XYZ linux.menuconfig to save back into /home/user/heads/build/x86/linux-5.10.5/linux-librem_common/.config for manual verification until satisfaction. And then make BOARD=librem_15v4 linux.saveconfig can be used to save the file under savedefconfig format under git tree for verification.

A quick play resulted into the following from my branch:
```
diff --git a/config/linux-librem_common.config b/config/linux-librem_common.config
index 7a69da36..8ff81d1a 100644
--- a/config/linux-librem_common.config
+++ b/config/linux-librem_common.config
@@ -16,6 +16,7 @@ CONFIG_INITRAMFS_SOURCE="@BLOB_DIR@/dev.cpio"
 # CONFIG_RD_LZ4 is not set
 # CONFIG_RD_ZSTD is not set
 CONFIG_CC_OPTIMIZE_FOR_SIZE=y
+# CONFIG_MULTIUSER is not set
 # CONFIG_SGETMASK_SYSCALL is not set
 # CONFIG_SYSFS_SYSCALL is not set
 # CONFIG_FHANDLE is not set
@@ -35,12 +36,12 @@ CONFIG_SMP=y
 # CONFIG_RETPOLINE is not set
 # CONFIG_X86_EXTENDED_PLATFORM is not set
 CONFIG_PROCESSOR_SELECT=y
+# CONFIG_CPU_SUP_AMD is not set
 # CONFIG_CPU_SUP_HYGON is not set
 # CONFIG_CPU_SUP_CENTAUR is not set
[... 188 lines cut ...]
+# CONFIG_XZ_DEC_POWERPC is not set
+# CONFIG_XZ_DEC_IA64 is not set
+# CONFIG_XZ_DEC_ARMTHUMB is not set
+# CONFIG_XZ_DEC_SPARC is not set
 CONFIG_XZ_DEC_TEST=m
 CONFIG_IRQ_POLL=y
 CONFIG_PRINTK_TIME=y
 CONFIG_BOOT_PRINTK_DELAY=y
-CONFIG_DYNAMIC_DEBUG=y
 # CONFIG_DEBUG_BUGVERBOSE is not set
-CONFIG_DEBUG_INFO=y
-CONFIG_DEBUG_INFO_DWARF4=y
-CONFIG_GDB_SCRIPTS=y
 # CONFIG_ENABLE_MUST_CHECK is not set
 CONFIG_FRAME_WARN=1024
 CONFIG_MAGIC_SYSRQ=y
-CONFIG_DEBUG_FS=y
 CONFIG_HARDLOCKUP_DETECTOR=y
 CONFIG_WQ_WATCHDOG=y
 # CONFIG_SCHED_DEBUG is not set
-CONFIG_STACKTRACE=y
 # CONFIG_RCU_TRACE is not set
 # CONFIG_FTRACE is not set
 # CONFIG_STRICT_DEVMEM is not set
```

JonathonHall-Purism, 2022-10-25:
Hi @Unb0rn, thank you for all the work on this so far!

I generally agree that most of the suggested config changes make sense.  I ran a few builds to check some of them to see how much space is actually saved.  I can also help test other Librem devices when we have a CI build (I'll comment on the PR too regarding that).

Overall it looks like we can save roughly 500KB which is pretty significant.

I have notes about a few configs that we might want to keep or that I'm unsure of:
`CONFIG_ISA_DMA_API` - I'm not familiar with this config specifically, but LPC still exists on modern computers and is designed to look like ISA to software.  TPMs, SuperIOs, and ECs can often be connected by LPC - the Librem Server's TPM is on LPC IIRC, so we may need to keep this.  We can try it and see.
`CONFIG_ZONE_DMA` - I don't know of any devices requiring this offhand, but for what it's worth, this and ISA_DMA_API together are only about 2KB.
`CONFIG_NR_CPUS` - L1UM-3X16C has 32 logical CPUs, so this must be at least 32.  I'd guess this probably only increases BSS though, not the size of the image file, in which case maybe just leave it at the default.
`CONFIG_PRINTK`, `CONFIG_BUG` - I think we should keep both of these.  There will always be unexpected issues in the field to support, and it's important that users are able to get some amount of information about a problem.  It sounds like the conditions caught by BUG are silently ignored if CONFIG_BUG is disabled, which seems risky to me.
`CONFIG_SYN_COOKIES` - This specifically seems removable.  For what it's worth regarding networking in general, current Librem devices do have Ethernet (L14, L1UM, Mini).
`CONFIG_MULTIUSER` - I'm totally fine with removing this as long as it works, i.e. there are no lurking dependencies on the syscalls this removes.  We can test it.

JonathonHall-Purism, 2022-11-03:
That's great @Unb0rn!  I have this on my to-do list to check some other devices.  I have to cover a handful of other tasks first but I have not forgotten about this, thank you for all the work here!

PR #1232 (2022-12-01): Librem image shrink
This should resolve #1229
Tried not to break anything for any Librem machine, however I am able to test it only on Librem 14

Still need some input on some options:
```
CONFIG_X86_IOPL_IOPERM - legacy. Is there something in heads still using it?
CONFIG_ZONE_DMA - are there any devices on librem machines still using it?
CONFIG_ISA_DMA_API - isn't it ancient?
CONFIG_RSEQ
CONFIG_X86_REROUTE_FOR_BROKEN_BOOT_IRQS - do we need it? I think its only needed for some pretty old specific HW
CONFIG_SYN_COOKIES=y - the kernel is almost network-less. Is it some kind of dependency?
CONFIG [... cut]
closed 2022-12-01 by tlaurion