linuxboot/heads #1239: USB boot to PureOS installer shows boot options that don't work

JonathonHall-Purism, 2022-11-09:
When booting to a PureOS installer on USB, Heads shows some options that always fail when selected.

Two of the options have `($root)` in them (a variable).  We could safely ignore these (they're the EFI boot options, and the BIOS boot options are fine), but some code in `kexec-parse-boot` makes me think it is actually trying to expand variables, so I can't just ignore any boot options containing variables:

```
		echo $(eval "echo \"$entry\"")
```

@tlaurion Is there any reason to keep any boot options with variables in them?  I checked a handful of ISOs and have not found any (Debian live, Fedora live, Qubes installer, although there are tons more...).

tlaurion, 2022-11-09:
@JonathonHall-Purism no! Propose anything!

tlaurion, 2022-11-09:
Related https://github.com/osresearch/heads/issues/1001#issuecomment-1159640544

JonathonHall-Purism, 2022-11-09:
Cool, I will get a PR up.  It looks like it also picks up a memtest image that doesn't work either, hmm, looking into some way to identify that

tlaurion, 2022-11-09:
Memtest can be booted from latest beta image from memtest, look at separate the solved issue for instructions

PR #1240 (2022-11-11): Ignore grub device specs in boot entry paths
Fixes #1239, #1001

Ignore device specs in GRUB boot entry paths, assume they point to /boot.  Check that boot kernel and initrd are valid files; ignore options that would not be able to boot.

Tested booting the following ISOs (directly written to USB disk): Debian (live KDE, netinst still has graphics init issues), Fedora KDE, Qubes, PureOS.

Tested booting Qubes ISO from ext4-formatted USB disk (loopback ISO boot).

Tested booting installed PureOS and Qubes.
closed 2022-11-11 by tlaurion