linuxboot/heads #1248: additional files in `/boot/` are ignored

3hhh, 2022-11-24:
**Describe the bug**

Additional files in `/boot/` are ignored by `heads`.

**To Reproduce**
Steps to reproduce the behavior:
1. `sudo touch /boot/testfile`
2. `sudo reboot`

**Expected behavior**

`heads` complains about a `/boot/` integrity difference (additional file).

**Actual behaviour**

OS boots / `heads` doesn't complain.

**Side Note**

I haven't (yet) tested how it behaves when I remove files.

tlaurion, 2022-11-24:
Duplicate of #1108 

Please reopen if you disagree.

-----

As far as Heads is concerned for preboot security (making sure that what is going to be booted is in the expected state) the goal is to make sure that nothing trusted would refer to those added files. For example, a modified grub entrey pointing to a new kernel/initrd/Xen.

tlaurion, 2022-11-24:
@3hhh reopening to clarify your expectations and to document this a little bit more.

You would expect Heads to
- Prevent default boot if new file present even though nothing detached signed changed?

Actual behavior:
- existing but modified files are covered
- deleted files are covered

#1108 would expend what is showed when files are changed, to also cover presence of new files. The problem with Heads is that there is no way, from within Heads itself to troubleshoot the origin of this. The OS could inform the user, but outside of that, I'm not sure of the best behavior to implement.

To me, I expect Heads to tell me what *changed* since content was signed, where what is important is what would impact the behavior of booted OS.

The habit here currently requires the user to check Qubes dom0 updates, where I posted on Qubes OS some validation tricks to make sure that what is under /boot is desired.

We can expect in the longer terms that fwupd will eventually drop firmware images under /boot for Heads to pick up. That will imply Heads validating detached signatures that will be added in the firmware, but is related to files being added under /boot

Another, but more complicated approach, would be to have helpers scripts under dom0, maybe pushed as contribution packages to be installed by Heads users, to have dom0 updater report on the changes, report on files not having been deployed by Qubes under /boot, and have that helper script prompt a report prior of gently reminding the user that those modified/deleted/added files will be asked to be signed on next default boot operation which will ask to set up a new boot default if grub changed, and or pushing the user for him to generate a new digest of /boot content and sign it.

Otherwise, the best approach would be for Qubes to sign kernel and Xen binaries, or detach sign them and have Heads verify those. And once more, have Qubes provide a static and bigger initramfs containing all the drivers needed, otherwise initrd is different for all Qubes users.

-----

But to go back to this issue.

- booting the default boot could open a warning for found new files ( otherwise if files were changed/modified/deleted, default boot behavior is to show those changes but cannotnreport on new files)
  - it could then permit to boot as normal, since generation of new digest + detach signing is not required now, asking for user to investigate those new additional files
  - it could give option to generate new hashes + detach sign them.

Thoughts?

3hhh, 2022-11-24:
On 11/24/22 19:17, tlaurion wrote:
> You would expect Heads to
> - Prevent default boot if new file present even though nothing detached signed changed?

Yes, that's exactly what I would expect.

Additional files can be an equal security risk as missing or changed files, so they should be treated the same way.

For example I noticed this heads behaviour while I was considering to extend Qubes OS to load some files from `/boot` under certain circumstances. Assuming a user had not configured that file, an attacker could create it to produce an unwanted insecure configuration and then stage other attacks.

With your `fwupd` example, one could e.g. install some insecure firmware.

> The habit here currently requires the user to check Qubes dom0 updates, where I posted on Qubes OS some validation tricks to make sure that what is under /boot is desired.

If one's really scared about "did something change since the last time I booted", I'd recommend the following Opsec practice: Shut down Qubes, boot it, update, then reboot - all while being present in person.
  
> Another, but more complicated approach, would be to have helpers scripts under dom0, maybe pushed as contribution packages to be installed by Heads users, to have dom0 updater report on the changes, [...]

I prefer the Opsec practice. It's more simple and doesn't rely on additional code. ;-)
  
> Otherwise, the best approach would be for Qubes to sign kernel and Xen binaries, or detach sign them and have Heads verify those. And once more, have Qubes provide a static and bigger initramfs containing all the drivers needed, otherwise initrd is different for all Qubes users.

Qubes initrd may already contain user-specific configuration (e.g. usbguard configuration), i.e. there'll likely always be custom initramfs for some users. Alternatively Qubes OS would have to put more configuration into `/boot`, which would make this issue even more relevant.

> -----
>
> But to go back to this issue.
>
> - booting the default boot could open a warning for found new files ( otherwise if files were changed/modified/deleted, default boot behavior is to show those [...]

Please prevent boot unless there's a resign or the insecure boot option is chosen.

tlaurion, 2022-12-01:
> For example I noticed this heads behaviour while I was considering to extend Qubes OS to load some files from `/boot` under certain circumstances. Assuming a user had not configured that file, an attacker could create it to produce an unwanted insecure configuration and then stage other attacks.

Heads is pre-boot security only. Its goal is to make sure that what the user accepted, through basic opsec (boot Qubes and if dom0 has updates, apply them and sign changes affecting /boot asap).  The reason of my disagreement is that if nothing *refers* to the new files that were added under /boot,  outside of files that changed, then the new files in question that were added are not referred by any files that will be considered when booting. 

Let's go back at current protections as explained under https://forum.qubes-os.org/t/verifying-installation
1- High level detached signed digest (authenticity+quick integrity check) of all saved digests, signed by the user *when a change is detected on /boot files* (deletion/modification, but not addition of files) https://forum.qubes-os.org/t/verifying-installation/11739/4
`sha256sum $(find /boot/kexec*.txt) | gpg --verify /boot/kexec.sig -`
This takes shecksums of all /boot/kexec*.txt files and verify that they match integrity/authenticity of detached signature under /boot/kexec.sig
Those verify that, with a default boot configured and a TPM disk unlock key setuped:
`/boot/kexec_default.1.txt /boot/kexec_default_hashes.txt /boot/kexec_hashes.txt /boot/kexec_key_devices.txt /boot/kexec_lukshdr_hash.txt /boot/kexec_rollback.txt`
Are in expected state.

Let's detail
- `/boot/kexec_default.1.txt` contains the user accepted boot options to be passed when selecting configured boot option. This included any option that would be passed to either Xen or the kernel to boot into the system 
- `/boot/kexec_default_hashes.txt` contains all /boot accepted hashes, and their checksums are verified prior of booting.
One could replicate this from within booted OS through `cd /boot && sudo sha256sum -c /boot/kexec_hashes.txt` 
- `/boot/kexec_key_devices.txt`  and `/boot/kexec_lukshdr_hash.txt` are related to currently managed LUKS containers managed through Heads, where default boot will refuse to boot default boot option if the LUKS header changed
- `/boot/kexec_rollback.txt` is a file containing TPM counter value, which is an increment-only that can be used to prevent roll-back attacks on the signed kernel and initramfs configurations. (needs to be enforced)

> stage other attacks

In current Heads state, Heads not warning about new files under /boot is not helping the OS to prevent OS chained attacks. 

This post https://forum.qubes-os.org/t/verifying-installation/11739/11 contains what I would implement.
The idea here would not to "block" current trusted and accepted boot configuration to be booted if new files are present.
But would go in more depth *prior of making the user accept new files to be part of detached, and trusted files under boot when a trusted /boot configuration changed*. Thje difference between the two statements is that currently, when /boot files monitored are not changed, nothing says to the user that something was "Added there"

> With your `fwupd` example, one could e.g. install some insecure firmware.

Including a firmware image that would be dropped in place, where there is no suppot, as of today, to propose to the user to flash such image if it was dropped under /boot anyway, unless the user calls `flash.sh /boot/firmware.rom` manually from the recovery shell. Dropping a firmware image to /boot is actually really convenient, and is the desired way for the future. But mixing concepts here should not be made. that firmware image should be accompanied by a detached signature as well. And that detached signature would be provided to LVFS as part of the firmware update package. And prior of being proposed to the user as trusted, this also means that Heads will need to have those public keys for supported FWUPD distributors, or otherwise, provide a big fat warning if a firmware is being dropped not being verified in any way. This is why this is not part of Heads as of now, the pieces to make this work correctly are not present as of now. Which requires users to verify those firmware images themselves prior of manually flashing them, without any automatic update mechanisms in place.

> Qubes initrd may already contain user-specific configuration (e.g. usbguard configuration), i.e. there'll likely always be custom initramfs for some users. Alternatively Qubes OS would have to put more configuration into `/boot`, which would make this issue even more relevant.

Having a Quebes signed, static initrd by default would permit supported hardware to benefit from detached signed or signed kernel initrd and xen binaries and would permit Heads to verify those way more easily, if they were available. As you pointed out, a user created initrd upon dom0 pushed updates (as of today) could not be verified in any other way then current way from Heads. when dom0 changes /boot, this is because either grub, grub.cfg, Xen or kernel packages were deployed. When that happens, we know grub.cfg will be modified to point to new Xen kernel initrd binaries, and Heds currently and correctly (in my opinion) of the changes related to the changes. Ideally, I would love to see more details in what default boot configuration changed to quickly audit changes in enforced boot parameters. That [... cut ...]
> 7df8ea17b75e0a38d0840769732e3192006611c10e5a516beb732e47560e35ae  ./efi/EFI/qubes/xen.efi
> 7df8ea17b75e0a38d0840769732e3192006611c10e5a516beb732e47560e35ae  ./efi/EFI/qubes/xen-4.14.5.efi
22c22
< 49de5d9fd4286e5ff6d61d21bd7ae5982f765efe71d19d01e2e81e7a5cfc4794  ./xen-4.14.5.gz
---
> 43467546e0a7787eeb93d49523b01c5d4e15deddbbbd1315df95048350139def  ./xen-4.14.5.gz
```

But then what?
Propose to the user after this output is showed to him the following?:
`Unsafe boot + investigate / Generate and sign new digest`

Thoughts welcome!

3hhh, 2022-12-03:
> The idea here would not to "block" current trusted and accepted boot configuration to be booted if new files are present.
> But would go in more depth _prior of making the user accept new files to be [...]

No, please don't do that for the reasons I outlined: You don't really know whether the new files aren't referenced from any `initrd` code. You only knew for sure if you exclusively cared about Qubes OS and would review their code whenever they push an update and 100% trust your review skills.

Otherwise any new file may be used to break previously made boot integrity assumptions by the user.

With the idea you outlined above, the attacker would not need to fear any heads message informing the user about new files until the next time the user needs to sign again. That's why it's a bad idea IMHO.

If you don't want to block the boot process as I proposed, feel free to only warn the user before booting. The user needs to be able to make an informed decision about boot integrity _every_ time s/he boots IMHO.

3hhh, 2022-12-08:
> Content of the script I run personally under dom0 prior and after dom0 updates, which simulates what Heads does for detached signature validatation + output of files that changed:
> 
> ```
> sha256sum $(find /boot/kexec*.txt) | gpg --verify /boot/kexec.sig - && ( if [ -e /usr/bin/sudo ] ; then SUDO=sudo; fi; cd /boot && $SUDO sha256sum -c /boot/kexec_hashes.txt ) | grep -v OK
> ```

Well yes, things would get more complicated than a single line.

Here's what I came up with:
```bash
#!/bin/bash

#known issues:
# - all options require root access rights as files may otherwise remain "hidden" due to access right issues
# - TBD

[ $EUID -eq 0 ] || { >&2 echo "ERROR: Not root." ; exit 1 ; }

cd /boot/

#GPG verification commented out for test reasons
#sha256sum "$(find /boot/kexec*.txt)" | gpg --verify /boot/kexec.sig - && ( if [ -e /usr/bin/sudo ] ; then SUDO=sudo; fi; cd /boot && $SUDO sha256sum -c /boot/kexec_hashes.txt ) | grep -v OK
sha256sum -c /boot/kexec_hashes.txt | grep -v OK

#option 1: sort & uniq
#problem: displays added _and_ removed files
(sed -rn 's/^[^\s]+\s+(.*)$/\1/p' kexec_hashes.txt ; find . -type f ! -path './kexec*') | sort | uniq -u | sed -rn 's/^(.*)$/\1 ADDED OR REMOVED/p'

#option 2: comm or diff with sort
comm -23 <(find . -type f ! -path './kexec*' | sort) <(sed -rn 's/^[^\s]+\s+(.*)$/\1/p' kexec_hashes.txt | sort) | sed -rn 's/^(.*)$/\1 ADDED/p'

#option 3: sh/bash/awk/whatever code
#...

exit 0
```

Such information should at least be displayed as warning to be confirmed by the user IMHO.

3hhh, 2022-12-10:
> Btw none of the above options identify added directories as `heads` currently doesn't save the directory structure at all.

So IMHO `heads` should add something like `sudo find /boot/` as e.g. `/boot/kexec_tree.txt` to its output, sign and verify that file as well and warn the user, if this directory structure is different on boot.

commit accd9f47 (2023-01-17): initrd: track files in /boot in kexec_tree.txt

Fixes #1248
closed 2023-01-17 by tlaurion