linuxboot/heads #1250: Kernel config - possibly unneeded IOMMU support

JonathonHall-Purism, 2022-12-01:
The kernel configurations have IOMMU support enabled.  We're not sure if this is needed for i915 graphics to work, and if so, on which devices.  If disabling IOMMU support saves a significant amount of space, we should check whether it is needed on these boards.

Previously discussed in #1232

tlaurion, 2023-01-27:
@JonathonHall-Purism this setups the IOMMU through the first kernel initialization.

DemiMarie, 2023-03-17:
IOMMU is needed for security reasons.  Linux has an option to enable early DMA protection which should be used.

tlaurion, 2023-09-06:
To be clearer, iommu needs to be setup as early as possible but since boards are now not needing to load i915 related drivers in Heads, the kernel drivers options related to iommu for gfx can now vanish.

As a consequence, the kernel configs can now have the default iommu being on.

This will need review of all kernel configs.

tlaurion, 2023-12-12:
Done under #1522 

Shows linux configs  that do not have `CONFIG_INTEL_IOMMU_DEFAULT_ON=y` enabled:
```
user@heads-tests-deb12:~/heads$ find config/linux* | while read config; do grep -q "CONFIG_INTEL_IOMMU_DEFAULT_ON=y" $config || echo $config; done
config/linux-kgpe-d16_server.config
config/linux-kgpe-d16_server-whiptail.config
config/linux-kgpe-d16_workstation.config
config/linux-linuxboot.config
config/linux-talos-2.config
```

Which is normal because not Intel CPUS. Closing.

closed 2023-12-12 by tlaurion