linuxboot/heads #1254: Coreboot 4.11 based boards fail building (removed from CircleCI)

tlaurion, 2022-12-05:
## Context of the Build

**1. What board are you trying to build?**
kgpe-d16 and librem_l1um 

**2. What repository:branch are you using to build from?**
- [x] Heads:Master
- [ ] Other (please specify)

**3. What version of coreboot are you trying to build**
- [x] 4.11
- [ ] 4.13
- [ ] 4.14
- [ ] 4.15
- [ ] 4.17
- [ ] Other (please specify)

**4. In building the rom where did you get the blobs?**
- [x] No blobs required
- [ ] Provided by the company that installed Heads on the device
- [ ] Extracted from a backup rom taken from this device
- [ ] Extracted from another backup rom taken from another device (please identify the board model)
- [ ] Extracted from the online bios using the automated tools provided in Heads
- [ ] I don't know

**5. If using the automated tools to get the blobs did you run the relevant scripts in the blobs directory**
- [ ] Yes
- [ ] No

**6. What operating system are you using**
debian-11, debian-10 (as of current build with patchset for 4.11 patches to be applied successfully) more work needed:
https://github.com/osresearch/heads/pull/1253

## Please describe the problem

**Describe the bug**
Will edit

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

JonathonHall-Purism, 2022-12-05:
@tlaurion I have a couple of related patches. I hadn't had a chance yet to go back and figure out if these were caused by our other patches or need to be upstreamed.
* One of the coreboot-4.11 patches: https://source.puri.sm/firmware/pureboot/-/commit/701437e08a00fe7c410a851352e918ae6ed3a108
* Building with make 4.2.1: https://source.puri.sm/firmware/pureboot/-/commit/78fb53946fc1de8caeda333f814bc87d265ab30a
    * would guess that there is probably something that should be patched in coreboot-4.11 instead, but I haven't dug that deeply yet

tlaurion, 2022-12-05:
@JonathonHall-Purism thanks for replying
- gawk and make 4.2.1 hacks were removed since irrelevant. They were added at a time where the buildsystems were not providing the right OS tools and we needed to make sure we at least had the base version to build musl-cross-make correctly, and also coreboot 4.11 build toolchain. 
  - agreed that maybe that removal is breaking 4.11 builds though, but the tests were successful to build 4.11 boards, which l1um and kgpe-d16 were still part of CircleCI at that time.
- Also to note that concurently around the same time, CircleCI docker image changed from debian-10 to debian-11. Again, make+gawk were required under debian-10, but are not required under debian-11.
- What I witnessed in the past with CircleCI was the need to kick the builds because of a race condition on car.data if I remember well, which was not happening at each builds.
- Now when attemtping to build kgpe-d16 on local host with debian-11, the error happens at each build. I will check if that applied also to l1um and if so, we will be able to merge forces to troubleshoot this more effectively. 

I have readded l1um as a buld test so we can discuss under PR at #1253 based on logs produced.

tlaurion, 2022-12-05:
Here are interesting logs for comparison.

librem_l1um builds successfully on debian-10:
[librem_l1um_deb10.txt](https://github.com/osresearch/heads/files/10157921/librem_l1um_deb10.txt)

librem_l1um fails to build on debian-11:
[librem_l1um_deb11.txt](https://github.com/osresearch/heads/files/10157922/librem_l1um_deb11.txt)

The diff shows pretty clearly what is happening when its working on debian10 vs failing on debian11:
[diff.txt](https://github.com/osresearch/heads/files/10157945/diff.txt)

Will try to dig into that, if your eyes spots something more quickly then I do @JonathonHall-Purism, let me know.

tlaurion, 2022-12-05:
@JonathonHall-Purism Found the cause: make 4.3 is more capricious on the spec files.

Two patches in the loop for testing:
https://review.coreboot.org/plugins/gitiles/coreboot/+/0f6f70c3942c152c512b1aa51b6f6079a05e003b%5E%21/#F0
https://review.coreboot.org/plugins/gitiles/coreboot/+/1c08a9a9c4986f2b3c47322f041e289121536dc0%5E%21/

tlaurion, 2022-12-08:
Unfortunately, I am stalled under #1253.

Mitigation is to build kgpe-d16 and librem_l1um on debian-10.

tlaurion, 2026-03-25:
been fixed for a while

closed 2026-03-25 by tlaurion