linuxboot/heads #1257: GPG verification errors in OS boot menu silently make the user return to previous menu

3hhh, 2022-12-17:
**To reproduce:**

1. Make sure that GPG verification fails by modifying some file.
2. Reboot & select "Show OS boot menu" & select some boot option.

**Expected behaviour:**

Error is shown.

**Actual behaviour:**

User is returned to the previous menu, no error is shown.

tlaurion, 2022-12-19:
https://github.com/osresearch/heads/blob/315febdf74938691a93d8f2cc657bd55d7d922e6/initrd/bin/gui-init#L535-L541
is faulty

https://github.com/osresearch/heads/blob/315febdf74938691a93d8f2cc657bd55d7d922e6/initrd/bin/gui-init#L535-L541

is ok

tlaurion, 2022-12-19:
@3hhh show boot options shows failed checksum that I manually modified under /boot/kexec_hashes.txt for the purpose of this test, tested from x230-hotp-maximized rom, downloaded from artifacts from https://app.circleci.com/pipelines/github/tlaurion/heads/1323/workflows/09a18f86-de56-49e2-83c8-60e2bcc1d50e

tlaurion, 2022-12-19:
@JonathonHall-Purism please review!

tlaurion, 2022-12-19:
> @3hhh #1259 boot options shows failed checksum that I manually modified under /boot/kexec_hashes.txt for the purpose of a test. 

I then tested modified /boot/kexec_hashes.txt modified file against master's produced rom and obtained the same behavior: show boot options report files having different hashes then under kexec_hashes.txt per https://github.com/osresearch/heads/blob/bf3898a2a1465ae5360c85947c25abf85fc7b443/initrd/bin/gui-init#L67 without verifying kexec.sig detached signed content, which is verified per default boot and not show boot option.

default boot doesn't call kexec-select-boot with -m option:
https://github.com/osresearch/heads/blob/bf3898a2a1465ae5360c85947c25abf85fc7b443/initrd/bin/gui-init#L552

show boot optopns calls kexec-select-boot with -m option:
https://github.com/osresearch/heads/blob/bf3898a2a1465ae5360c85947c25abf85fc7b443/initrd/bin/gui-init#L539

where calling kexec-select-boot with -m option is forcing boot without verifying kexec.sig, just relaying on kexec_hashes.txt:
https://github.com/osresearch/heads/blob/bf3898a2a1465ae5360c85947c25abf85fc7b443/initrd/bin/kexec-select-boot#L27

Again @3hhh:
> 1. Make sure that GPG verification fails by modifying some file in /boot/.
> 2. Reboot & select "Show OS boot menu" & select some boot option.

Can you be more precise into what you modify, what you expect and what you obtain as a result?

3hhh, 2022-12-25:
> Again @3hhh:
> 
> > 1. Make sure that GPG verification fails by modifying some file in /boot/.
> > 2. Reboot & select "Show OS boot menu" & select some boot option.
> 
> Can you be more precise into what you modify, what you expect and what you obtain as a result?

I recall I modified `/boot/.auditing-0` / wrote some different content inside.

I expected some GPG verification failure error message, but was returned to the previous GUi menu without any error at all.