linuxboot/heads #1260: Question Tails OS support boot USB/DVD?

PowerPress, 2022-12-26:
Could I use Heads to secure my system hasn't been tampered and boot off tails USB drive or DVD of tails?

Would it be possible to do this without any hard drives on the PC or I wouldn't have to worry about Tails getting compromised and being able to read information from the drives?

Thank you.

tlaurion, 2023-01-16:
@PowerPress Confused by this ticket, since documentation explains how this works directly for Tails use case.
If no OS is installed, you can use non-hotp maximized boards variants only though, since HOTP requires a disk to be provisioned and an installation of an OS (need /boot to store htop counter).

Documentation is at:
https://osresearch.net/InstallingOS/#generic-os-installation

Since Tails is botha live CD and an installer, the documentation doesn't make any distinction about booting an ISO alongside its iso.asc detached signature file.

If this is the problem for you, I would love you to tell me how you would want the documentation to be different.

This would be a documentation issue, not a code issue, and would need to be opened under https://github.com/osresearch/heads-wiki/issues and not here.

Closing accordingly here.

closed 2023-01-16 by tlaurion

tlaurion, 2023-01-16:
This is the welcome screen you will get if no OS is installed, which should answer your question from a documentation perspective: https://osresearch.net/Configuring-Keys/#no-os-installed