linuxboot/heads #1263: GitHub repository metadata points to non-TLS URL

JeremyRand, 2023-01-01:
The repository metadata for https://github.com/osresearch/heads points to http://osresearch.net/ (without TLS).  Can the metadata be updated to use the HTTPS version?

tlaurion, 2023-01-16:
@JeremyRand I tried to find where on several occasions since you opened the issue, but I can't either modify the about section myself or do not find any place where to do it as a collaborator?

JeremyRand, 2023-01-17:
> @JeremyRand I tried to find where on several occasions since you opened the issue, but I can't either modify the about section myself or do not find any place where to do it as a collaborator?

@tlaurion I believe changing that metadata requires either Admin or Owner privs on the repo; collaborators can't do it.

tlaurion, 2023-01-17:
@osresearch?

closed 2023-11-01 by tlaurion