linuxboot/heads #1279: Boot Error into Boot-able Boot-and-Nuke Software

newbieAtGithub, 2023-01-15:
hi @tlaurion 

Please kindly help with below error,

Boards: x230-maximized + Yubikey
1st time flash, self installed, external flash, IFD unlocked, downloaded from Circle CI, 

Steps to reproduce: 

First, plugin USB device, containing boot-able boot-and-nuke software, [Aban](https://aban.derobert.net/),
Second, turn on Heads laptop, & navigate to boot options > USB boot > Aban, like below screenshot: 

[image]

Third, choose Aban disk wiper

[image]

Fourth, then 1st failure happen here "fail to identify", like below screenshot, 
the system stuck at this screen for around 2-3 minutes. 

[image]

Fifth, then after some time, finally boot failed, like below screenshot,

[image]

Sixth, reboot the laptop, & navigate to another boot option, like below screenshot,

[image]

Seventh, after choosing the boot option, then the process stuck here, "starting new kernel", 
wait for long time, maybe 5-10 minutes, but nothing happen. 

[image]

Expectation: boot successfully to [Aban](https://aban.derobert.net/), and wiping HD. 

Thanks and Regards,

tlaurion, 2023-01-16:
Downloading https://aban.derobert.net/dl/ABAN-amd64.hybrid.iso

@newbieAtGithub 
Pertinent errors should be found under reported screenshot:

building Qemu and injecting key (following instructions to test Heads without hardware on a provisioned debian-11-xcfe OS):
`user@heads-tests:~/heads$ make BOARD=qemu-coreboot-fbwhiptail-tpm1 PUBKEY_ASC=~/QubesIncoming/Insurgo/Insurgo_2023_pub.asc USB_TOKEN=NitrokeyStorage ROOT_DISK_IMG=~/heads/build/x86/qemu-coreboot-fbwhiptail-tpm1-hotp/root.qcow2 inject_gpg`

Using qemu-image tools to take a snapshot prior of trying to use a wiper tool and making sure snapshot exists:
```
user@heads-tests:~/heads$ qemu-img snapshot ~/heads/build/x86/qemu-coreboot-fbwhiptail-tpm1-hotp/root.qcow2 -c working
user@heads-tests:~/heads$ qemu-img snapshot ~/heads/build/x86/qemu-coreboot-fbwhiptail-tpm1-hotp/root.qcow2 -l
Snapshot list:
ID        TAG               VM SIZE                DATE     VM CLOCK     ICOUNT
1         working               0 B 2023-01-16 15:41:50 00:00:00.000          0
```

Launching iso with provisioned disk from qemu doc:
`make BOARD=qemu-coreboot-fbwhiptail-tpm1 PUBKEY_ASC=~/QubesIncoming/Insurgo/Insurgo_2023_pub.asc USB_TOKEN=NitrokeyStorage ROOT_DISK_IMG=~/heads/build/x86/qemu-coreboot-fbwhiptail-tpm1-hotp/root.qcow2 INSTALL_IMG=~/Downloads/ABAN-amd64.hybrid.iso run`

Booting from USB up to ABAN on QEMU board (no hardware)
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

I could not replicate on Qemu.
@newbieAtGithub you are 

> Fifth, then after some time, finally boot failed, like below screenshot,
> 
> [image]

What does `cat /boot.log` provides here? What is the platform you are testing (x230?)

tlaurion, 2023-01-16:
So we know Heads is able to kexec into ABAN from Qemu testing.

Redoing for:
> Boards: x230-maximized + Yubikey

Before inserting USB thumb drive I will overwrite with ABAN iso content:
```
user@heads-tests:~/heads$ lsblk
NAME    MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
xvda    202:0    1    15G  0 disk 
├─xvda1 202:1    1   200M  0 part 
├─xvda2 202:2    1     2M  0 part 
└─xvda3 202:3    1  14.8G  0 part /
xvdb    202:16   1   100G  0 disk /rw
xvdc    202:32   1    10G  0 disk 
├─xvdc1 202:33   1     1G  0 part [SWAP]
└─xvdc3 202:35   1     9G  0 part 
xvdd    202:48   1 467.5M  1 disk 

```
After inserting USB thumb drive (Qubes OS passing device to qube):
```
user@heads-tests:~/heads$ lsblk
NAME    MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda       8:0    1  14.8G  0 disk 
└─sda1    8:1    1     8G  0 part 
xvda    202:0    1    15G  0 disk 
├─xvda1 202:1    1   200M  0 part 
├─xvda2 202:2    1     2M  0 part 
└─xvda3 202:3    1  14.8G  0 part /
xvdb    202:16   1   100G  0 disk /rw
xvdc    202:32   1    10G  0 disk 
├─xvdc1 202:33   1     1G  0 part [SWAP]
└─xvdc3 202:35   1     9G  0 part 
xvdd    202:48   1 467.5M  1 disk
```

Inserted device is sda.

dd'ing ISO to thumb drive (iso confirmed working from Qemu, please verify checksum prior which could be be source of error):
```
user@heads-tests:~/heads$ sudo dd if=~/Downloads/ABAN-amd64.hybrid.iso of=/dev/sda bs=16M  status=progress
279969792 bytes (280 MB, 267 MiB) copied, 10 s, 27.0 MB/s
16+1 records in
16+1 records out
279969792 bytes (280 MB, 267 MiB) copied, 29.5348 s, 9.5 MB/s
```

Then redoing USB boot from above option `ABAN_Disk_wiper_amd64` with SSD drive I don't care about its content:
[image]
[image]
[image]

Works perfectly on x230-maximized from master with downloaded ISO: https://aban.derobert.net/dl/ABAN-amd64.hybrid.iso
```
user@heads-tests:~/heads$ sha256sum ~/Downloads/ABAN-amd64.hybrid.iso
6381be3b8a3e8d1e0add9130e8011fd8cb7262a17bb369e240e9352d4d74bff4  /home/user/Downloads/ABAN-amd64.hybrid.iso
```
 
@newbieAtGithub : Hypothesis is corrupted ISO. Please verify checksums and reopen if you consider this was closed by error.

tlaurion, 2023-01-16:
> Fourth, then 1st failure happen here "fail to identify", like below screenshot, the system stuck at this screen for around 2-3 minutes.
> 
> [image]
> 
> Fifth, then after some time, finally boot failed, like below screenshot,
> 
> [image]

You seem to have issues with your disk.
`I/O error` are generally not a good thing to see.

closed 2023-01-16 by tlaurion

tlaurion, 2023-01-16:
Not related to Heads booting problem but related to that specific drive to wipe issues. Try another drive!

newbieAtGithub, 2023-01-22:
hi @tlaurion 

i know which part was wrong, Heads failed to boot into ABAN before, 
because i used i386 ISO, instead of amd64 ISO, 
assuming that amd64 is only for AMD processor, while i use Intel, 
but i386 ISO is for 32 bit processor, while mine is 64 bit,
finally i tried amd64 ISO, and it works, thanks. 
because i see above, that you also use amd64, while i guess yours also Intel. 

okay, thanks and regards,