linuxboot/heads #1283: Setting a TPM disk encryption key prevents OS discards down to the LUKS and actual disk layer

tlaurion, 2023-01-17:
Revisiting #894 codepaths, testing #1188 with debian-11-xfce (unencrypted /boot and optional LUKS container for OS), while trying to fix TPM disk unlock key to work for something else then Qubes OS under #1280 made me realize that as of today, the cpio that is created as part of unsealing TPM disk encryption key contains a crypttab and the secret that is applied on top of initramfs.

That codepath was created prior of Qubes 4.1 changing the defaults to have its installer apply discards (TRIM) properly from the FS layers (EXT4 in Templates/qubes), down to the LVM pool layer to setting LUKS correctly to have discard applied there as well, so that TRIM operations can be applied correctly per SSD firmware (which then varies per vendor implementation).

Current Heads implementation
------
When a user's decide to setup a boot default (highly recommended), the user is offered to setup a TPM disk unlock key, which will take not only[ firmware measurements](https://osresearch.net/Keys/#tpm-pcrs) (coreboot's reported measured boot under PCR2) until Heads payload is loaded, which then extend measurements of ifs persistently stored components into PCR7 (GPG key material and config.user override kept under coreboot's CBFS storage area), but also measurements of kernel modules at time of sealing in PCR5 and the bootmode in PCR4, while making sure that booting the default boot option won't unseal the secret if the LUKS header hash changed from what was measured at sealing under PCR6. All of the boot options (kernel options, xen options, kernel and xen binaries hashes) are measured (hashed) and signed/authentified by the user after an OS core component upgrade (dom0 boot components upgrades) prior of being discovered and changed.

All is good until now. A user booting its default boot option will only be permitted if the same detached signed hashes are verified against he public key fused inside of the ROM, and if the sealed TPM nvram region can be unsealed with above PCRs+defined TPM disk enryption key passphrase used to seal the disk unlock key "passphrase" (256 characters key) which is unsealed, packed into an additional cpio to be unpacked by the OS after having unpacked its own initrd to construct its initramfs, which is the the pre-os environement containing OS drivers and pre-boot environement to boot into the end OS.

**The actual problem comes with the fact that that additional cpio that Heads constructs, which contains the secret.key released by the TPM after authentication, also contains a crypttab override that changes expected OS' crypttab behavior from initrd, and deployed in OS' paths.** We will over that.

This happens here on master: https://github.com/osresearch/heads/blob/c3b0bd6ffbe816430dd41ef54e649af52ed1ff3b/initrd/bin/kexec-insert-key#L78

What was proposed under #894: https://github.com/osresearch/heads/blob/3b3ce56bafa674c9a2af76e23c6be1a2d01c01aa/initrd/bin/kexec-insert-key#L98

And now, some examples of what is expected/deployed on some OSes.

debian-11-xfce:
[image]

Q4.1:
```
[user@dom0 ~]$ sudo cat /etc/crypttab 
luks-a4ba89e4-65a3-48f9-a045-d128c2a422cc UUID=a4ba89e4-65a3-48f9-a045-d128c2a422cc none discard
```

Conclusion:
Heads currently overwrites the discard/TRIM behavior **to not be applied**:
[image]

Controversial and probably desired implementation
----
Since the OS is responsible to deal with the upper layer discard/trim configuration on filesystem level (EXT4,etc) and pool level (LVM etc), it is thought that having Heads force a LUKS discard override as part of its deployed cpio to be applied on top of initramfs would be the best approach.

So #1280's proposed implementation (to override multiple crypttab deployed files to have debian based OSes `/cryptroot/crypttab` initramfs files overwritten, which neglects /etc/crypttab...):
https://github.com/osresearch/heads/blob/990150703bc6b8da4c5f6790838cf48b4972984b/initrd/bin/kexec-insert-key#L102
Look more like:
` echo "luks-$uuid UUID=$uuid /secret.key luks,discard" | tee "$INITRD_DIR/$crypttab_path" "$INITRD_DIR/cryptroot/crypttab" > /dev/null `
**Where `luks,discard` would force discard/trim operations on selected default boot defined LUKS containers**

Possible current impact of lack of discard operations on LUKS container
----
- Impact performance
- Impact anti-forensic
- No TRIM applied down to drives

Litterature
----
- https://patrick-nagel.net/blog/archives/337 (2011)
- https://lwn.net/Articles/787272/ (2019)
- Qubes OS 4.0 enables LUKS discard automatically: https://github.com/QubesOS/openqa-tests-qubesos/commit/2f49b7502e5b94d46db8499febff40c0b8dd5751
- Qubes 4.0 blkdiscard applied on shutdown only so that TRIM is not locking IO https://github.com/QubesOS/qubes-core-admin/commit/16d069289cb01e73272e82f9f2a1c7f8ebdf0403
- Qubes 4.0: enable discard on all supporting filesystem at installer: https://github.com/QubesOS/qubes-anaconda/commit/2d242cb577b835313b67c14aa0fd3380c00bf78d
- Qubes 4.0: apply discard operation on dom0 filesystems by default: https://github.com/QubesOS/qubes-anaconda/commit/6b814910db8db712d82fc0355c54cb878dc0faa0
- Qubes 4.0: add automatic fstrim to trim/discard thin pool volumes down to LUKS layer: https://github.com/QubesOS/qubes-core-admin-linux/commit/bae443dfceffc43c44fd9b3bf06132b8ac28303d

Latest: Q4.1 implements trimming operations automatically to all layers https://github.com/Qubes-Community/Contents/blob/master/docs/configuration/disk-trim.md

> There are some security implications to permitting TRIM (read for example [this article](https://asalor.blogspot.com/2011/08/trim-dm-crypt-problems.html)), but in most cases not exploitable. Conversely, TRIM can improve security against local forensics when using SSDs, because with TRIM enabled deleting data (usually) results in the actual data being erased quickly, rather than remaining in unallocated space indefinitely. However deletion is not guaranteed, and can fail to happen without warning for a variety of reasons.

tlaurion, 2023-01-17:
Still not perfect since mismatch in naming of volumes.

As current PR:
[image]
Disk unlock works automatically, filesystem checks happens correctly....

But at time of passing control to OS:
[image]
There is double prompt for vda5_crypt passphrase even though volume is unlocked per UUID.... So again, we rely on OS initramfs scripts to do the right thing:
[image]

Which under debian-11-xfce can be ctrl-c'ed to be bypassed:
[image]

But this shows that there is no one-fits-all behavior here, and we might have to unpack initrd to check crypttab and replace `none` for key passphrase to pass `/secret.key` to comply with OS initramfs scripts and final OS cryptab which is used to make sure everything is right.

Put to draft for the moment. 

Next steps:
---

Will test passing expected config in cryptab to validate hypothesis, and if that works, will think on right avenue to provide basic parser to inspect initrd content on time of setting default boot option (which is when we should define crypttab file location and define proper override).

tlaurion, 2023-01-17:
> ## Next steps:
>
> Will test passing expected config in cryptab to validate hypothesis, and if that works, will think on right avenue to provide basic parser to inspect initrd content on time of [...]

For the sake of experimentation:
```
diff --git a/initrd/bin/kexec-insert-key b/initrd/bin/kexec-insert-key
index 95a50832..6dde5298 100755
--- a/initrd/bin/kexec-insert-key
+++ b/initrd/bin/kexec-insert-key
@@ -99,7 +99,7 @@ if [ "$unseal_failed" = "n" ]; then
        for uuid in `cat "$TMP_KEY_DEVICES" | cut -d\  -f2`; do
                # In Debian, the "luks" option at last should not be omitted
                # crypttab files in initramfs are overwritten from the ones passed below
-               echo "luks-$uuid UUID=$uuid /secret.key luks,discard" | tee "$INITRD_DIR/$crypttab_path" "$INITRD_DIR/cryptroot/crypttab" > /dev/null
+               echo "vda5_crypt UUID=$uuid /secret.key luks,discard" | tee "$INITRD_DIR/$crypttab_path" "$INITRD_DIR/cryptroot/crypttab" > /dev/null
        done
        ( cd "$INITRD_DIR" ; find . -type f | cpio -H newc -o ) >> "$SECRET_CPIO"
 fi
```

[image]
[image]

Conclusion:
A parser is needed, otherwise, there is no guarantee we would oblige the OS initramfs scripts. Lucky for us, that should not be so complicated since Heads has the proper tools to do so, including cpio and decompression tools, while it is confirmed here that debian-11-xfce doesn't include two crypttab files, while PurismOS includes both /cryptroot/crypttab and /etc/crypttab being different of each other.

@JonathonHall-Purism : Can you post content of both files here to see differences?

tlaurion, 2023-01-17:
practical experimentations:
---

Qubes:
----
```
mkdir -p /tmp/test
sudo file /boot/initramfs-5.15.76-1.fc32.qubes.x86_64.img 
/boot/initramfs-5.15.76-1.fc32.qubes.x86_64.img: ASCII cpio archive (SVR4 with no CRC)
cpio -idv < initramfs-5.15.76-1.fc32.qubes.x86_64.img
find ./
./
./kernel
./kernel/x86
./kernel/x86/microcode
./kernel/x86/microcode/GenuineIntel.bin
./early_cpio

```
Nothing interesting would be found here for crypttab under initramfs but still an example to be considered.

debian-11-xfce
----
Shows that some handling will need to be done on initrd without extensions. Will continue and edit later on

JonathonHall-Purism, 2023-01-18:
> @JonathonHall-Purism : Can you post content of both files here to see differences?

Sure, this is from a VM with Heads.  I also checked my L14 and it's the same other than the UUID.  They only differ in whitespace and comments.

`etc/crypttab`:
```
# /etc/crypttab: mappings for encrypted partitions.
#
# Each mapped device will be created in /dev/mapper, so your /etc/fstab
# should use the /dev/mapper/<name> paths for encrypted devices.
#
# See crypttab(5) for the supported syntax.
#
# NOTE: Do not list your root (/) partition here, it must be set up
#       beforehand by the initramfs (/etc/mkinitcpio.conf). The same applies
#       to encrypted swap, which should be set up with mkinitcpio-openswap
#       for resume support.
#
# <name>               <device>                         <password> <options>
luks-5598dac8-4ebf-4338-a9ea-01c66dae569c UUID=5598dac8-4ebf-4338-a9ea-01c66dae569c     none luks
```

`cryptroot/crypttab`:
```
luks-5598dac8-4ebf-4338-a9ea-01c66dae569c UUID=5598dac8-4ebf-4338-a9ea-01c66dae569c none luks
```

tlaurion, 2023-01-28:
Fixed by #1284.

Needs firmware upgrade

closed 2023-01-28 by tlaurion