linuxboot/heads #1285: Issue Booting debian (devuan) 11 on Librem 14 with Heads

123ahaha, 2023-01-18:
## Please identify some basic details to help process the report
Hi there ! I recently acquired a librem 14 and I have an issue booting debian.
### A. Provide Hardware Details

**1. What board are you using (see list of boards [here](https://github.com/eganonoa/heads/tree/master/boards))?**
So I have a Librem_14 with Intel Core i7-10710U & Mesa intel UHD Graphics

**2. Does your computer have a dGPU or is it iGPU-only?**
- [x] dGPU
- [ ] iGPU-only

**3. Who installed Heads on this computer?**
- [x] Purism (it was preinstalled, but i had to switch to coreboot to install debian on the nvme)
- [ ] Self-installed (and then I tried heads again with the https://source.puri.sm/firmware/utility/-/blob/master/coreboot_util.sh installer, and i compiled a recent version (on another laptop) of heads and updated it)

**4. What PGP key is being used?**
- [x] Librem Key
- [x] Nitrokey Pro 2
I tried both, they work, i don't think this is the issue

**5. Are you using the PGP key to provide HOTP verification?**
- [x] Yes

### B. Identify how the board was flashed

**1. Is this problem related to updating heads or flashing it for the first time?**
- [x} First-time flash
- [x] Updating heads 
Both, the only issue is to switch back to coreboot

**2. If the problem is related to an update, how did you attempt to apply the update?**
- [x] Using the Heads GUI
I could try the chips flashing ... but heads work very fine, until i have to boot

**3. How was Heads initially flashed**
- [x] Don't know
Probably external at librem factor

### C. Identify the rom related to this bug report

**1. Did you download or build the rom at issue in this bug report?**
- [x} I downloaded it
- [x] I built it
tried both

**2. If you downloaded your rom, where did you get it from?**
- [x] Purism

*Please provide the release number or otherwise identify the rom downloaded*
sorry i don't have it right now, but if really needed, I'll provide it

**3. If you built your rom, which repository:branch did you use?**
- [x] Heads:Master

**4. What version of coreboot did you use in building?**
- [x] 4.8.1 (current default in heads:master)
- [ ] Other (please specify)
Default But heads say coreboot 4.17

**5. In building the rom where did you get the blobs?**
- [x] Provided by the company that installed Heads on the device
- [x] Extracted from the online bios using the automated tools provided in Heads
- [ ] I don't know (i didn't see any option to put i915.bin into heads ...)

## Please describe the problem

**Describe the bug**
A clear and concise description of what the bug is.
The bug is : I can't boot debian with heads (compiled from source, master) or 'pureboot' from current release from purism, i can't either boot the install iso in graphical or non graphical mode to install debian. Each time it stops to "Starting the new kernel" and then freeze.

I can boot PureOS tho and i can boot the live desktop iso from debian

**To Reproduce**
Steps to reproduce the behavior:
1. On a librem 14, flash a debian (xfce live, netinstall, free, non-free) iso with balenaetcher, do the normal setup of the librem/nitrokey (which work)
Note that I have to install debian with coreboot. and then flash heads back.
2. Make default, Boot

**Expected behavior**
I have a t430 and a x230, it should ask me for the encryption password for the NVME disk

**Screenshots**
I put some screenshot of
1. the output when heads stop
[image]
2. the dmesg output after a reboot
[image]
4. the dmesg error
[image]

**Additional context**
I saw that there might be an issue with i915, but i have the firmware-linux-nonfree installed ...

Thanks in advance for any help

123ahaha, 2023-01-18:
I found something while trying thing i read on issues already openned/closed
This one had something very similar to me https://github.com/osresearch/heads/issues/1117 
So i tried to build a head version for the librem 14 but in  > Devices > Display > Framebuffer mode i switched from default Legacy VGA to Linear high res[...]
So I didn't tried that with Legacy VGA, but when the boot process freeze, i tried to just type my cryptsetup to unlock the drive, and then it happend : i arrived on the lock/welcome screen

So it is linked to graphic driver, but what do i have to do ? when i check the i915 pci status I have that : 

>lspci | grep ' VGA ' | cut -d" " -f 1 | xargs -i lspci -v -s {}

00:02.0 VGA compatible controller: Intel Corporation Device 9bca (rev 04) (prog-if 00 [VGA controller])
	DeviceName: VGA compatible controller
	Subsystem: Intel Corporation Device 9bca
	Flags: bus master, fast devsel, latency 0, IRQ 140
	Memory at b0000000 (64-bit, non-prefetchable) [size=16M]
	Memory at a0000000 (64-bit, prefetchable) [size=256M]
	I/O ports at 1000 [size=64]
	Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]
	Capabilities: [40] Vendor Specific Information: Len=0c <?>
	Capabilities: [70] Express Root Complex Integrated Endpoint, MSI 00
	Capabilities: [ac] MSI: Enable+ Count=1/1 Maskable- 64bit-
	Capabilities: [d0] Power Management version 2
	Capabilities: [100] Process Address Space ID (PASID)
	Capabilities: [200] Address Translation Service (ATS)
	Capabilities: [300] Page Request Interface (PRI)
	Kernel driver in use: i915
	Kernel modules: i915
	
So what did i missed ?

tlaurion, 2023-01-18:
@JonathonHall-Purism ?

JonathonHall-Purism, 2023-01-19:
Hi @123ahaha , I checked this out on L14 and on my mini v2 (similar chipset, but I have a UART on that which makes troubleshooting a lot easier).  I have the same behavior for Debian/Devuan; live desktop media work but net installers do not.

It's the video - the net installers do boot, but they can't initialize video since they lack the i915 driver.  Eventually I tried removing the i915 driver from PB to see if it would work if the firmware doesn't switch to graphics mode, and it does.

So, probably the shortest path to get these working is to somehow switch back to text mode before invoking kexec -e.  Currently we can't unload i915 because it's built-in, and it's still not confirmed that unloading the driver would work.

@tlaurion Does the Debian net installer work currently on any boards you support?

JonathonHall-Purism, 2023-01-19:
The KDSETMODE ioctl with KD_TEXT might also be an option here, need to test it.

tlaurion, 2023-01-19:
@JonathonHall-Purism will have to recheck since t430/x230 have discrepencies.
the x230 has no gfx init under coreboot, and relies solely on i915 driver where the kexec normally works under those conditions, at the condition that i915 is also provided inside of the initrd to bring drm+console early, prior of plymouth to be able to propose a prompt for disk unlock. On t430, it seems that we needed to activate libgfxinit for some reasons that I do not remember (Nitrokey tested those + community members)

@123ahaha confirms here that typing disk unlock key passphrase blindly unlocks disk and then GUI shows correctly.
Hypothesis here is that either ADD_KERNEL_OPTION misses something in the board configs to properly pass good options to i915 driver. Older i915 driver doesn't need any other thing then DRM to activate display (ivy/sandy).

I will try to find time to test this again on x230 but I'm not quite sure my tests here would be helpful for Librems. Everything depends on what is provided in the initrd of the OS for early boot, where text based init normally works. The issues (untackled) with plymout are still a mystery to me (bmc AST2400 on my todo list) to have graphical installers work on vga for Talos II.

Sorry for not being more helpful here.

123ahaha, 2023-01-19:
> @JonathonHall-Purism will have to recheck since t430/x230 have discrepencies. 
Yes, I have a T430 (with and without dgpu) and a x230 all under devuan and they are working fine with heads !

>I will try to find time to test this again on x230 but I'm not quite sure my tests here would be helpful for Librems.
Let me know if you need my librem_14 to test rom. If i brick the board, I have the clip to externally flash the bios.

123ahaha, 2023-02-23:
An update here, i went with debian bookworm install, and i have no issue while kexec-ing into Debian 6.1.8-1 kernel. I just saw the module update of i915 :

```
$lspci | grep ' VGA ' | cut -d" " -f 1 | xargs -i lspci -v -s {}

00:02.0 VGA compatible controller: Intel Corporation Comet Lake UHD Graphics (rev 04) (prog-if 00 [VGA controller])
	DeviceName: VGA compatible controller
	Subsystem: Intel Corporation Comet Lake UHD Graphics
	Flags: bus master, fast devsel, latency 0, IRQ 140
	Memory at b0000000 (64-bit, non-prefetchable) [size=16M]
	Memory at a0000000 (64-bit, prefetchable) [size=256M]
	I/O ports at 1000 [size=64]
	Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]
	Capabilities: [40] Vendor Specific Information: Len=0c <?>
	Capabilities: [70] Express Root Complex Integrated Endpoint, MSI 00
	Capabilities: [ac] MSI: Enable+ Count=1/1 Maskable- 64bit-
	Capabilities: [d0] Power Management version 2
	Capabilities: [100] Process Address Space ID (PASID)
	Capabilities: [200] Address Translation Service (ATS)
	Capabilities: [300] Page Request Interface (PRI)
	Kernel driver in use: i915
	Kernel modules: i915
```

tlaurion, 2023-03-01:
@123ahaha a recap and close of this issue would be appreciated!

123ahaha, 2023-03-10:
I will close the exit but I could not find a solution to this problem.
Recap:
on computers with a graphics card requiring the i915 module, and under debian 11 (and derivatives), heads will boot fine, if you have an FDE, enter the password blindly, the OS should boot and load the module.
I read the issue Test T440p with Coreboot NO_GFX_INIT [ #1323](https://github.com/osresearch/heads/issues/1323), The solution would be to load the i915 module in the initramfs, I thought I did it ... but I do not know

closed 2023-03-10 by 123ahaha

tlaurion, 2023-03-10:
@123ahaha I think this is same debate happening under #1323.
@JonathonHall-Purism [you talked](https://github.com/osresearch/heads/issues/1285#issuecomment-1397501964) about:
> The KDSETMODE ioctl with KD_TEXT might also be an option here, need to test it.

I'm not sure how that would be handled. Would we be inspecting initramfs to see if expected KEXEC_KERNEL_OPTION_ADD contains i915 parameters or something and if not, switch back KDSETMODE back to text so that installers fallback into basic text framebuffer if expected driver isn't found?