linuxboot/heads #1320: Heads cant boot from USB-drive with Ventoy

fhvyhjriur, 2023-02-23:
I am using Ventoy to be able to have many ISO files on one USB-Drive and boot the ISO file i need at the moment.
Latest Heads version from yesterday ( heads-x230-maximized-v0.2.0-1380-gc4b964c.rom ) cant start Ventoy. Ventoy is able to start on any Coreboot payload i tried. It work fine with SeaBIOS and it works fine with TianoCore. Reason for this wide compatibility is, that Ventoy adds a boot menu for BIOS-mode and a boot menu for EFI-Mode. But Heads cant start Ventoy BIOS-mode and also cant start Ventoy EFI-mode.
Ventoy is free software and widely used for the use case of many ISO files on one flash drive to be able to choose from. Could someone take a look why it is not starting with Heads? Such basic functionality (booting from Ventoy USB-drive) should work.

https://github.com/ventoy/Ventoy

tlaurion, 2023-02-24:
Screenshots welcome or explanation why you think heads should boot (kexec into another grub.cfg defined boot entries, multiboot compatible Linux OSes only) would be welcome. Otherwise I understand you expect me to download, test, and make Heads compatible with XYZ for free?

tlaurion, 2023-03-01:
@fhvyhjriur input needed

natterangell, 2023-09-15:
I also use ventoy from time to time. The reason why @fhvyhjriur might be a little confused is because heads by nature is used to kexec into another linux kernel, not chainload into another boot loader. 

If you want to reuse your ventoy usb drive to do USB boot on a machine running heads without having to reformat the drive, there is a workaround. It's actually a little more workable now due to exFAT support in heads, which is what I think most ventoy users will have on the partition containing their ISO-files.

1. Copy the (linux) iso files you want to use to the root of the partition where you have your ISO folder.
2. boot heads and enter the recovery console
3. run `mount-usb` to detect and mount the usb-drive (this happens in read-only mode)
4. remount the partition containing ISO-files read-write, e.g.: `umount /dev/sdb1 && mkdir /iso && mount /dev/sdb1 /iso`. Adjust as necessary if you have several disks in your machine.
5. `cd /iso`
6. `gpg --output <iso_name>.sig --detach-sig <iso_name>`. This will prompt you for your GPG key PIN, so it needs to be attached. Repeat for all the ISO files you want
7. reboot and choose usb boot from boot menu, you should now be able to boot the ISO-files from your usb-partition.

This only works for linux distributions, heads (obviously) still need to kexec into the grub entries on the ISO.

Works fine for me. You can boot TAILS this way, if you want, or a live distro if you need to mount LUKS partitions and fix something etc.

@tlaurion, I find the wiki good enough to boot installation media, but maybe it could be expanded a little to clarify exactly what's going on when heads boot from USB. Thoughts?

tlaurion, 2023-09-16:
> Works fine for me. You can boot TAILS this way, if you want, or a live distro if you need to mount LUKS partitions and fix something etc.\n\n@tlaurion, I find the wiki good enough to boot installation media, but maybe it could be expanded a little to clarify exactly what's going on when heads boot from USB. Thoughts?

Yes. The reason it is not more documented is that as usual I have difficulty documenting things that ideally shouldn't.

Some context at https://github.com/osresearch/heads/issues/1438

Ventoy kinda redo what Heads does natively. Ventoy doesn't do authenticity +integrity validation. Heads only does it for ISOs that are detached signed. And not all ISOs are detached signed. 

Most of the ISOs have hashes file + detached signature of the hash file. This is what #1438 considers so advanced users could generate hashes to be verified prior of signing them.

A bit of history can explain why newer hashing+detached signature are not currently supported under Heads, but is not reconsidered. As you know, Heads uses busybox to provide tools needed, including sha256sum. And then gnupg as gpg toolstacks for signature. 

Concretely, if we take Fedora as an example, there was a time where Fedora was providing ISOs and detached signatures, just like arch Linux tails and QubesOS do today. 

But as most of other OSes, whey migrated to separate hash file + detached signature. https://labs.fedoraproject.org/en/verify

The ideal would be to support that scheme, and  #1438 comes with the same conclusion to present a hash to be verified to the user prior of detach signing it. If upstream distributions don't provide a way to verify integrity of the ISO then signing an iso that isn't verified is not really a good idea, waiting for disaster to happen while installing, creating weird errors and causing random issues that are really hard to troubleshoot and blame to a bad thumb drive of memory issues. 

Tldr: https://labs.fedoraproject.org/en/verify should be enforced under Heads alongside detached signature verification currently supported under Heads by limited number of Linux distributions. It's easier to demand upstream to produce hash files and detach sign them after the fact then ask them to detach sign their ISOs on the ISO builder. The reasoning is simple, producing a detached signature of the hashes doesn't require to expose the private key to the Continuous Integration (CI) or the builder, they can be signed after the fact. The hash file and iso can be downloaded and tested independently, and if they match be detached signed and the detached signature uploaded after validation, not by the CI.

tlaurion, 2023-09-16:
Bing generated workaround on busybox limitation:

`#!/bin/bash
# split-checksum.sh: Split a CHECKSUM file into separate files for each algorithm

# Check if a CHECKSUM file is provided as an argument
if [ $# -eq 0 ]; then
    echo "Usage: $0 CHECKSUM"
    exit 1
fi

# Loop through each line of the CHECKSUM file
while read line; do
    # Extract the algorithm and the checksum from the line
    algo=$(echo $line | cut -d' ' -f1)
    checksum=$(echo $line | cut -d' ' -f2)

    # Append the line to a file named after the algorithm
    echo $line >> $algo-CHECKSUM
done < $1

# Verify the checksums of each file using sha256sum
for file in *-CHECKSUM; do
    sha256sum -c $file
done`

tlaurion, 2023-09-16:
| Linux Distribution | Example of iso file | Example of checksum file | Example of signature file |
| ------------------ | ------------------- | ------------------------ | ------------------------- |
| Fedora             | Fedora-Workstation-Live-x86_64-34-1.2.iso | Fedora-Workstation-Live-x86_64-34-1.2.iso-CHECKSUM | Fedora-Workstation-Live-x86_64-34-1.2.iso-CHECKSUM.asc |
| CentOS             | CentOS-Stream-8-x86_64-latest-dvd1.iso    | CentOS-Stream-8-x86_64-latest-dvd1.iso-CHECKSUM    | CentOS-Stream-8-x86_64-latest-dvd1.iso-CHECKSUM.asc    |
| openSUSE           | openSUSE-Leap-15.3-DVD-x86_64.iso          | openSUSE-Leap-15.3-DVD-x86_64.iso-CHECKSUM          | openSUSE-Leap-15.3-DVD-x86_64.iso-CHECKSUM.asc          |
| Debian             | debian-live-10.9.0-amd64-standard.iso      | debian-live-10.9.0-amd64-standard.iso-CHECKSUM      | debian-live-10.9.0-amd64-standard.iso-CHECKSUM.asc      |

tlaurion, 2023-09-17:
Needs to be reworked

```
#!/bin/bash
# verify-checksum.sh: Verify the checksum file against its signature and then verify the SHA256 hash from the checksum file

# Check if an iso file is provided as an argument
if [ $# -ne 1 ]; then
    echo "Usage: $0 ISO"
    exit 1
fi

# Search for a checksum file that matches the iso file name with a wildcard pattern using find
checksum_file=$(find . -type f -name "$1-*")

# Check if a checksum file is found
if [ -z "$checksum_file" ]; then
    echo "No checksum file found."
    exit 2
fi

# Search for a signature file that matches the checksum file name with either .asc or .sig extension using find
signature_file=$(find . -type f \( -name "$checksum_file.asc" -o -name "$checksum_file.sig" \))

# Check if a signature file is found
if [ -z "$signature_file" ]; then
    echo "No signature file found."
    exit 3
fi

# Verify the checksum file against its detached signature using gpg
gpg --verify $signature_file $checksum_file || die "The checksum file is not verified."

# Check if the checksum file is a valid hash file using grep
grep -qP 'SHA256 \(\S+\) = \K\S+' $checksum_file || die "The checksum file is not valid."

# Extract only the SHA256 hash from the checksum file using grep
sha256=$(grep -oP 'SHA256 \(\S+\) = \K\S+' $checksum_file)

# Compare the SHA256 hash with the iso file using sha256sum
echo "$sha256  $1" | sha256sum -c || die "The iso file is not verified."

# Print a success message
echo "The checksum and the iso file are verified."
```

tlaurion, 2024-08-19:
Interest for Ventoy restated on Dahsaro Premier support channel (subscribers only) https://matrix.to/#/!RNcjJXCGHiyxXCHpKv:matrix.org/$oVg45lpw9Wev7EL3NTlgBGZin1C-NFVZ8YIAUeHlFU4?via=matrix.org&via=nitro.chat&via=matrix.3mdeb.com

Reopening to see interest and how Heads could support this (booting unverified integrity/authenticity ISOs)

tlaurion, 2024-08-19:
PoC code that could be integrated into "Unsafe USB boot" at https://github.com/linuxboot/heads/issues/1438#issuecomment-1722386799

Comments?

natterangell, 2024-09-10:
I would also welcome this. 

Personally, I'd prefer that ISOs could be booted unsafely without any verification (maybe with a warning) as several distributions don't distribute detached signatures, and signing the iso with your own key does little to increase security of a downloaded ISO, and it's a little clunky to do. Since we can already boot any iso written to a USB without verification, I don't see the difference.

natterangell, 2024-09-10:
Not to mention, it makes a whole lot of sense to use ventoy now that most flash drives are pretty large, so it's a better use of space. And with exfat now supported by Heads, the ISO-partition on a ventoy flash drive can be used with exfast and thus available cross-platform. Now problem with large ISO files either.

tlaurion, 2026-10-05:
fixed by https://github.com/linuxboot/heads/pull/2130

closed 2026-10-05 by tlaurion

tlaurion, 2026-10-05:
(just to be clear: Ventoy is not recommended, but if used to boot other laptops, isos on the same ventoy drive (exfat expected) should work with Heads as long as exfat is supported by the distro and at usb boot (mounting usb exfat partition said to contain iso) finds the iso and continues booting from iso's initramfs: not a heads problem, which #2130 will warn if iso can't be booted so users open issues in distributions to fix: not a heads problem at this point.