linuxboot/heads #1322: Plausible deniability of encryption

ZooWooHoo, 2023-02-28:
Is this something that’s possible? I’m not sure if Heads requires attached LUKS headers but I think this would be a good feature that some individuals would benefit from.

tlaurion, 2023-03-01:
@ZooWooHoo @rafaelreis-r can you give more info on what you would want?

Example. Heads boots. No /boot content?

FDE support (duplicate issue).

The Heads would do what? 

Ask for a USB storage Luks header to load?

Youbwoukd want the LUKS header to be completely stored under TPM? 

Tom 1.2? TPM 2?

ZooWooHoo, 2023-03-02:
Plausible deniability in my mind means you can say you haven’t got around to installing an os.

I was under the impression an encrypted drive without LUKS headers would look like a blank drive (maybe freshly wiped) with random data (unpartitioned) to forensics.

I guess heads would ask for a USB with LUKS headers in the case the drive isn’t bootable on its own. That doesn’t help the plausible deniability but it also doesn’t prove the existence of hidden data as it would be normal heads behavior.

PaulKMandal, 2024-09-21:
> Basically, this feature would depend on Heads supporting full FDE (/boot inside of the LUKS container) which would need a bit of work.
> 
> Heads checks for valid /boot content on boot. Heads prompt when no OS is found is currently to install an OS by booting from USB. Once Full FDE is supported, that prompt would simply need to propose to the user to load an external LUKS header from a USB thumb drive, prior of validating the HOTP counter under /boot (to validate firmware integrity) or doing anything else, unlocking LUKS and then mounting /boot as normal and propose boot options. There should also be a menu option permitting to export that header on USB storage and then wipe it.
> 
> Note that the OEM factory reset menu/Re-Ownership stores the boot device name inside of the config.user overlay injected in ROM. That device name would probably need to be switched to UUID there, or the boot detection script modified to detect LUKS container mapped partitions and select it dynamically.
> 
> But yeah, first task here would be to support full FDE under Heads.
> 
> And then discuss the details of how this plausible encryption deniability feature should be seen for encryption deniability. But in my head, this is more additional data security than plausible encryption deniability as proposed in this issue.
> 
> A full forensic on the drive would most probably show discrepancies in entropy across regions of the disk, proving to the forensic analyser that the content of the drive was not wiped but that there is actually encrypted content there. And then by adding a prompt in the menus for loading a detached LUKS header would clearly show that possibility and the most plausible case here: drive contains encrypted material.
> 
> An interrogator would still request the release of that LUKS header and its passphrase. I understand the dream, the desire for such feature, but the truth here is that it won't really provide it if easy to use and where Heads would offer, for usability reasons, to load an external LUKS header and where that hint would push the auditor to clone the drive even more and look for that LUKS header. All thread models being valid, why not store that content encrypted on the cloud (self-hosted if needed) to download upon arrival, or again, travel without data that would prevent you to type your decryption passphrase if requested. The basic opsecs principles still prevails: if you think you will be asked to type a passphrase to decrypt your drive, you should minimize the content that can be extracted if under threat with a hammer. Encryption here provides privacy, added security and authentication. But plausible deniability with encryption normally requires hidden volumes and such, which would then happen on top of a booted OS. I think not having any OS installed is a big trigger at security screening. Having your disk shipped to destination might be a better idea. Or having the tools and data in the cloud might be a way better option then trying to hide a whole installation on top of Heads.
> 
> Thoughts/propositions?

Unfortunately I'm very late to the discussion here, but veracrypt has a feature that allows someone to hide an encrypted partition inside another encrypted partition, thus hiding the entropy discrepancy. Is Heads only compatible with LUKS, or is there any support with VeraCrypt? I cannot find any mentions of veracrypt on the Wiki.

tlaurion, 2024-10-10:
Nobody is saying what it woukd look like to them since the opening of this issue.

Random data on drive would tilt forensic.
What heads should be doing when no partitions found on drive outside of what it's doing now and prompts user to boot from usb?

Please help me help you.
This issue is stalling without useful input.

What would you want heads to do? What would it change that it supports veracrypt?

Nothing will happen here without a clear use case. Desire. Goal.