linuxboot/heads #1332: Improve whiptail dual console board configurations (would be useful for qemu)

tlaurion, 2023-03-09:
Not really sure why the following gives to qemu a non-complete interactive shell on host's console, launching qemu:

```
user@heads-tests:~/heads$ git diff
diff --git a/Makefile b/Makefile
index 3dfab86594..579b4cf25f 100644
--- a/Makefile
+++ b/Makefile
@@ -472,6 +472,7 @@ bin_modules-$(CONFIG_NKSTORECLI) += nkstorecli
 bin_modules-$(CONFIG_OPENSSL) += openssl
 bin_modules-$(CONFIG_TPM2_TOOLS) += tpm2-tools
 bin_modules-$(CONFIG_BASH) += bash
+bin_modules-$(CONFIG_USE_AGETTY) += util-linux
 
 $(foreach m, $(bin_modules-y), \
        $(call map,initrd_bin_add,$(call bins,$m)) \
diff --git a/boards/qemu-coreboot-whiptail-tpm1/qemu-coreboot-whiptail-tpm1.config b/boards/qemu-coreboot-whiptail-tpm1/qemu-coreboot-whiptail-tpm1.config
index 37ae90a5d1..852280cd45 100644
--- a/boards/qemu-coreboot-whiptail-tpm1/qemu-coreboot-whiptail-tpm1.config
+++ b/boards/qemu-coreboot-whiptail-tpm1/qemu-coreboot-whiptail-tpm1.config
@@ -41,6 +41,10 @@ CONFIG_MSRTOOLS=y
 CONFIG_NEWT=y
 CONFIG_SLANG=y
 
+#To output whiptail to both qemu and local console
+export CONFIG_USE_AGETTY=y
+export CONFIG_BOOT_EXTRA_TTYS="ttyS0"
+
 endif
 
 export CONFIG_LINUX_USB_COMPANION_CONTROLLER=y
@@ -55,7 +59,7 @@ export CONFIG_BOOTSCRIPT=/bin/gui-init
 #export CONFIG_BOOTSCRIPT=/bin/generic-init
 export CONFIG_BOOT_REQ_HASH=n
 export CONFIG_BOOT_REQ_ROLLBACK=n
-export CONFIG_BOOT_RECOVERY_SERIAL="/dev/ttyS0"
+#export CONFIG_BOOT_RECOVERY_SERIAL="/dev/ttyS0"
 export CONFIG_BOOT_KERNEL_ADD="console=ttyS0 console=tty systemd.zram=0"
 export CONFIG_BOOT_KERNEL_REMOVE="quiet rhgb splash"
```

On talos II server board having same config, but used over ssh, correcting pseudo-terminal through ssh call:
`ssh -t root@talos "obmcutil poweron && obmc-console-client"`

Works perfectly and permits us to have a dual whiptail output on both VGA and bmc terminal, accessed through ssh.
Note that the problem to get this working was to pass `ssh -t` above.

Not sure how to have the same on qemu call so that the `--serial stdio` can offer the same functionality.

qemu whiptail board could be really useful in that configuration, since whiptail output would contain trails of console output in addition to TRACE and DEBUG output under /tmp/debug.log

krystian-hebel, 2023-03-09:
I've noticed that for one of the configurations for Talos (can't remember if it was server or workstation) input given through serial console would result in action on VGA output, but not on serial. Should those outputs be mirrored or are they two separate ttys?

tlaurion, 2023-03-09:
@krystian-hebel as of now for server board (whiptail), output is on different tty, not mirrored. Issue you talk about is with workstation (fbwhiptail) which is to be deprecated in favor of a single Talos board, actually repurposing server board config.

Whiptail on different TTy is a plus in our use case, where either one can be used to drive either menu and have TOTP refreshed on either terminal, but if two are used at same time and one goes to recovery, the other one would not know until either going back to main menu and having TOTP secret unable to unseal or when trying to release TPM disk unlock key since measurements would not unseal, going to recovery extending PCR4 and invalidating both sealed secret at unseal operation.

Mirroring would be interesting though. Didn't know it would be possible?

tlaurion, 2023-09-06:
There is currently one issue in dual console boards like qemu.

When prompts of applications are asked, only qemu console (not console having called qemu) shows prompt for pintentry.

Have not had time to look into that but it is annoying when testing.

@JonathonHall-Purism ideas?