linuxboot/heads #1348: LUKS encrypted NixOS installation does not boot

septem9er, 2023-03-21:
## Please identify some basic details to help process the report

### A. Provide Hardware Details
Librem 14
**1. What board are you using (see list of boards [here](https://github.com/eganonoa/heads/tree/master/boards))?**
Librem 14
**2. Does your computer have a dGPU or is it iGPU-only?**
- [ ] dGPU
- [X] iGPU-only

**3. Who installed Heads on this computer?**
- [ ] Insurgo
- [ ] Nitrokey
- [X] Purism
- [ ] Other provider
- [ ] Self-installed

**4. What PGP key is being used?**
- [X] Librem Key
- [ ] Nitrokey Pro 2
- [ ] Nitrokey Storage
- [ ] Yubikey
- [ ] Other

**5. Are you using the PGP key to provide HOTP verification?**
- [X] Yes
- [ ] No
- [ ] I don't know

### B. Identify how the board was flashed

**1. Is this problem related to updating heads or flashing it for the first time?**
- [ ] First-time flash
- [ ] Updating heads 

**2. If the problem is related to an update, how did you attempt to apply the update?**
- [ ] Using the Heads GUI
- [ ] Flashrom via the Recovery Shell
- [ ] External flashing

**3. How was Heads initially flashed**
- [ ] External flashing
- [ ] Internal-only / 1vyrain
- [X] Don't know

**4. Was the board flashed with a maximized or non-maximized/legacy rom?**
- [ ] Maximized
- [ ] Non-maximized / legacy
- [X] I don't know

**5. If Heads was externally flashed, was IFD unlocked?**
- [ ] Yes
- [ ] No
- [ ] Don't know

### C. Identify the rom related to this bug report

**1. Did you download or build the rom at issue in this bug report?**
- [ ] I downloaded it
- [ ] I built it

**2. If you downloaded your rom, where did you get it from?**
- [ ] Heads CircleCi
- [ ] Purism
- [ ] Nitrokey
- [ ] Somewhere else (please identify)

*Please provide the release number or otherwise identify the rom downloaded*

**3. If you built your rom, which repository:branch did you use?**
- [ ] Heads:Master
- [ ] Other (please identify)

**4. What version of coreboot did you use in building?**
- [ ] 4.8.1 (current default in heads:master)
- [ ] 4.13
- [ ] 4.14
- [ ] 4.15
- [ ] Other (please specify)
- [ ] I don't know

**5. In building the rom where did you get the blobs?**
- [ ] No blobs required
- [ ] Provided by the company that installed Heads on the device
- [ ] Extracted from a backup rom taken from this device
- [ ] Extracted from another backup rom taken from another device (please identify the board model)
- [ ] Extracted from the online bios using the automated tools provided in Heads
- [ ] I don't know

## Please describe the problem

**Describe the bug**
When installing NixOS with LUKS encrypted root partition (boot is unencrypted), the system does not boot. When installing nix unencrypted it does work.

I did not fill out some of the information above, as the Librem 14 was preinstalled with coreboot and heads by purism. So I do cannot answer some of those questions.

If I can provide any additional information or help identify the issue, I'll gladly do that.

**To Reproduce**
- Install NixOS
- Choose manual partiioning
- Choose MBR as partition table
- Create an unencrypted boot partition (set legacy-boot flag)
- Create an LUKS encrypted root partition (set root flag)
- Choose that the bootloader should be installed into /boot
- After the installation is complete, try to boot into the system

**Expected behavior**
The system should be able to boot NixOS
**Actual behavior**
Nothing happens after `starting the new kernel`
**Screenshots**

**Additional context**
In the matrix chat one of you told me they would like to see the kexec line vs the line in the grub.cfg. So here is the full `/boot/grub/grub.cfg`
```
# Automatically generated.  DO NOT EDIT THIS FILE!

        search --set=drive1 --fs-uuid 8B97-CA37
        if [ -s $prefix/grubenv ]; then
          load_env
        fi

        # ‘grub-reboot’ sets a one-time saved entry, which we process here and
        # then delete.
        if [ "${next_entry}" ]; then
          set default="${next_entry}"
          set next_entry=
          save_env next_entry
          set timeout=1
          set boot_once=true
        else
          set default=0
          set timeout=5
        fi

        function savedefault {
            if [ -z "${boot_once}"]; then
            saved_entry="${chosen}"
            save_env saved_entry
            fi
        }

        # Setup the graphics stack for bios and efi systems
        if [ "${grub_platform}" = "efi" ]; then
          insmod efi_gop
          insmod efi_uga
        else
          insmod vbe
        fi
    
            insmod font
            if loadfont ($drive1)//converted-font.pf2; then
              insmod gfxterm
              if [ "${grub_platform}" = "efi" ]; then
                set gfxmode=auto
                set gfxpayload=keep
              else
                set gfxmode=1024x768
                set gfxpayload=text
              fi
              terminal_output gfxterm
            fi
        
            background_color '#2F302F'
            
            insmod png
            if background_image --mode 'normal' ($drive1)//background.png; then
              set color_normal=white/black
              set color_highlight=black/white
            else
              set menu_color_normal=cyan/blue
              set menu_color_highlight=white/blue
            fi
        

menuentry "NixOS - Default" --class nixos --unrestricted {
search --set=drive1 --fs-uuid 8B97-CA37
  linux ($drive1)//kernels/z6vmhagxd308pffxnx47pvi27pbi5y5q-linux-5.15.102-bzImage init=/nix/store/y0l8j0n3g5v9zyxw04lrxhxq2qk8np2h-nixos-system-nixos-22.11.3184.9b8e5abb183/init loglevel=4
  initrd ($drive1)//kernels/nv544pzpzldq0vpgajmgjkfym448nha4-initrd-linux-5.15.102-initrd ($drive1)//kernels/nv544pzpzldq0vpgajmgjkfym448nha4-initrd-linux-5.15.102-initrd-secrets
}

submenu "NixOS - All configurations" --class submenu {
menuentry "NixOS - Configuration 1 (2023-03-21 - 22.11.3184.9b8e5abb183)" --class nixos {
search --set=drive1 --fs-uuid 8B97-CA37
  linux ($drive1)//kernels/z6vmhagxd308pffxnx47pvi27pbi5y5q-linux-5.15.102-bzImage init=/nix/store/y0l8j0n3g5v9zyxw04lrxhxq2qk8np2h-nixos-system-nixos-22.11.3184.9b8e5abb183/init loglevel=4
  initrd ($drive1)//kernels/nv544pzpzldq0vpgajmgjkfym448nha4-initrd-linux-5.15.102-initrd ($drive1)//kernels/nv544pzpzldq0vpgajmgjkfym448nha4-initrd-linux-5.15.102-initrd-secrets
}

}

```

tlaurion, 2023-03-21:
I915 doesn't seem to be compiled in initrd.

septem9er, 2023-03-21:
> I915 doesn't seem to be compiled in initrd.

I did have a similiar issue when installing arch, where adding i915 to the initramfs solved the issue.

So I thought this nicht be the issue and already tried the following:
- Waiting a bit, typing in the LUKS password and Pressung enger. This did work on the arch systemy because the i915 issue was just about it not showing the password prompt
- I tried to add the i915 module to the initramfs. I did ADS the module to the nix config and told nix to redo the Installation of the bootloader. However, I am not sure if that worked correctly. Can I test this somehow?

JonathonHall-Purism, 2023-03-21:
You can list the contents of your initrd and see if i915 is in there.  Something like:

`zcat <your-initrd> | cpio -t | grep i915`

septem9er, 2023-03-21:
Thanks. Seems like I added the i915 module succesfully to the initrd. I checked the contents and there are paths for i915 now.

However, the boot still doesn't work even though the new initrd is used (it was in the kexec line).

tlaurion, 2023-03-21:
Something else is at play with 5.x kernel config. I was able to see difference in behavior between 4.x and 5.x under other issue where even without i915 driver things worked before but not anymore. Unfortunately, that will need a bit more troubleshooting

tlaurion, 2023-03-22:
Note that we are trying to figure out right now what is happening with vgaarb, fbdev and i915 from 5.x kernel kexec'ing into something else. For some reason it works from 4.1x but not 5.x, and 4.x is able to boot into Tinycore (other opened issue) without Tinycore providing i915. So.... No idea for the moment, troubleshooted past two days, now going back to reading doc.

tlaurion, 2023-06-14:
@septem9er originally said in OP

> Describe the bug
> When installing NixOS with LUKS encrypted root partition (boot is unencrypted), the system does not boot. When installing nix unencrypted it does work.

@septem9er What is the output of 
`zcat <your-initrd> | cpio -t | grep i915`

tlaurion, 2023-06-14:
```

menuentry "NixOS - Default" --class nixos --unrestricted {
search --set=drive1 --fs-uuid 8B97-CA37
  linux ($drive1)//kernels/z6vmhagxd308pffxnx47pvi27pbi5y5q-linux-5.15.102-bzImage init=/nix/store/y0l8j0n3g5v9zyxw04lrxhxq2qk8np2h-nixos-system-nixos-22.11.3184.9b8e5abb183/init loglevel=4
  initrd ($drive1)//kernels/nv544pzpzldq0vpgajmgjkfym448nha4-initrd-linux-5.15.102-initrd ($drive1)//kernels/nv544pzpzldq0vpgajmgjkfym448nha4-initrd-linux-5.15.102-initrd-secrets
}

submenu "NixOS - All configurations" --class submenu {
menuentry "NixOS - Configuration 1 (2023-03-21 - 22.11.3184.9b8e5abb183)" --class nixos {
search --set=drive1 --fs-uuid 8B97-CA37
  linux ($drive1)//kernels/z6vmhagxd308pffxnx47pvi27pbi5y5q-linux-5.15.102-bzImage init=/nix/store/y0l8j0n3g5v9zyxw04lrxhxq2qk8np2h-nixos-system-nixos-22.11.3184.9b8e5abb183/init loglevel=4
  initrd ($drive1)//kernels/nv544pzpzldq0vpgajmgjkfym448nha4-initrd-linux-5.15.102-initrd ($drive1)//kernels/nv544pzpzldq0vpgajmgjkfym448nha4-initrd-linux-5.15.102-initrd-secrets
}
```

That won't fly under Heads.
There is no logic to be able to replace $drive1 with 8B97-CA37. Its just removed.

@septem9er : please provide kexec call picture on boot.
From my understanding init is searched on mounted / (root) and kernel and initrd are expected to be found per fs-uuid replacement through grub normally.

JonathonHall-Purism, 2023-06-14:
@septem9er The issue with i915 framebuffer boot is resolved in Heads now and will be in PureBoot Release-27 (I'm working on building and testing this right now).

From the snippets above, it looks like the kernel is indeed on the boot filesystem (`8B97-CA37` is a FAT32 filesystem ID, btrfs would have an actual UUID), so Heads/PB might be able to load this kernel/initrd correctly just by ignoring drive1.  Maybe i915 is the only issue.

Could you please try Release-27 as soon as it is out to see if this will work with encrypted root for you?

JonathonHall-Purism, 2023-06-22:
@septem9er PureBoot Release-27 is out, could you please update and give it a try?  (https://source.puri.sm/firmware/utility/#usage)

septem9er, 2023-06-23:
@JonathonHall-Purism After installing the Release-27, and installing NixOS with an encrypted root, the boot still fails. But in another way than before, so it did help! However, I'm not sure if heads is the issue here. Could you have a look?

That is how I installed (and partitioned) NixOS:
[image]

And this is how the boot fails because of a kernel panic:
[image]

septem9er, 2023-06-25:
> That won't fly under Heads. There is no logic to be able to replace $drive1 with 8B97-CA37. Its just removed.
>
> @septem9er : please provide kexec call picture on boot. From my understanding init is [...]
@tlaurion 
Ah, okay. So I guess this is the issue now. Here is the current grub.cfg:

```
# Automatically generated.  DO NOT EDIT THIS FILE! 

    search --set=drive1 --fs-uuid d6fa6ca3-71f6-4a65-8521-927494eded8c 
    if [ -s $prefix/grubenv ]; then 
      load_env 
    fi 

    # ‘grub-reboot’ sets a one-time saved entry, which we process here and 
    # then delete. 
    if [ "${next_entry}" ]; then 
      set default="${next_entry}" 
      set next_entry= 
      save_env next_entry 
      set timeout=1 
      set boot_once=true 
    else 
      set default=0 
      set timeout=5 
    fi 

    function savedefault { 
        if [ -z "${boot_once}"]; then 
        saved_entry="${chosen}" 
        save_env saved_entry 
        fi 
    } 

    # Setup the graphics stack for bios and efi systems 
    if [ "${grub_platform}" = "efi" ]; then 
      insmod efi_gop 
      insmod efi_uga 
    else 
      insmod vbe 
    fi 

        insmod font 
        if loadfont ($drive1)//converted-font.pf2; then 
          insmod gfxterm 
          if [ "${grub_platform}" = "efi" ]; then 
            set gfxmode=auto 
            set gfxpayload=keep 
          else 
            set gfxmode=1024x768 
            set gfxpayload=text 
          fi 
          terminal_output gfxterm 
        fi 
     
        background_color '#2F302F' 
         
        insmod png 
        if background_image --mode 'normal' ($drive1)//background.png; then 
          set color_normal=white/black 
          set color_highlight=black/white 
        else 
          set menu_color_normal=cyan/blue 
          set menu_color_highlight=white/blue 
        fi 
     

menuentry "NixOS - Default" --class nixos --unrestricted { 
search --set=drive1 --fs-uuid d6fa6ca3-71f6-4a65-8521-927494eded8c 
  linux ($drive1)//kernels/q2828ynf790bql5k3jgn0nham38p4gpl-linux-6.1.34-bzImage init=/nix/store/bsfxa9xc6vgpr9nn6cl0mhnsxnmm3ssp-nixos-system-nixos-23.05.1249.ef0bc397634/init loglevel=4 
  initrd ($drive1)//kernels/5m6v3lymbzv0zz74iawlbdzs2r0hgzq6-initrd-linux-6.1.34-initrd ($drive1)//kernels/bsfxa9xc6vgpr9nn6cl0mhnsxnmm3ssp-nixos-system-nixos-23.05.1249.ef0bc397634-secrets 
} 

submenu "NixOS - All configurations" --class submenu { 
menuentry "NixOS - Configuration 1 (2023-06-25 - 23.05.1249.ef0bc397634)" --class nixos { 
search --set=drive1 --fs-uuid d6fa6ca3-71f6-4a65-8521-927494eded8c 
  linux ($drive1)//kernels/q2828ynf790bql5k3jgn0nham38p4gpl-linux-6.1.34-bzImage init=/nix/store/bsfxa9xc6vgpr9nn6cl0mhnsxnmm3ssp-nixos-system-nixos-23.05.1249.ef0bc397634/init loglevel=4 
  initrd ($drive1)//kernels/5m6v3lymbzv0zz74iawlbdzs2r0hgzq6-initrd-linux-6.1.34-initrd ($drive1)//kernels/bsfxa9xc6vgpr9nn6cl0mhnsxnmm3ssp-nixos-system-nixos-23.05.1249.ef0bc397634-secrets 
} 

}
```
And here is the kexec line of heads
[image]

> Can you please detail exactly how you did it so others can replicate and then we can find the issue and fix it?

I did it like this andI can replicate it:

- Download [NixOS] from (https://nixos.org/download.html#nix-more)
  - Click on more... -> ISO Image
  - Choose Plasma Desktop 64-Bit AMD/Intel
- Boot into the installer
- In the Graphical installer, choose ...
  - Welcome-User: anything
  - Desktop: Choose "No Desktop" (cause the install is faster)
  - Unfree Software: Allow
  - Partitioning: Manual partitioning
    - New partition Table -> MBR
    - Create partition: 1024MB, ext4, encrypt: no, /boot, label: boot, Flags: legacy-boot
    - Create Partition: 65536MB, swap,encrypt: no, -, label: swap, Flags: swap
    - Create Partition: space left, ext4, encrypt: yes, /, label:root, Flags: root
    - Install Bootloader on -> Boot partition (/boot)
  - Install
  - Reboot
  - Error as in the comment above.

JonathonHall-Purism, 2023-06-27:
Thanks @septem9er.

The encrypted setup has two initrds:

```
initrd ($drive1)//kernels/5m6v3lymbzv0zz74iawlbdzs2r0hgzq6-initrd-linux-6.1.34-initrd ($drive1)//kernels/bsfxa9xc6vgpr9nn6cl0mhnsxnmm3ssp-nixos-system-nixos-23.05.1249.ef0bc397634-secrets
```

Heads understands this but then tries to pass two `--initrd` arguments to kexec, which doesn't work - kexec just takes the last one.  This explains the error that's now visible, most of the initrd is missing, and since there's no init program Linux falls back to old logic to try to mount root on its own, which then fails, and the kernel panics with that error meaning it couldn't mount root.

`kexec-boot` will have to concatenate the two initrds, let me look into that.

JonathonHall-Purism, 2023-06-28:
Nix developers are addressing this upstream, and it sounds like the intent is that the second initrd would not be present at all once this is fixed.

This renders the PR moot - though GRUB allows passing multiple initrds, I don't know of any other case where this happens and it is probably not worth the baggage to carry this as nearly-untestable code.  Maybe Heads should just take the first one for simplicity, would have been less-wrong in this case.

@septem9er You might want to follow the discussion in the last Matrix link for updates, if Nix fixes this and the second initrd is gone, then this will work with PB/Heads as-is since the i915 graphics fix is already in.

@tlaurion If the fixes to Nix do eliminate the second initrd (not just leave it empty or something), then we can close #1426 and this issue IMO.

tlaurion, 2023-06-28:
They are waiting for their security team, but here is one of the fixes proposed https://github.com/vlinkz/calamares-nixos-extensions/commit/837ca4da5521a74d3b5ca6f7b88890a6713faa22

RaitoBezarius, 2023-06-28:
This was published as https://github.com/NixOS/calamares-nixos-extensions/security/advisories/GHSA-3rvf-24q2-24ww.

tlaurion, 2023-06-28:
Testing iso here if you are faster then myself to test https://drive.google.com/file/d/1ZIUc2POI6Jdil56VITDlMib5qHVvpvpc/view?usp=drive_link

septem9er, 2023-06-29:
Using this iso, I can boot into an encrypted NixOS instalation using heads. The initrd -secrets file isn't created.

Thanks for the help to both of you!

tlaurion, 2023-06-29:
@septem9er will reopen until official OS is released?

@RaitoBezarius please close with comment when the next release of ISO with updated graphical installer happens?

tlaurion, 2023-08-03:
@RaitoBezarius can we close?

RaitoBezarius, 2023-08-04:
Sorry for this, @vlinkz can you confirm this? I believe this is the case, but just in case.

vlinkz, 2023-08-04:
> Sorry for this, @vlinkz can you confirm this? I believe this is the case, but just in case.

We need to merge this first: https://github.com/NixOS/nixpkgs/pull/247163

tlaurion, 2023-10-22:
@septem9er @vlinkz @tlaurion 

> > Sorry for this, @vlinkz can you confirm this? I believe this is the case, but just in case.
> 
> We need to merge this first: [NixOS/nixpkgs#247163](https://github.com/NixOS/nixpkgs/pull/247163)

This is merged. Can we close?

RaitoBezarius, 2023-10-22:
Correct!

Nintorac, 2024-07-05:
This may have regressed in the 24.05 graphical installer (specifically `nixos-gnome-24.05.2411.706eef542dec-x86_64-linux.iso`)

I have a similar setup with an unencrypted `/boot` and an encrypted `/`. When I start the OS it just boots all the way to user login not requiring my decryption password at any time.

Here is my `/boot/kernels`, including the secrets one that tlaurion mentioned being problematic.
[image]

I don't have the knowledge to validate my secret is in here though

septem9er, 2024-07-05:
Hey @Nintoracy,
thank you for the report. This is a NixOS issue, so here is not really the right place to report it. However, I was able to reproduce this issue and just now reported it to the NixOS security Team via Matrix. https://matrix.to/#/#security:nixos.org

Nintorac, 2024-07-05:
ooh my bad, didn't even look where I was :! 

Awesome, thanks!!

tlaurion, 2024-07-05:
Ongoing discussion around https://matrix.to/#/!NBBFPbiuttRgTqbrcY:nixos.org/$OP4Zb70brQuqRGpiPLpVhq2-iocmRydo8lkbSBEFKbg?via=matrix.org&via=tchncs.de&via=nixos.dev

tlaurion, 2024-07-05:
Switched to https://matrix.to/#/!NBBFPbiuttRgTqbrcY:nixos.org/$6hZzQzfqcjGYxj1qtlGDkKjrBZbf0sikdYCnbGNqx-U?via=matrix.org&via=tchncs.de&via=nixos.dev

emilazy, 2024-08-15:
We’ve fixed this and posted an advisory at https://github.com/NixOS/calamares-nixos-extensions/security/advisories/GHSA-vfxf-gpmj-2p25. Sorry that we let this regression slip through the cracks and that it took a while to address it; we have less in‐house Calamares expertise than I’d like. Thanks to @dalto8 for helping us review the fix.

closed 2024-09-06 by tlaurion