linuxboot/heads #1359: qemu-coreboot-whiptail-tpm2 fails with openssl error

ThePlexus, 2023-03-31:
## Context of the Build

**1. What board are you trying to build?**
qemu-coreboot-whiptail-tpm2

**2. What repository:branch are you using to build from?**
- [X] Heads:Master
- [ ] Other (please specify)

**3. What version of coreboot are you trying to build**
- [X] 4.13
- [ ] 4.14
- [ ] 4.15
- [ ] 4.17
- [ ] Other (please specify)

**4. In building the rom where did you get the blobs?**
- [X] No blobs required
- [ ] Provided by the company that installed Heads on the device
- [ ] Extracted from a backup rom taken from this device
- [ ] Extracted from another backup rom taken from another device (please identify the board model)
- [ ] Extracted from the online bios using the automated tools provided in Heads
- [ ] I don't know

**5. If using the automated tools to get the blobs did you run the relevant scripts in the blobs directory**
- [ ] Yes
- [ ] No

**6. What operating system are you using**

*please specify*

debian 11 x64 / mint LMDE 5

## Please describe the problem

This board fails to build with an OpenSSL output error
```
$ apt install -y build-essential zlib1g-dev uuid-dev libdigest-sha-perl libelf-dev bc bzip2 bison flex git gnupg gawk iasl m4 nasm patch python python2 python3 wget gnat cpio ccache pkg-config cmake libusb-1.0-0-dev autoconf texinfo ncurses-dev doxygen graphviz udev libudev1 libudev-dev automake libtool rsync innoextract sudo libssl-dev device-tree-compiler u-boot-tools sharutils e2fsprogs parted curl unzip
--SNIP
0 upgraded, 0 newly installed, 0 to remove and 34 not upgraded.
```
```
$ make BOARD=qemu-coreboot-whiptail-tpm2
---SNIP
2023-03-31 15:14:00+01:00 MAKE openssl
tail /home/user/heads/build/x86/log/openssl.log
-----
	apps/openssl-bin-openssl.o apps/openssl-bin-passwd.o \
	apps/openssl-bin-pkcs12.o apps/openssl-bin-pkcs7.o \
	apps/openssl-bin-pkcs8.o apps/openssl-bin-pkey.o \
	apps/openssl-bin-pkeyparam.o apps/openssl-bin-pkeyutl.o \
	apps/openssl-bin-prime.o apps/openssl-bin-progs.o \
	apps/openssl-bin-rand.o apps/openssl-bin-rehash.o \
	apps/openssl-bin-req.o apps/openssl-bin-rsa.o \
	apps/openssl-bin-rsautl.o apps/openssl-bin-s_client.o \
	apps/openssl-bin-s_server.o apps/openssl-bin-s_time.o \
	apps/openssl-bin-sess_id.o apps/openssl-bin-smime.o \
	apps/openssl-bin-speed.o apps/openssl-bin-spkac.o \
	apps/openssl-bin-srp.o apps/openssl-bin-storeutl.o \
	apps/openssl-bin-ts.o apps/openssl-bin-verify.o \
	apps/openssl-bin-version.o apps/openssl-bin-x509.o \
	apps/libapps.a -lssl -lcrypto -ldl -pthread 
x86_64-linux-musl-gcc: error: apps/openssl: No such file or directory
make[2]: *** [Makefile:18051: apps/openssl] Error 1
```

**To Reproduce**
Steps to reproduce the behavior:
install prerequisites, try to build 

probably something im missing, as ive never done a qemu before.

tlaurion, 2023-03-31:
@ThePlexus can you redo that with
`make BOARD=qemu-coreboot-whiptail-tpm2 V=1`

And/OR drop output of referred 
/home/user/heads/build/x86/log/openssl.log and additionally /home/user/heads/build/x86/log/openssl.configure.log ?

ThePlexus, 2023-03-31:
it seems to build OK now? very strange! but its OK and I get a ROM ... dmesg doesnt seem to show a disk error either but i suspect ive got a failing ssd. 

```
66d4a4638960f0cf21259a06b4d5e44afcbd69dc12c2cdd515bbfeab33dbf261  /home/user/heads/build/x86/qemu-coreboot-whiptail-tpm2/heads-qemu-coreboot-whiptail-tpm2-.rom
```

closed 2023-03-31 by ThePlexus