linuxboot/heads #1364: Fail to build binary under Debian 12

123ahaha, 2023-04-04:
## Context of the Build

**1. What board are you trying to build?**

**2. What repository:branch are you using to build from?**
- [x] Heads:Master
- [ ] Other (please specify)

**3. What version of coreboot are you trying to build**
- [ ] 4.13
- [ ] 4.14
- [ ] 4.15
- [x] 4.17
- [ ] Other (please specify)
this doesn't work for the librem_14 board, but I've also tried for the x230 htop maximized and the t430 htop maximized

**4. In building the rom where did you get the blobs?**
- [ ] No blobs required
- [ ] Provided by the company that installed Heads on the device
- [ ] Extracted from a backup rom taken from this device
- [ ] Extracted from another backup rom taken from another device (please identify the board model)
- [x] Extracted from the online bios using the automated tools provided in Heads
- [ ] I don't know

**5. If using the automated tools to get the blobs did you run the relevant scripts in the blobs directory**
- [x] Yes
- [ ] No

**6. What operating system are you using**

DEBIAN 12 (Linux 6.1.0-7-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.20-1 (2023-03-19) x86_64 GNU/Linux)

## Please describe the problem

I try to build heads (make BOARD=librem=14, t430... or x230....) but there is a compilation error :
When building librem_14 bin
```
Building packages ...
Building GMP v6.2.1 for host ... ok
Building MPFR v4.1.0 for host ... ok
Building MPC v1.2.1 for host ... ok
Building BINUTILS v2.37 for target ... ok
Building GCC v11.2.0 for target ... failed. Check 'build-i386-elf-GCC/build.log'.
make[3]: *** [Makefile:19: build_gcc] Error 1
make[2]: *** [Makefile:35: build-i386] Error 2
make[1]: *** [util/crossgcc/Makefile.inc:30: crossgcc-i386] Error 2
make[1]: Leaving directory '/home/user/applications/heads/build/x86/coreboot-4.17'
make: *** [modules/coreboot:106: /home/user/applications/heads/build/x86/coreboot-4.17/.xcompile] Error 2
```
librem_14 build.log : https://privatebin.net/?9e1e4af7ea744768#EWcCXse8c9kUkFtPbeMtisYzcschgdp5RoZSoNtxh5gq
[build.log](https://github.com/osresearch/heads/files/11147612/build.log)

When building t430-hotp-maximized
```
Unpacked and patched ... ok
Building packages ...
Building GMP v6.2.0 for host ... ok
Building MPFR v4.1.0 for host ... ok
Building MPC v1.2.0 for host ... ok
Building BINUTILS v2.35 for target ... ok
Building GCC v8.3.0 for target ... failed. Check 'build-i386-elf-GCC/build.log'.
make[3]: *** [Makefile:26: build_gcc] Error 1
make[2]: *** [Makefile:51: build-i386] Error 2
make[1]: *** [util/crossgcc/Makefile.inc:34: crossgcc-i386] Error 2
make[1]: Leaving directory '/home/user/applications/heads/build/x86/coreboot-4.13'
make: *** [modules/coreboot:106: /home/user/applications/heads/build/x86/coreboot-4.13/.xcompile] Error 2

```

t430-hotp-maximized build.log : https://privatebin.net/?5b4cd00951a608ab#6ve5y7zyLUz7Qt1jLmr8MKmtPzGmZ937pERNvzJ3iSkJ
[t430-bin-build.log](https://github.com/osresearch/heads/files/11147636/t430-bin-build.log)

I have a thinkpad t430 under debian 11 that I use to update the heads (I recompile each time) and everything is fine
So I think it must be a version problem (of Ada/gnat ? I don't know) but I'm not enough knowledgeable in Makefile to be able to find and propose a solution ...

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Download and install debian bookworm
2. install  the required packages sudo apt install -y build-essential zlib1g-dev uuid-dev libdigest-sha-perl libelf-dev bc bzip2 bison flex git gnupg gawk iasl m4 nasm patch python python2 python3 wget gnat cpio ccache pkg-config cmake libusb-1.0-0-dev autoconf texinfo ncurses-dev doxygen graphviz udev libudev1 libudev-dev automake libtool rsync innoextract sudo libssl-dev device-tree-compiler u-boot-tools sharutils e2fsprogs parted curl unzip
python is to replace by python-is-python3
python2 is to replace by 2to3
4. clone the repo, make=your_board
5. See error

**Expected behavior**
It should build the binary

**Additional context**
All is working under debian 11 I'll put the version of some of the software that are used to build heads
```
$gnat --version
GNAT 12.2.0
Copyright (C) 1996-2022, Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$nasm --version
NASM version 2.16.01

$gcc --version
gcc (Debian 12.2.0-14) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

tlaurion, 2023-04-04:
@123ahaha 

The sad story with gnat, and coreboot buildstack that heads builds so that coreboot is built with each coreboot buildstack, cannot be built with a newer gnat version that was provided in that buildstack.

What this means is that coreboot 4.13 (current xx20/xx30) will not be possible to build with bookworm (debian-12)
Just like coreboot 4.11 (kgpe-d16, librem1um) cannot be built with debian-11 and requires to be built with debian-10.

This will require patches, and might or may not work.

I'm surprised that librem14 cannot be built from a fresh heads checkout though, since I thought coreboot 4.17 could be built from debian-12.

@123ahaha Can you wipe install and build directories and try again for `make BOARD=librem14 V=1`?
Also, if you could upload your log.txt here instead of a file dropping service for which your file won't be availabe in less then 24h would be nice for prosperity.

tlaurion, 2023-04-04:
From the last linges of your log, you can see that you are attemtping to build gnat 11.3 from gnat 12. This is impossible.
Unfortunately, bookworm provides gnat 12.2 and gnat 11.3, none of which will be able to build an earlier version of it (11.2 for coreboot 4.17).
Src: https://packages.debian.org/search?suite=bookworm&searchon=names&keywords=gnat

The rule, for gnat, is that you can build it from a previous version, hence why it works for CircleCI which builds for roms from debian-11.

tlaurion, 2023-04-04:
This problem is really tiring.... 
The only short-mid-term solution would be to try to build coreboot buildstack from musl-cross-make, as all other modules.
The would need this path to be reinvestigated: https://www.openwall.com/lists/musl/2017/11/02/1

I do not remember why we stopped doing that in the past, where https://github.com/osresearch/heads/blob/a447674a893156a21407c6a2dc31c7e877292942/modules/coreboot#L78-L81
Is a trace from the era we last tried to do so....

123ahaha, 2023-04-04:
Thanks for the infos ! I didn't now about the gnat issue ...
I could help for the heads docker image server for free, but i'll need some specifcations however ...
Another fix could be instead of NixOS, juste use a flake.nix file ? as Nix package manager is easier to install than a whole nixos distro

tlaurion, 2023-04-04:
@123ahaha this issue is now linked to #1269 where if you have ideas, please join the discussion.
The problem is that NixOS is not providing gnat6 anymore, which is required to build librem1um and kgpe-d16 from coreboot 4.11 as said earlier.

I'm no NixOS expert. 
What I understood from upstream documentation would be to pin repository definition to a fixed commit. Then defined the OS packages requirements, the same way we do for debian-11. And then do as we do now. Build musl-cross-make, then build all modules, then build coreboot's version buildstack, and be happy for once and know that that work won't need to be done again. And then, maybe, I say maybe, work our way up to having musl-cross-make built for both PPC64 and x86. And then, maybe, use NixOS prepackaged coreboot buildstack for newer version.

But the point here, really, is that in our use case (CircleCI builds and local builds to users), once that buildstack is used, its reused, most of the time. Local builds will keep things there, where modules updates simply change directory names and builds will work forward in time. Same happens with CircleCI, where the cache system would just pick it up with a couple of tweaks, including whatever NixOS directories that were compiled/downloaded, and where we could hash things up there as we do now, with better guarantees that nothing bleeds into the builds, outside of time and dates. even rpath and other reproducibility inducing problems would be gone.

And at the end of the day, if we have to publish instructions to use docker to build Heads, that would be a lesser headach then now where people wants to build Heads on XYZ and where we say "use debian-11... for now" is not necessary a good solution since we know that it won't work 10 years from now :/

So again:
- gnat6 would need to be buildable under NixOS. That is the missing link today.

123ahaha, 2023-04-04:
Then I'll close the issue

closed 2023-04-04 by 123ahaha

tlaurion, 2023-04-04:
But. building x230 with coreboot 4.19 etc will work on bookworm. As will librem14.
But this is why CircleIC and roms are produced there as well: to promote a buildsystem that works for the moment until things transit to something else.

And this is not the whole story either. Under Heads, gnat is needed only if coreboot configuration enables LIBGFXINIT.
There is no reason, really under Heads, to have such support if we also pack the proper kernel drivers to drive vga console.

Food for thought. for the following platforms, if you uninstall gnat, you would be able to build those boards successfully:
```
user@heads-tests:~/heads$ grep GFX config/coreboot-*
config/coreboot-librem_l1um.config:CONFIG_NO_GFX_INIT=y
config/coreboot-t430-legacy-flash.config:CONFIG_NO_GFX_INIT=y
config/coreboot-x220.config:CONFIG_NO_GFX_INIT=y
config/coreboot-x220-maximized.config:CONFIG_NO_GFX_INIT=y
config/coreboot-x230-legacy.config:CONFIG_NO_GFX_INIT=y
config/coreboot-x230-legacy-flash.config:CONFIG_NO_GFX_INIT=y
config/coreboot-x230-maximized.config:CONFIG_NO_GFX_INIT=y
```