linuxboot/heads #1366: Collaborate with ownerboot

tlaurion, 2023-04-05:
Ownerboot needs a payload.  @amjoseph-nixpkgs ?

amjoseph-nixpkgs, 2023-04-24:
Apologies for the slow reply!

The subject line was a bit vague so this settled near the bottom of my to-do list.

I knew for a long time that "heads is the thing you use if you want coreboot for qubes" but incorrectly assumed you were doing the whole SeaBIOS-UEFI-grub dance, which I try pretty hard to avoid.  I apologize for that.  At first glance it appears that you guys build an `initrd` with a boot menu that `kexec()`s into the primary operating system.

Let me look at this more closely and see if ownerboot can boot that intird instead of the `init=/bin/sh` stand-in that is currently used.

tlaurion, 2023-08-27:
@amjoseph-nixpkgs are you on matrix? Poke me @(insurgo:matrix.org)

tlaurion, 2024-05-25:
Ha. Gonna be more complicated talking async with @amjoseph-nixpkgs  since not on github anymore https://github.com/NixOS/nixpkgs/issues/282862

closed 2025-08-25 by tlaurion