linuxboot/heads #1376: cryptsetup error debian

xalatore, 2023-04-17:
I just installed debian-live-11.6.0-amd64-xfce.iso on my Thinkpad x230 (Heads and QubesOS were already on there (self-installed)).
During installation I selected manual partitioning, deleted the old partitions (QubesOs) and set up a 1gib ext4 partion mounted at boot (unencrypted) and another big partion with luks encryption mounted at / (like it is recommended on the osresearch.net page under Install OSes/Compatibility).
During install I entered my new luks password.
Install went fine without errors.

Afterwards I upgraded heads internally to heads-x230-maximized-v0.2.0-1523-g8ff4b9a.rom with the option erase settings.
Everything went fine, and afterwards I was greeted with the OEM Factory Reset screen and followed the instructions to set everything up, new TPM, new GPG Keys. I use a Yubikey. Setup a new TOTP Secret as well.

Heads is working great, everything fine.

When I boot Debian it gives me the following error:

cryptsetup: ERROR: luks-769d1276-e51c-4556-8a69-ca13818d96d2: cryptsetup failed, bad password or options?

and also cryptsetup ....: maximum number of tries exceeded

and then drops me into initramfs..

Where did I go wrong?

I would appreciate any help!

Thanks a lot in advance!

tlaurion, 2023-04-17:
@xalatore hello there!

Well, all leads to a wrongly typed disk recovery passphrase? You seem to not use heads TPM released disk encryption key and relying on OS passphrase prompt to type decryption passphrase. Nothing wrong with that per se.

From heads recovery shell, you could
cryptsetup luksOpen /dev/sda2 test

And see if you can unlock the FDE luks partition as a start.

Only hypothesis here is maybe a keyboard mapping mismatch. Or again being in capslock (x230 doesn't have a LED for that).

tlaurion, 2023-04-17:
Reboot and try again. Gut feeling is about wrongly typed passphrase or capslock really. Your test relying on Debian only with proper partitioning, it seems that the passphrase typed is simply wrong.

xalatore, 2023-04-20:
Sorry for late reply!
Indeed keyboard mismatch, can be closed!

Thank you!

closed 2023-04-20 by xalatore