linuxboot/heads #1383: QubesOS public key

naive-HA, 2023-04-22:
## Please identify some basic details to help process the report

### A. Provide Hardware Details

**1. What board are you using (see list of boards [here](https://github.com/eganonoa/heads/tree/master/boards))?**
T420
**2. Does your computer have a dGPU or is it iGPU-only?**
- [ ] dGPU
- [ ] iGPU-only

**3. Who installed Heads on this computer?**
- [ ] Insurgo
- [ ] Nitrokey
- [ ] Purism
- [ ] Other provider
- [ x] Self-installed

**4. What PGP key is being used?**
- [ ] Librem Key
- [ x] Nitrokey Pro 2
- [ ] Nitrokey Storage
- [ ] Yubikey
- [ ] Other

**5. Are you using the PGP key to provide HOTP verification?**
- [ x] Yes
- [ ] No
- [ ] I don't know

### B. Identify how the board was flashed

**1. Is this problem related to updating heads or flashing it for the first time?**
- [ x] First-time flash
- [ ] Updating heads 

**2. If the problem is related to an update, how did you attempt to apply the update?**
- [ ] Using the Heads GUI
- [ ] Flashrom via the Recovery Shell
- [x ] External flashing

**3. How was Heads initially flashed**
- [ x] External flashing
- [ ] Internal-only / 1vyrain
- [ ] Don't know

**4. Was the board flashed with a maximized or non-maximized/legacy rom?**
- [ x] Maximized
- [ ] Non-maximized / legacy
- [ ] I don't know

**5. If Heads was externally flashed, was IFD unlocked?**
- [ ] Yes
- [ ] No
- [ x] Don't know

### C. Identify the rom related to this bug report

**1. Did you download or build the rom at issue in this bug report?**
- [x ] I downloaded it
- [ ] I built it

**2. If you downloaded your rom, where did you get it from?**
- [x ] Heads CircleCi
- [ ] Purism
- [ ] Nitrokey
- [ ] Somewhere else (please identify)

*Please provide the release number or otherwise identify the rom downloaded*
heads-t420-maximized-v.0.2.0-1540-g400f714d.rom

**3. If you built your rom, which repository:branch did you use?**
- [ ] Heads:Master
- [ ] Other (please identify)

**4. What version of coreboot did you use in building?**
- [ ] 4.8.1 (current default in heads:master)
- [ ] 4.13
- [ ] 4.14
- [ ] 4.15
- [ ] Other (please specify)
- [ ] I don't know

**5. In building the rom where did you get the blobs?**
- [ ] No blobs required
- [ ] Provided by the company that installed Heads on the device
- [ ] Extracted from a backup rom taken from this device
- [ ] Extracted from another backup rom taken from another device (please identify the board model)
- [ ] Extracted from the online bios using the automated tools provided in Heads
- [ ] I don't know

## Please describe the problem

**Describe the bug**
A clear and concise description of what the bug is.
Tried to install QubesOS for the first time and upon iso verifcation got error: 
can't check signature: no public key
Should I provide the public key? Is it suposed to be fused into heads?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen
I was expecting the public key to be fused into heads. Am I wrong?

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

naive-HA, 2023-04-22:
[image]

tlaurion, 2023-04-22:
@naive-HA can you re-upload picture in last comment?

naive-HA, 2023-04-22:
[image]

naive-HA, 2023-04-22:
https://drive.google.com/file/d/1IZFKLM_s9fVTzPVYGvS7BHN0M0W-dFRC/view

tlaurion, 2023-04-22:
@naive-HA i will have to dowload 4.2 ISO ans asc file and reproduce.

But if I understand correctly, it should work and its a bug.

mount -o remount,rw /media
Do your changes
mount -o remount,ro /media 

Should get around your trace of attempts that were failing here.

naive-HA, 2023-04-23:
use 4.1.2 iso and asc

naive-HA, 2023-04-23:
there is something wrong with pubring and trustdb in etc distro
i have deleted both, imported qubes-4 key from etc distro keys, copied pubring and trustdb from //.gnupg to etc distro and managed to start qubes installer

tlaurion, 2023-04-23:
Repro notes. Will edit

Downloading ISO and detached signature:
- https://mirrors.edge.kernel.org/qubes/iso/Qubes-R4.1.2-x86_64.iso
- https://mirrors.edge.kernel.org/qubes/iso/Qubes-R4.1.2-x86_64.iso.asc

Public signing key shouldbe equal to online:
- https://keys.qubes-os.org/keys/qubes-release-4-signing-key.asc

tlaurion, 2023-04-23:
https://github.com/osresearch/heads/issues/1383#issuecomment-1518630973 doesn't include valid picture. (You posted comment when image was uploading)

Second picture is not showing anymore.

Please upload, through github, a screenshot of when its failing and a date output, from recovery shell.

As shown above, calling usb-init from recovery shell will give proper traces (without debug since not enabled under your board config) but not finding valid public key is not right. System clock has to be wrong or something else.

naive-HA, 2023-04-23:
I managed to install qubes os as detailed above.
I will reflash and I will post a new screenshot

naive-HA, 2023-04-24:
yes, my bad. I have made an oopsie
it seems that during the external flashing I have messed some things up. The hwclock was off by a lot.
I have updated with
date -s '2023-04-24 08:00:00'
hwclock -w
The hwclock survived a reboot, flashed internally heads-t420-maximized-v0.2.0-1540-g400f71d, initiated a new qubes os install and the verification process succeeded.

However, the behaviour ia a bit confusing when hwclock is off. With hwclock messed up I have managed to trigger the installation process by recreating the files in etc distro...
anyway: sorry if I wasted your time
I am continuing here with the fun: it seems that external flashing messed up more than the hwclock: qubes os says iommu is not enabled...

closed 2023-04-24 by naive-HA