linuxboot/heads #1387: x230-fhd/edp variant now under coreboot 4.19 : UNTESTED

tlaurion, 2023-04-25:
@n4ru @computer-user123 @Tonux599 @househead

Can you please test latest rom https://output.circle-artifacts.com/output/job/82bedd7c-74ac-479b-bd4a-7a1ada8da664/artifacts/0/build/x86/x230-hotp-maximized-fhd_edp/heads-x230-hotp-maximized-fhd_edp-v0.2.0-1529-g3a38ac0.rom

Originating from merge of PR where your feedback was requested a while ago at https://github.com/osresearch/heads/pull/1312#issuecomment-1432062401

(This patch and board under Heads has been created for you guys!)
Also note that some participation under https://review.coreboot.org/c/coreboot/+/28950. Better from board owners then from me (where past comments have not led to a merge upstream, meaning I would have to maintain this, which I have no intention of doing so).

Pinning this issue.
Please test only if you have an external reprogrammer to unbrick if needed.

computer-user123, 2023-04-25:
@tlaurion  : Thank you!!! Will get on this asap!
Appreciate your commitment to this project!

I can test x230-FHD-eDP(nitro caster board) 
Hopefully today or tomorrow.
Will also w530 / & w530 k2000 . (I prefer to not use dGPU , for my uses, less heat and power draw are benefits).

pcm720, 2023-04-29:
Built myself a ROM from ab1faf5389dbfbc174e3cbf6cdec0be1cba14772.
Can confirm that it works great on a X230 freshly modded with eDP 4.0 board and 1440p display.

Thanks!

tlaurion, 2023-04-29:
@computer-user123 thanks for testing this! Closing issue!

closed 2023-04-29 by tlaurion

tlaurion, 2023-07-19:
Updated https://github.com/osresearch/heads/issues/692 w530 and x230-edp board owner information to ease tracking of this, also to #1444 and upstreaming under coreboot (no way I take a long life commitment in maintaining this downstream)