linuxboot/heads #1391: Mention in the wiki to reset CMOS after a flash

underdogest, 2023-04-30:
**Is your feature request related to a problem? Please describe.**
Spent 2 hours pinpointing the problem when I just had to reset the CMOS via removing the battery.

**Describe the solution you'd like**
Just put it in the wiki.

**Additional context**
1) Flashed a x230 from legacy to maximized internaly: black screen on startup
2) Flashed the same rom externaly: black screen on startup
3) Flashed multiple tags externaly and finaly reset CMOS: normal startup

tlaurion, 2023-05-01:
@underdogest Sorry you had this issue.
Do you change CMOS settings? Should they be hardcoded firmware to make sure such settings are not causing a brick?

Questioning in a desire to unify things here. Users of t430 and other boards with gpu are told pr board configuration notes to apply some tweaks on the rom itself to change cmos settings. Maybe this should be unified?

underdogest, 2023-05-01:
I did not change any CMOS defaults when building and afaik heads doesnt even use anything there apart from the RTC. Maybe the GPU read something from there?

Anyway, I think it would be best if heads mentioned clearing CMOS via battery removal after internal flashing and in the wiki, so users who experience an apparent brick try this first.