linuxboot/heads #1392: 5.10.5 kernel build failure under Debian bookworm

srgrint, 2023-05-01:
## Context of the Build

**1. What board are you trying to build?**
T440p
**2. What repository:branch are you using to build from?**
- [ X] Heads:Master
- [ ] Other (please specify)

**3. What version of coreboot are you trying to build**
- [ ] 4.13
- [ ] 4.14
- [ ] 4.15
- [X ] 4.17
- [ ] Other (please specify)

**4. In building the rom where did you get the blobs?**
- [ ] No blobs required
- [ ] Provided by the company that installed Heads on the device
- [ ] Extracted from a backup rom taken from this device
- [ ] Extracted from another backup rom taken from another device (please identify the board model)
- [ X] Extracted from the online bios using the automated tools provided in Heads
- [ ] I don't know

**5. If using the automated tools to get the blobs did you run the relevant scripts in the blobs directory**
- [X ] Yes
- [ ] No

**6. What operating system are you using**
Debian bookworm/12

Have tried building the t440p-maximized rom using my test install of Debian 12 (bookworm) which ships with GCC 12

Once I have built the toolchain without gnat support, the build of the 5.10.5 kernel fails due to finding a use after free (due to the new compiler being more aware than what shipped with Debian 11).  The bug has already been fixed upstream last year - if I apply the following patch, heads builds as expected:

https://github.com/torvalds/linux/commit/52a9dab6d892763b2a8334a568bd4e2c1a6fde66

I don't know whether you want to cherry pick this into heads?

tlaurion, 2023-05-01:
@srgrint linux 5.10.178 (LTS) has the fix backported.

@JonathonHall-Purism you might want to backport referred patch, where my helper to prompt devel (who will bump kernel version) works really sweetly under debugging branch.

Note that bookworm on builder host is not officially supported as of now.

I'm personally tired of supporting multiple hosts buildstacks, where goal would be to have Nix layer on top of any linux distrib to fix this as well as building under docker image, which is more interesting on a maintainership perspective. 

As of now, it is recommended to build on top of debian-10.

srgrint, 2023-05-01:
> I'm personally tired of supporting multiple hosts buildstacks, where goal would be to have Nix layer on top of any linux distrib to fix this as well as building under docker image, which is more interesting on a maintainership perspective.

Completely agree - Nix layer is definitely the way forward.

The use-after-free kernel bug is real though - just Debian 10 buildstack isn't astute enough to catch it

tlaurion, 2023-05-01:
> Once I have built the toolchain without gnat support, the build of the 5.10.5 kernel fails due to finding a use after free (due to the new compiler being more aware than what shipped with Debian 11). The bug has already been fixed upstream last year - if I apply the following patch, heads builds as expected:
> 
> [torvalds/linux@52a9dab](https://github.com/torvalds/linux/commit/52a9dab6d892763b2a8334a568bd4e2c1a6fde66)
> 
> I don't know whether you want to cherry pick this into heads?

Right. Bug is real and currently affects Purism and t440p as of now. Will refer PR to this comment. Thanks.

srgrint, 2023-05-02:
Thanks - not sure whether you wished to also fix boards currently using 4.14.62 which is also affected.

Clearly we want to be bumping all boards to 5.10 in the medium term.  Have opened a PR  (#1394) in case you want to fix 4.14 branch in the interim.

closed 2023-05-03 by tlaurion