linuxboot/heads #1396: Review kernel crypto backend requirements

tlaurion, 2023-05-03:
I recompiled the kernel with the following options set, cryptsetup is working fine now:
```
CONFIG_CRYPTO_XTS=y
CONFIG_CRYPTO_AES=y
CONFIG_CRYPTO_AES_TI=y
CONFIG_CRYPTO_AES_NI_INTEL=y
CONFIG_CRYPTO_USER_API_HASH=y
CONFIG_CRYPTO_USER_API_SKCIPHER=y
```
Can confirm that other than that, there has been no regressions for me on 5.10.178.

_Originally posted by @pcm720 in https://github.com/osresearch/heads/issues/958#issuecomment-1530342983_

tlaurion, 2023-05-03:
https://www.suse.com/support/update/announcement/2020/suse-ru-20200952-1/

Clarifies that:

> Update from version 2.0.5 to version 2.0.6 (jsc#SLE-5911, bsc#1165580): * Fix support of larger metadata areas in LUKS2 header. This release properly supports all specified metadata areas, as documented in LUKS2 format description (see docs/on-disk-format-luks2.pdf in archive). Currently, only default metadata area size is used (in format or convert). Later cryptsetup versions will allow increasing this metadata area size. * If AEAD (authenticated encryption) is used, cryptsetup now tries to check if the requested AEAD algorithm with specified key size is available in kernel crypto API. This change avoids formatting a device that cannot be later activated. For this function, the kernel must be compiled with the CONFIG_CRYPTO_USER_API_AEAD option enabled. Note that kernel user crypto API options (CONFIG_CRYPTO_USER_API and CONFIG_CRYPTO_USER_API_SKCIPHER) are already mandatory for LUKS2. * Fix setting of integrity no-journal flag. Now you can store this flag to metadata using --persistent option. * Fix cryptsetup-reencrypt to not keep temporary reencryption headers if interrupted during initial password prompt. * Adds early check to plain and LUKS2 formats to disallow device format if device size is not aligned to requested sector size. Previously it was possible, and the device was rejected to activate by kernel later. * Fix checking of hash algorithms availability for PBKDF early. Previously LUKS2 format allowed non-existent hash algorithm with invalid keyslot preventing the device from activation. * Allow Adiantum cipher construction (a non-authenticated length-preserving fast encryption scheme), so it can be used both for data encryption and keyslot encryption in LUKS1/2 devices. For benchmark, use: # cryptsetup benchmark -c xchacha12,aes-adiantum # cryptsetup benchmark -c xchacha20,aes-adiantum For LUKS format: # cryptsetup luksFormat -c xchacha20,aes-adiantum-plain64 -s 256 The support for Adiantum will be merged in Linux kernel 4.21. For more info see the paper https://eprint.iacr.org/2018/720.

tlaurion, 2023-05-03:
origin linux-qemu has been changed to include nothing.
final config is enough to run benchmark and see what can be used from crypto backend vs what is desired.

```
diff --git a/config/linux-qemu.config b/config/linux-qemu.config
index ac638a4316..042df1b7e0 100644
--- a/config/linux-qemu.config
+++ b/config/linux-qemu.config
@@ -2824,6 +2824,8 @@ CONFIG_CRYPTO_NULL2=y
 CONFIG_CRYPTO_CRYPTD=y
 CONFIG_CRYPTO_AUTHENC=y
 # CONFIG_CRYPTO_TEST is not set
+CONFIG_CRYPTO_SIMD=y
+CONFIG_CRYPTO_GLUE_HELPER_X86=y
 
 #
 # Public-key cryptography
@@ -2911,9 +2913,9 @@ CONFIG_CRYPTO_SHA512=y
 #
 # Ciphers
 #
-# CONFIG_CRYPTO_AES is not set
-# CONFIG_CRYPTO_AES_TI is not set
-# CONFIG_CRYPTO_AES_NI_INTEL is not set
+CONFIG_CRYPTO_AES=y
+CONFIG_CRYPTO_AES_TI=y
+CONFIG_CRYPTO_AES_NI_INTEL=y
 # CONFIG_CRYPTO_BLOWFISH is not set
 # CONFIG_CRYPTO_BLOWFISH_X86_64 is not set
 # CONFIG_CRYPTO_CAMELLIA is not set
@@ -2956,16 +2958,19 @@ CONFIG_CRYPTO_SHA512=y
 # CONFIG_CRYPTO_ANSI_CPRNG is not set
 # CONFIG_CRYPTO_DRBG_MENU is not set
 # CONFIG_CRYPTO_JITTERENTROPY is not set
-# CONFIG_CRYPTO_USER_API_HASH is not set
-# CONFIG_CRYPTO_USER_API_SKCIPHER is not set
+CONFIG_CRYPTO_USER_API=y
+CONFIG_CRYPTO_USER_API_HASH=y
user@heads-tests:~/heads$ git diff > crypto
user@heads-tests:~/heads$ cat crypto
diff --git a/config/linux-qemu.config b/config/linux-qemu.config
index ac638a4316..042df1b7e0 100644
--- a/config/linux-qemu.config
+++ b/config/linux-qemu.config
@@ -2824,6 +2824,8 @@ CONFIG_CRYPTO_NULL2=y
 CONFIG_CRYPTO_CRYPTD=y
 CONFIG_CRYPTO_AUTHENC=y
 # CONFIG_CRYPTO_TEST is not set
+CONFIG_CRYPTO_SIMD=y
+CONFIG_CRYPTO_GLUE_HELPER_X86=y
 
 #
 # Public-key cryptography
@@ -2911,9 +2913,9 @@ CONFIG_CRYPTO_SHA512=y
 #
 # Ciphers
 #
-# CONFIG_CRYPTO_AES is not set
-# CONFIG_CRYPTO_AES_TI is not set
-# CONFIG_CRYPTO_AES_NI_INTEL is not set
+CONFIG_CRYPTO_AES=y
+CONFIG_CRYPTO_AES_TI=y
+CONFIG_CRYPTO_AES_NI_INTEL=y
 # CONFIG_CRYPTO_BLOWFISH is not set
 # CONFIG_CRYPTO_BLOWFISH_X86_64 is not set
 # CONFIG_CRYPTO_CAMELLIA is not set
@@ -2956,16 +2958,19 @@ CONFIG_CRYPTO_SHA512=y
 # CONFIG_CRYPTO_ANSI_CPRNG is not set
 # CONFIG_CRYPTO_DRBG_MENU is not set
 # CONFIG_CRYPTO_JITTERENTROPY is not set
-# CONFIG_CRYPTO_USER_API_HASH is not set
-# CONFIG_CRYPTO_USER_API_SKCIPHER is not set
+CONFIG_CRYPTO_USER_API=y
+CONFIG_CRYPTO_USER_API_HASH=y
+CONFIG_CRYPTO_USER_API_SKCIPHER=y
 # CONFIG_CRYPTO_USER_API_RNG is not set
-# CONFIG_CRYPTO_USER_API_AEAD is not set
+CONFIG_CRYPTO_USER_API_AEAD=y
+# CONFIG_CRYPTO_USER_API_ENABLE_OBSOLETE is not set
 # CONFIG_CRYPTO_STATS is not set
 CONFIG_CRYPTO_HASH_INFO=y
 
 #
 # Crypto library routines
 #
+CONFIG_CRYPTO_LIB_AES=y
 # CONFIG_CRYPTO_LIB_BLAKE2S is not set
 # CONFIG_CRYPTO_LIB_CHACHA is not set
 # CONFIG_CRYPTO_LIB_CURVE25519 is not set

```

------
```

~ # cryptsetup benchmark
# Tests are approximate using memory only (no storage IO).
PBKDF2-sha1          N/A
PBKDF2-sha256        N/A
PBKDF2-sha512        N/A
PBKDF2-ripemd160     N/A
PBKDF2-whirlpool     N/A
argon2i       4 iterations, 13774 memory, 4 parallel threads (CPUs) for 256-bit key (requested 2000 ms time)
argon2id      4 iterations, 13924 memory, 4 parallel threads (CPUs) for 256-bit key (requested 2000 ms time)
[  164.596922] random: cryptsetup: uninitialized urandom read (16 bytes read)
[  164.598969] random: cryptsetup: uninitialized urandom read (16 bytes read)
#     Algorithm |       Key |      Encryption |      Decryption
        aes-cbc        128b        10.0 MiB/s         8.2 MiB/s
[  166.707187] random: cryptsetup: uninitialized urandom read (16 bytes read)
[  166.711022] random: cryptsetup: uninitialized urandom read (16 bytes read)
    serpent-cbc        128b               N/A               N/A
[  166.718407] random: cryptsetup: uninitialized urandom read (16 bytes read)
    twofish-cbc        128b               N/A               N/A
        aes-cbc        256b         7.9 MiB/s         6.2 MiB/s
[  168.875203] urandom_read: 3 callbacks suppressed
[  168.875266] random: cryptsetup: uninitialized urandom read (16 bytes read)
    serpent-cbc        256b               N/A               N/A
    twofish-cbc        256b               N/A               N/A
        aes-xts        256b               N/A               N/A
    serpent-xts        256b               N/A               N/A
    twofish-xts        256b               N/A               N/A
        aes-xts        512b               N/A               N/A
    serpent-xts        512b               N/A               N/A
    twofish-xts        512b               N/A               N/A
```

tlaurion, 2023-05-03:
compared to above:
```
diff --git a/config/linux-qemu.config b/config/linux-qemu.config
index b53d154437..109bfea95a 100644
--- a/config/linux-qemu.config
+++ b/config/linux-qemu.config
@@ -2860,7 +2860,7 @@ CONFIG_CRYPTO_ECB=y
 # CONFIG_CRYPTO_LRW is not set
 # CONFIG_CRYPTO_OFB is not set
 # CONFIG_CRYPTO_PCBC is not set
-# CONFIG_CRYPTO_XTS is not set
+CONFIG_CRYPTO_XTS=y
 # CONFIG_CRYPTO_KEYWRAP is not set
 # CONFIG_CRYPTO_NHPOLY1305_SSE2 is not set
 # CONFIG_CRYPTO_NHPOLY1305_AVX2 is not set
```

```
# cryptsetup benchmark
# Tests are approximate using memory only (no storage IO).
PBKDF2-sha1          N/A
PBKDF2-sha256        N/A
PBKDF2-sha512        N/A
PBKDF2-ripemd160     N/A
PBKDF2-whirlpool     N/A
argon2i       4 iterations, 13260 memory, 4 parallel threads (CPUs) for 256-bit key (requested 2000 ms time)
argon2id      4 iterations, 13599 memory, 4 parallel threads (CPUs) for 256-bit key (requested 2000 ms time)
[  106.196295] random: cryptsetup: uninitialized urandom read (16 bytes read)
[  106.200119] random: cryptsetup: uninitialized urandom read (16 bytes read)
#     Algorithm |       Key |      Encryption |      Decryption
        aes-cbc        128b         7.4 MiB/s         8.0 MiB/s
[  108.412820] random: cryptsetup: uninitialized urandom read (16 bytes read)
[  108.416666] random: cryptsetup: uninitialized urandom read (16 bytes read)
    serpent-cbc        128b               N/A               N/A
[  108.425426] random: cryptsetup: uninitialized urandom read (16 bytes read)
    twofish-cbc        128b               N/A               N/A
        aes-cbc        256b         5.8 MiB/s         6.2 MiB/s
[  110.607983] urandom_read: 3 callbacks suppressed
[  110.608054] random: cryptsetup: uninitialized urandom read (16 bytes read)
[  110.612615] random: cryptsetup: uninitialized urandom read (32 bytes read)
    serpent-cbc        256b               N/A               N/A
    twofish-cbc        256b               N/A               N/A
        aes-xts        256b         7.5 MiB/s         8.0 MiB/s
    serpent-xts        256b               N/A               N/A
    twofish-xts        256b               N/A               N/A
        aes-xts        512b         5.9 MiB/s         6.1 MiB/s
    serpent-xts        512b               N/A               N/A
    twofish-xts        512b               N/A               N/A
```

tlaurion, 2023-05-03:
For good measure, diff from previous config, tuned for up to ivybridge (No AVX2 here : newer platforms should enable it. Here SSE2+AVX cpu acceleration enabled):
```
user@heads-tests:~/heads$ git diff
diff --git a/config/linux-qemu.config b/config/linux-qemu.config
index 109bfea95a..607887bca6 100644
--- a/config/linux-qemu.config
+++ b/config/linux-qemu.config
@@ -2932,15 +2932,16 @@ CONFIG_CRYPTO_AES_NI_INTEL=y
 # CONFIG_CRYPTO_SALSA20 is not set
 # CONFIG_CRYPTO_CHACHA20 is not set
 # CONFIG_CRYPTO_CHACHA20_X86_64 is not set
-# CONFIG_CRYPTO_SERPENT is not set
-# CONFIG_CRYPTO_SERPENT_SSE2_X86_64 is not set
-# CONFIG_CRYPTO_SERPENT_AVX_X86_64 is not set
+CONFIG_CRYPTO_SERPENT=y
+CONFIG_CRYPTO_SERPENT_SSE2_X86_64=y
+CONFIG_CRYPTO_SERPENT_AVX_X86_64=y
 # CONFIG_CRYPTO_SERPENT_AVX2_X86_64 is not set
 # CONFIG_CRYPTO_SM4 is not set
 # CONFIG_CRYPTO_TWOFISH is not set
-# CONFIG_CRYPTO_TWOFISH_X86_64 is not set
-# CONFIG_CRYPTO_TWOFISH_X86_64_3WAY is not set
-# CONFIG_CRYPTO_TWOFISH_AVX_X86_64 is not set
+CONFIG_CRYPTO_TWOFISH_COMMON=y
+CONFIG_CRYPTO_TWOFISH_X86_64=y
+CONFIG_CRYPTO_TWOFISH_X86_64_3WAY=y
+CONFIG_CRYPTO_TWOFISH_AVX_X86_64=y
 
 #
 # Compression
```

```
~ # cryptsetup benchmark
# Tests are approximate using memory only (no storage IO).
PBKDF2-sha1          N/A
PBKDF2-sha256        N/A
PBKDF2-sha512        N/A
PBKDF2-ripemd160     N/A
PBKDF2-whirlpool     N/A
argon2i       4 iterations, 13321 memory, 4 parallel threads (CPUs) for 256-bit key (requested 2000 ms time)
argon2id      4 iterations, 13670 memory, 4 parallel threads (CPUs) for 256-bit key (requested 2000 ms time)
[  133.376027] random: cryptsetup: uninitialized urandom read (16 bytes read)
[  133.380390] random: cryptsetup: uninitialized urandom read (16 bytes read)
#     Algorithm |       Key |      Encryption |      Decryption
        aes-cbc        128b         9.8 MiB/s         7.5 MiB/s
[  135.478067] random: cryptsetup: uninitialized urandom read (16 bytes read)
[  135.482054] random: cryptsetup: uninitialized urandom read (16 bytes read)
    serpent-cbc        128b        15.1 MiB/s         4.0 MiB/s
[  137.807642] random: cryptsetup: uninitialized urandom read (16 bytes read)
[  137.810037] random: cryptsetup: uninitialized urandom read (16 bytes read)
    twofish-cbc        128b        19.1 MiB/s        10.7 MiB/s
[  139.885919] random: cryptsetup: uninitialized urandom read (16 bytes read)
[  139.889818] random: cryptsetup: uninitialized urandom read (32 bytes read)
        aes-cbc        256b         7.7 MiB/s         5.7 MiB/s
[  141.992113] random: cryptsetup: uninitialized urandom read (16 bytes read)
[  141.994108] random: cryptsetup: uninitialized urandom read (32 bytes read)
    serpent-cbc        256b        15.0 MiB/s         3.6 MiB/s
    twofish-cbc        256b        17.9 MiB/s        10.6 MiB/s
        aes-xts        256b         9.7 MiB/s         7.3 MiB/s
    serpent-xts        256b         3.9 MiB/s         3.9 MiB/s
    twofish-xts        256b        10.4 MiB/s        10.5 MiB/s
        aes-xts        512b         7.8 MiB/s         5.5 MiB/s
    serpent-xts        512b         3.9 MiB/s         3.9 MiB/s
    twofish-xts        512b        10.6 MiB/s        10.5 MiB/s
```

tlaurion, 2023-05-03:
So here is a reference for cryptsetup requirements as of now (AVX2/SSE3):

```
user@heads-tests:~/heads$ grep CRYPTO config/linux-qemu.config | grep "=y"
CONFIG_CRYPTO=y
CONFIG_CRYPTO_ALGAPI=y
CONFIG_CRYPTO_ALGAPI2=y
CONFIG_CRYPTO_AEAD=y
CONFIG_CRYPTO_AEAD2=y
CONFIG_CRYPTO_SKCIPHER=y
CONFIG_CRYPTO_SKCIPHER2=y
CONFIG_CRYPTO_HASH=y
CONFIG_CRYPTO_HASH2=y
CONFIG_CRYPTO_RNG2=y
CONFIG_CRYPTO_AKCIPHER2=y
CONFIG_CRYPTO_KPP2=y
CONFIG_CRYPTO_ACOMP2=y
CONFIG_CRYPTO_MANAGER=y
CONFIG_CRYPTO_MANAGER2=y
CONFIG_CRYPTO_USER=y
CONFIG_CRYPTO_MANAGER_DISABLE_TESTS=y
CONFIG_CRYPTO_NULL=y
CONFIG_CRYPTO_NULL2=y
CONFIG_CRYPTO_CRYPTD=y
CONFIG_CRYPTO_AUTHENC=y
CONFIG_CRYPTO_SIMD=y
CONFIG_CRYPTO_GLUE_HELPER_X86=y
CONFIG_CRYPTO_CBC=y
CONFIG_CRYPTO_ECB=y
CONFIG_CRYPTO_XTS=y
CONFIG_CRYPTO_ESSIV=y
CONFIG_CRYPTO_CRC32C=y
CONFIG_CRYPTO_CRC32C_INTEL=y
CONFIG_CRYPTO_MD5=y
CONFIG_CRYPTO_SHA1=y
CONFIG_CRYPTO_SHA1_SSSE3=y
CONFIG_CRYPTO_SHA256_SSSE3=y
CONFIG_CRYPTO_SHA512_SSSE3=y
CONFIG_CRYPTO_SHA256=y
CONFIG_CRYPTO_SHA512=y
CONFIG_CRYPTO_AES=y
CONFIG_CRYPTO_AES_TI=y
CONFIG_CRYPTO_AES_NI_INTEL=y
CONFIG_CRYPTO_SERPENT=y
CONFIG_CRYPTO_SERPENT_SSE2_X86_64=y
CONFIG_CRYPTO_SERPENT_AVX_X86_64=y
CONFIG_CRYPTO_SERPENT_AVX2_X86_64=y
CONFIG_CRYPTO_TWOFISH_COMMON=y
CONFIG_CRYPTO_TWOFISH_X86_64=y
CONFIG_CRYPTO_TWOFISH_X86_64_3WAY=y
CONFIG_CRYPTO_TWOFISH_AVX_X86_64=y
CONFIG_CRYPTO_USER_API=y
CONFIG_CRYPTO_USER_API_HASH=y
CONFIG_CRYPTO_USER_API_SKCIPHER=y
CONFIG_CRYPTO_USER_API_AEAD=y
CONFIG_CRYPTO_HASH_INFO=y
CONFIG_CRYPTO_LIB_AES=y
CONFIG_CRYPTO_LIB_SHA256=y

```

tlaurion, 2023-05-06:
Where the following van ne set to n

>CONFIG_CRYPTO_AES=n
>CONFIG_CRYPTO_AES_TI=n

If
> CONFIG_CRYPTO_AES_NI_INTEL=y

On Intel platforms

pcm720, 2023-05-07:
```
PBKDF2-sha1          N/A
PBKDF2-sha256        N/A
PBKDF2-sha512        N/A
```

This turns out to be a significant problem for disks using PBKDF2 as digest.
Just found out that I couldn't add a key to TPM nor mount the drive with cryptsetup on Heads based on 5.10 kernel.

tlaurion, 2023-05-07:
@pcm720 what os? What would it need?

pcm720, 2023-05-07:
Initial encryption was set up with Arch Linux, but the cryptsetup I'm trying it on is Heads' embedded binary.
As for what kernel config options it needs to work with PBKDF2 — I'm still trying to figure that out.
There's surprisingly little information on that, I'm currently looking at a diff between 4.19 and 5.10, but there's nothing obvious there.
Everything seems to imply that enabling digest options (CONFIG_CRYPTO_SHA1/256/512) should be enough for cryptsetup to do its job, but as we can see, this is not the case.

tlaurion, 2023-05-07:
Yeah, rhat ils what i figured out as well un staging PR

tlaurion, 2023-05-07:
@pcm720 if you want to test ROM produced by CI and report results there, that would be much appreciated.

tlaurion, 2023-05-08:
Will review once again and add fastest options under shared linux-x230-legacy and linux-x230-maximized as well. If CPU accelerated options supported under sandy/ivy, it should be activated here, and then cleaned up later on on separated board configs if space is a problem later on. Also to note that linux-x230-* configs should probably renamed to linux-xx30 for clarity (since for example, haswell+ has avx2 which sandy/ivy doesn't)

tlaurion, 2023-05-08:
So this is the current crypto backend requirements for current use case of Heads (cryptsetup and gnupg)
```
user@heads-tests:~/heads$ grep "CRYPTO" config/linux-x230-maximized.config | grep "=y"
CONFIG_CRYPTO=y
CONFIG_CRYPTO_ALGAPI=y
CONFIG_CRYPTO_ALGAPI2=y
CONFIG_CRYPTO_AEAD=y
CONFIG_CRYPTO_AEAD2=y
CONFIG_CRYPTO_SKCIPHER=y
CONFIG_CRYPTO_SKCIPHER2=y
CONFIG_CRYPTO_HASH=y
CONFIG_CRYPTO_HASH2=y
CONFIG_CRYPTO_RNG=y
CONFIG_CRYPTO_RNG2=y
CONFIG_CRYPTO_AKCIPHER2=y
CONFIG_CRYPTO_KPP2=y
CONFIG_CRYPTO_ACOMP2=y
CONFIG_CRYPTO_MANAGER=y
CONFIG_CRYPTO_MANAGER2=y
CONFIG_CRYPTO_USER=y
CONFIG_CRYPTO_MANAGER_DISABLE_TESTS=y
CONFIG_CRYPTO_NULL=y
CONFIG_CRYPTO_NULL2=y
CONFIG_CRYPTO_CRYPTD=y
CONFIG_CRYPTO_AUTHENC=y
CONFIG_CRYPTO_SIMD=y
CONFIG_CRYPTO_GLUE_HELPER_X86=y
CONFIG_CRYPTO_CBC=y
CONFIG_CRYPTO_ECB=y
CONFIG_CRYPTO_XTS=y
CONFIG_CRYPTO_ESSIV=y
CONFIG_CRYPTO_HMAC=y
CONFIG_CRYPTO_CRC32C=y
CONFIG_CRYPTO_CRC32C_INTEL=y
CONFIG_CRYPTO_MD5=y
CONFIG_CRYPTO_SHA1=y
CONFIG_CRYPTO_SHA1_SSSE3=y
CONFIG_CRYPTO_SHA256_SSSE3=y
CONFIG_CRYPTO_SHA512_SSSE3=y
CONFIG_CRYPTO_SHA256=y
CONFIG_CRYPTO_SHA512=y
CONFIG_CRYPTO_AES_NI_INTEL=y
CONFIG_CRYPTO_USER_API=y
CONFIG_CRYPTO_USER_API_HASH=y
CONFIG_CRYPTO_USER_API_SKCIPHER=y
CONFIG_CRYPTO_USER_API_RNG=y
CONFIG_CRYPTO_USER_API_AEAD=y
CONFIG_CRYPTO_HASH_INFO=y
CONFIG_CRYPTO_LIB_AES=y
CONFIG_CRYPTO_LIB_SHA256=y
```

closed 2023-05-08 by tlaurion