linuxboot/heads #1402: Heads needs a logo!

tlaurion, 2023-05-11:
Someone has a proposition for a Heads logo?

As per https://github.com/osresearch/heads/pull/1381#issuecomment-1542610357 picture:
[image]

Now could be a great time to combine concepts.
What I have in mind is advertising linuxboot (as a concept) with coreboot (propeller of OSF) and Heads (which misses a logo currently).

Any propositions?

Today's limitation:
- Logo needs to be 1024x768 (limitation of jpeg parser into coreboot that is not matching seabios's codebase (14 years old...)
- jpeg needs to be saved from gimp restrictive mode (scroll down at end of post https://puri.sm/posts/librem-13-coreboot-report-february-25th-2017/

tlaurion, 2023-05-11:
@thrillatplay that also resolves one of your old issue for your coreboot logo project :)

Thrilleratplay, 2023-05-11:
@tlaurion I never cross-linked the same problem I listed in Thrilleratplay/coreboot-builder-scripts#15 but I think this [open coreboot PR](https://review.coreboot.org/c/coreboot/+/42806) fixes the issue within the coreboot build.

Thrilleratplay, 2023-05-11:
A long time ago I remember serendipitously searching for something related the Heads ROM and coming across an image for Janus.

[image]
[source](https://commons.wikimedia.org/wiki/File:Janus_coin.png)

From [wikipedia](https://en.wikipedia.org/wiki/Janus)
> In ancient Roman religion and myth, Janus is the god of beginnings, gates, transitions, time, duality, doorways, passages, frames, and endings.

@osresearch mentioned that is was also the name of a super computing cluster he used in college.  (Sorry if I misremember any detail)

tlaurion, 2023-05-11:
> @tlaurion I never cross-linked the same problem I listed in [Thrilleratplay/coreboot-builder-scripts#15](https://github.com/Thrilleratplay/coreboot-builder-scripts/issues/15) but I think this [open [...]

@Thrilleratplay no, this is different. this is coreboot's libgfxinit, doing bootsplash just like where your attempts failed. Not seabios jpg/bmp issue. Limitation is that jpeg parser is old and only deal properly with old combination of resolution; this is 1024x768, not native resolution (and why you have empty space left and right of the picture and on the framebuffer after). That would deserve fixing upstream.

ThePlexus, 2023-05-11:
Hello team

I thought about this a little bit and would like to submit one for consideration. Ive not really done design or symbolism before so I don't expect this to be much liked, but i thought I would try my hand at it. Whats the worst that can happen ;) 

The stylised H circled at the top is designed to represent binary 011 - to show the triad of Heads, Coreboot and Linuxboot together, 

The design also nods to the requirement of including the logos of both coreboot and linuxboot, which handily also represent the triad in the form of a three sided dice

The dice motif is also a nod to QubesOS (cos, well .... of course I would)

And what security thing isnt complete without the shield in 2023?

The backing for this was [SVG shield/dice](https://game-icons.net/1x1/delapouite/dice-shield.html ) which is released under CC 3.0 

the format is JPEG, exported from with GIMP at 1024x768 using the recommended progressive 4:2:2 

[image]

tlaurion, 2023-05-11:
@ThePlexus as of now you have my vote!

walterchris, 2023-05-12:
I like the idea @ThePlexus . IMO I would leave out the shield and concentrate more on the "dice" itself. You have to imagine: " How does the logo look on a sticker" and how does it look on a boot screen ?

In both cases IMO it has too many details if you are relying on the shield. Imagine the logo to be printed as-is on a 4inch x 4inch sticker. The outline would be the shield, then the dice itself would be tiny :) 

Other than that it's a great idea :)

ThePlexus, 2023-05-12:
> I like the idea @ThePlexus . IMO I would leave out the shield and concentrate more on the "dice" itself. You have to imagine: " How does the logo look on a sticker" and how does it look on a boot [...]

Good points - heres an inverted version, no shield. 

> Other than that it's a great idea :)
Thanks!

[image]

walterchris, 2023-05-12:
I like that much better. Great!

d-wid, 2023-05-12:
@ThePlexus Is there a (good) way to make the H look like a couple of faces to make it look like Janus as @Thrilleratplay suggested?

d-wid, 2023-05-12:
Like some less rudimentary version of 
[image]

d-wid, 2023-05-13:
I am quite partial of what @Thrilleratplay did with his coreboot logo.
[image]

tlaurion, 2023-05-13:
Two things here and I'm sorry we are currently mixing thing.

- Heads logo
- Heads bootsplash

I think we have something with the H here (Janus Double heads) here for Heads logo, being separated of linuxboot and coreboot.

For Heads bootsplash, I have to say I am fan for the 3 equal parts of dice (representing indirectly Qubes) being sitting on Heads (top) which is a linuxboot  derivative (linux in firmware) sitting on coreboot.

If there was a 3 equal parts of that design, I would take that as a bootsplash today.

d-wid, 2023-05-13:
> If there was a 3 equal dice parts of that Hackerish design, I would take that as a bootsplash today for myself, simply because I would love to see that at each boot :). But for everyone using Heads by default, I do not know.

> As for Hacker/enthusiasts bootspash vs production ready bootsplash and or derivatives, I think a lot of variations could sit in a separate directory for enthusiasts to choose from at compilation time, but I would prefer if the one referred to/included by default in roms was the most professional looking one (as opposed to hacker enticing?

Leaving this discussion aside,
[image]

tlaurion, 2023-05-13:
Will add the two dices under https://github.com/osresearch/heads/pull/1403

With references to authors

tlaurion, 2023-06-21:
@d-wid @ThePlexus 

I have been having my tests boot up with all your logos
- https://github.com/osresearch/heads/issues/1402#issuecomment-1545404715 (sober)
- https://github.com/osresearch/heads/issues/1402#issuecomment-1546395983 (Janus related)

I do not know if i'm asking too much here, but what about we replace the concept of Janus with two sides of a brain, doing the H of Heads as a logo, and then integrate it in the bootpslash?

I never played around with graphic AI, but if there are people out there willing to do that, I am planning on using @ThePlexus logo under #1061 as a start until we come up with something we want to stick as Heads logo. I love Janus, but it seems not enough serious. I love the H of @ThePlexus, but using that H alone as a logo would be a bit boring. I'll spin that on matrix so see what happens next.

tlaurion, 2023-06-22:
Well. I was not so successful with AI generators, spending way too much of my night on it yersterday trying to communicate my need with AI.

What went up with the Janus idea is the following but not so convinced
[image]

tlaurion, 2023-07-14:
@verygreen shared an interesting idea at DAG#2
> publicly available binaries should have splashscreen with link to donation (qrcode?)

I would try to play with that concept and have a small down right qr code saying:
 "Please donate as much as you can at https://opencollective.com/insurgo"

Not a marketing guy here. Advices welcome.

tlaurion, 2023-08-03:
[image]

Logo proposition please thumb 👍👎

tlaurion, 2023-08-03:
[image]

Please 👎👍

tlaurion, 2023-08-03:
[image]

Please 👍👎

tlaurion, 2023-08-03:
[image]

Please 👍👎

tlaurion, 2023-08-03:
[image]

Please 👍👎

tlaurion, 2023-08-03:
> [image]
> 
> Please 👍👎

I'll try to find time to remove text on this one. My personal favorite in respect to Janus

tlaurion, 2023-08-04:
> > [image]
> > Please +1-1
> 
> I'll try to find time to remove text on this one. My personal favorite in respect to Janus

Tried under https://github.com/osresearch/heads/pull/1451#issue-1837074610 but please modify as you want. Maybe black and white under splashscreen?

natterangell, 2023-09-11:
This is by far my personally preferred suggestion so far.

> [image]

tlaurion, 2023-09-12:
> This is by far my personally preferred suggestion so far.
> 
> > [image]
> 
> 

You would see that being on everyone bootsplash? I like Janus reference but I'm not sure the double headed H fits the need for a logo, nor thr matrix background fit the bootsplash bill, but I could be wrong? 

@natterangell thoughts?

Otherwise I could switch it in a PR and see people's reactions. It's hard to gage those sort of things, so many users are silent voters.

natterangell, 2023-09-12:
@tlaurion well I'm far from being a graphics designer, so my taste might not be everyone's cup of tea. But I like this logo because it incorporates the janus face (btw, heads as in "the other side of tails", is an interesting reference, given that flipping coins goes back to ancient Rome, and the janus logo was found on several Roman coins).

I like the matrix/techy background, I think it fits well with the underlying technical philosophy behind this project. "The hidden code in the otherwise unknown firmware underpinning our daily computer use" kind of thing. But maybe that's just me.

I'd love to see this in a PR, but I'm a happy camper either way. 

FWIW, I've been thinking it would be nice to have an option for a somewhat "silent boot" process too. I have little need for PCR-values and kernel args on every boot once everything is set up and as long as there are no resigning needed etc. Like plymouth or the old framebuffer consoles where you can hit Esc to see whats going on underneath if you want to. I currently use "console=tty2", so "Finalizing chipset" is the last thing I see before my desktop shows.

d-wid, 2023-09-12:
@natterangell I think @tlaurion already has that jpg in master (or did at some point). It's not the default splash image but a little symlink manipulation should do it.

tlaurion, 2023-09-12:
> @natterangell I think @tlaurion already has that jpg in master (or did at some point). It's not the default splash image but a little symlink manipulation should do it.

Exactly. I would like to not fill the git repo with binary images too much but as of now some are under the blobs directory and the coreboot configs point to a symlink that points to @ThePlexus version. That symlink file can point to any other to be used, with master now dealing with jpg voodoo automagically since a patch was added to coreboot requiring imagemagik as an additional host dependency

tlaurion, 2023-09-12:
@d-wid @natterangell #1501

natterangell, 2023-09-13:
@tlaurion @d-wid Thanks, I didn't notice the picture was already in master. A new symlink was all it took, looks absolutely wonderful in my opinion :)

I get that the repo shouldn't be filled up with binary images.

d-wid, 2023-09-14:
I've added a new PR #1502 with images using the logo from #1451, for anyone interested.

closed 2023-10-09 by tlaurion

irelativism, 2024-07-25:
A bit late to this discussion but great logos ussually lack complexity and also have very litle relation to the name utilized itself, also the source of the idea "heads" comes from the inverse of TheIncognitoAmenesicSystem not a literal head. Do belueve the janus heads idea to be solid more options should be explored:
. define main principals behind project
. value that want to be portraited
. overall colour scheme used and aschtetics of the hole thing.

Cherrs irelativism.

Here small sketch on direction one could take:
[image]