linuxboot/heads #1404: QubesOS Blue Pill (Malicious Hypervisor) Attack

bluesteal, 2023-05-17:
Does Head's threat model include the Blue Pill (nested malicious hypervisor) attack which is loaded as a kernel module?

https://blog.invisiblethings.org/2006/06/22/introducing-blue-pill.html
https://theinvisiblethings.blogspot.com/2006/08/blue-pill-detection.html
https://papers.put.as/papers/macosx/2006/BH-US-06-Zovi.pdf

tlaurion, 2023-05-17:
@bluesteal

The blue pill attack, as you pointed out, would require loading an additional kernel module that is not present in coreboot's payload (kernel + initrd.cpio.xz).  (Same applying to linuxboot's DXE replacement with a linux kernel+initrd).

initrd.cpio.xz contains heads.cpio (policies under initrd), tools.cpio (tools compiled per modules definitions, included as build statemements under board configs) and modules.cpio (kernel modules not being built in the kernel, as board configs build statements define inclusion, as long as modules/linux defines them for inclusion as well and the linux config for a board also builds them as modules).

Heads requires a TPM to enforce firmware remote attestation. Coreboot enforces measured boot, which measures coreboot stages from bootblock up to the payload into PCR2. Then the linux payload is launched per coreboot, and Heads extends other PCRs as per https://osresearch.net/Keys/#tpm-pcrs.

If the user sets up a TPM disk unlock key into LUKS (which seals the measurements into nvram with measurements and an additional passphrase), then kernel modules loaded to provide linux runtime are also considered into the sealed secret into tpm nvram, and will only unseal the disk unlock key if they are the same.

Then, to boot your linux OS, Heads kexec into the next kernel + initrd. This means that Heads kernel is replaced with the one to boot into (kexec load) and then executed, at which point the old kernel modules are irrelevant.

It is also to note that Heads is a "controlled" environment, in the sense that at any moment you go into recovery shell, the TPM has its PCR4 extended, which invalidates the firmware integrity atestation measurements (TOTP/HOTP) and would also prevent runtime measurements (TPM disk unlock key will not release). One cannot go to recovery shell and back to the GUI without having measurements invalidated.

----

Now, nothing under Heads prevents to kexec (load a kernel+initrd) that would include a blue pill, which could then apply. Someone replacing Heads with a bluepilled version would apply. 

As asked multiple times in different issues and online, someone could craft a modified Heads firmware image including that additional kernel module. Reminder: coreboot measures the payload into PCR2, and then Heads measures runtime status of on-demand loaded kernel modules, wich is not specific to this raised issue. This firmware tampering, to go unnoticed, would need to fool coreboot measured boot measurements, so that they report the same PCR2 value as part of TOTP/HOTP sealed secret into TPM nvram, and Heads unsealing scripts would need to report fake results of measurements of modules loaded etc. Once again there, no platform under Heads today enforces TXT+ACM blobs so that the bootblock is part of IBB and the ACM blob itself is the one measuring IBB so that the Root of Trust of the measurement is enforced by the CPU (possibe in haswell+). So today, heads RoT, for most platforms is into the bootblock under coreboot, which then measures all following stages up to the payload (Heads kernel+initrd). One would need to fake the measurments reported per each coreboot stage there to report as being the same "firmware version" into TPM PCR2.

----

So the quick answer to your question is: highly improbable. 

An attacker successfully tampering the firmware as described above could then add an additional cpio to be passed to the next kernel (this is how secret.cpio is crafted under heads to inject crypttab and a LUKS decryption key "passphrase"). That additional cpio could replace OS init script and inject the bluepill module into the next kexec'ed kernel, yes. 

The content of /boot could be modified (initrd, kernel) but yet again, Heads defers trust of booted OS to the user signing /boot content upon system components upgrade (xen, kernel, initrd, grub config). The security mechanism in place is to make sure that the firmware and /boot components are in a known state, which implies that the user needs to reboot the system as soon as such updates are deployed to attest of the verified state of the core system upgrades. In short there, Heads will detect that /boot is different. It will report on changed files, deleted files and since recently, added files under /boot. But Heads requires of the user to detach sign digests it creates for those files, and then relies on gnupg to verify the detached signature of the digest, and relies on busybox's sha256sum applet to produce the digests. Yet again, user's public key is injected inside of CBFS, which is part of coreboot's payload section of the firmware. So with that chain of trust, Heads is different then UEFI proprietary firmware, of coreboot's grub/seabios payload, simply because the runtime environement provided by the firmware's GUI is linux, and that the bootloader's function of Heads is provided by Linux+ kexec. 

Coreboot enables the virtualization extensions of the CPU, the linux kernel configures parts of it (eg:iommu is activated on Heads kernel being active) and then Heads passes control to the next OS through kexec.

Note that some states are required to be passed from the first kernel to the second one. Kexec is responsible to either pass known graphic framebuffer to provide early console access, or to tell the second kernel how it should be entered in (64bits, 32bits entry point) and/or with certain options to reuse first kernel's graphical framebuffer. But we are drifting from your original question.

bluesteal, 2023-05-17:
How is the approach taken by Trenchboot different from that taken by Coreboot/Heads?

tlaurion, 2023-05-18:
This is the plan, but won't work with all platforms, needing funding and a lot of work/collaboration, new board configs for txt enablement, blobs download scripts modification etc.

Note that this is not directly related to your now closed issue and should be in another one if not existing.

Bluepill attacks take advantage of virtualization. Joana suggested to disable the extension to completely mitigate the risk herself. Hope we are not mixing things here.

D-rtm permits to dynamically measure state whenever needed as opposed to statically measuring on boot.

Those two concepts are complementary. Heads in this case would need to pack ACM and sinit inside of coreboot, and modify its measuring scheme accordingly. After that, drtm will be enabled and would depend on OS used.

In the type of attack describe in your issue opening, drtm doesn't mitigate this.

Other technologies might at runtime, but as also said previously, the best one can do to mitigate this is compartmentalization. That is, to separate risks and proper risk assessment.

Everytjing is threat modeling related. I would not let my passphrase manager unlocked, let it be my vault qube running side by side of an untrusted, disposable qube. I refrain typing passphrases manually in untrusted environment.

I would recommend not trusting technology too much and investing time in developing proper habits to mitigate your own risks. There will never be completely trustable security mechanisms that will deprecate proper opsec...

Better safe then sorry.

bluesteal, 2023-05-18:
Which extension did she recommend disabling? Is there a reference to that (a URL)?

A vault Qube running beside an untrusted, disposable qube; do you mean in terms of their location in memory? What
do you mean by running "side by side"?

tlaurion, 2023-05-18:
> Which extension did she recommend disabling? Is there a reference to that (a URL)?

I was referring to the same URLs you referred to https://theinvisiblethings.blogspot.com/2006/08/blue-pill-detection.html

> A vault Qube running beside an untrusted, disposable qube; do you mean in terms of their location in memory? What
> do you mean by running "side by side"?

I meant vault and a distrusted disposable qube running side by side. There is no reason to keep vault running (and having things in memory) if not needed. The logic here is that by running multiple disposable qubes for different tasks, even if an untrusted disposable qube was momentarily able to read the content of another qube, it would not be able to steal important secrets from memory. Simply because that secret would not be in memory.

The analogy always stands: you not not need to protect a secret that you do not know.

tlaurion, 2023-05-18:
> I'm assuming you mean the CPU extension that allows for a nested virtual machine right? What is the name of the extension?

That is not a cpu extension, but a property of hypervisor. Qubes ships xen with nested virt deactivated.

tlaurion, 2023-05-18:
> D-rtm permits to dynamically measure state whenever needed as opposed to statically measuring on boot.

This is moving target under coreboot.
3mdeb updated docs for coreboot 4.20 release. I will have to check qemu tpm2 support to make sure everything is alright but I remember we checked

https://review.coreboot.org/c/coreboot/+/68751/14/Documentation/security/vboot/measured_boot.md

bluesteal, 2023-05-19:
I'm a bit confused about nested virtualization. What is the technical name of this CPU extension?

bluesteal, 2023-05-19:
Does the X230 have a cpu with vt-x and EPT (Extended Page Tables)?

bluesteal, 2023-05-19:
and what about tboot(https://sourceforge.net/projects/tboot/)? Doesn't this project implement DRTM?

tlaurion, 2023-05-19:
@bluesteal you have a lot of questions. This is awesome, don't get me wrong.

Please open separate issues. It is also appreciated when internet searches are done prior of asking questions.

Yes the x230 has vtx and ept. That is, it is probably the most used hardware for qubesos, and those are qubesos hardware requirements.

tboot says on the same link you provided that it is an implementation of drtm.

You might want to look into qubesos news and forums as well, where most if those questions were already answered before.

tlaurion, 2023-05-19:
I consider this issue closed.