linuxboot/heads #1405: Analyzing Heads Payloads

bluesteal, 2023-05-17:
Can Heads payloads be analyzed using a firmware tool like CHIPSEC (https://github.com/chipsec/chipsec)

tlaurion, 2023-05-17:
@bluesteal It could. It has been a dream for a while now to have python under Heads to be able to drive those tests directly from within Heads, but Python is really big. Chipsec also requires building a kernel module that needs to be injected so chipsec can do its job close to the kernel. 

I made some tests combining u-root's exp/tcz (Tinycore downloader), which is definitely not ready for production and waiting for u-root to fix tinycore post-installation issues at https://github.com/u-root/u-root/issues/2665. That could be one of the most easiest ways to deploy python under Heads. You would most probably need to cross-compile chipsec's kernel module against comipled kernel with musl-cross-make as well. It would be nice to have something that could provide Chipsec to be ran on top of Heads, and was desired for a while (other issues exist on python and chipsec on this project. The desire for the future is to be able to pack things externally and require from the user to download those things once under /boot and detach sign them).

You could test chipsec enabled distro as referred per the project's README and README.md (3 projects vs 2 btw, you could open issues there if you investigate deeper).

Coreboot locks the chipset, and it is known that to be able to interact with IOs/SPI, you would need to pass `iomem=relaxed` on grub/syslinux config to be able to run chipsec tests from a livecd/installed OS, if not directly under Heads. This is common; one cannot expect to flash SPI from a booted os without passing those parameters. I'm sure their docs will guide you deeper in those needed directions.

Let me know how this goes for you. I would suggest the second approach more then the first. I will try to remember this issue when/if tinycore PoC goes deeper. I am looking forward, and for a while now, on a way to pack python3 inisde of the firmware image to a point of giving up, because it is too big to fit. So it would need to be sideloaded. Ideal would be to pack Nix's pinned repository drv outcome directly into a squashfs, but the more I think about that the more I see myself redoing the wheel which tinycore is doing great (while not reproducible, http only etc). So not quite sure of the best approach to go there.

bluesteal, 2023-05-17:
Is there a way to copy the firmware image file to a USB and then analyze it by booting into Qubes and launching one of the OSes mentioned (Blackarch) and then copying the image file into the live OS for forensic analysis?

tlaurion, 2023-05-17:
Chipsec's goal is to interact with live states of a firmware, poking around memory and testing things live.
There are other projects out there aimed at testing things statically, but as far as I know they won't provide as much information as chipsec would, even less on Intel platforms.

More importantly, Qubes is made to abstract from the hardware, so yes, only those unknown static tools would permit to analyze built roms, where only dom0 could interact with the firmware, and yet again, dom0 interfaces with Xen but is not Xen, and interfering ith dom0 config to have iomem=relaxed would not be recomended at all. Best you can do is boot from a live cd here, and install needed packages in ram and test desired config. Heads can have KERNEL_ADD board options added if you compile the rom yourself, which would be passed to kexec kernel. You could investigate that part, or install desired OS on a drive and modify grub.cfg on that OS to have that booted as you want. You could also copy from heads recovery shell /etc/config to /etc/config.user, modify CONFIG_BOOT_KERNEL_ADD line to include iomem=relaxed there and then call config-gui.sh to merge them under firmware. That would inject /etc/config.user into your rom, which would invalidate previous measurements and would require you to reseal secrets.

I will leave this one open for listing static analysers the community might find, otherwise tinycore support might land under Heads one day, and faster if community participates in the effort. I use my own PoC to strace things already when testing stuff that doesn't work, but the PoC is not ready to be put into a PR as of today, which would require work happening under u-root as referred above.

bluesteal, 2023-05-22:
Can you explain further what the iomem=relaxed setting is and why it is necessary to include this?
This setting is necessary for doing dynamic forensic analysis, versus static, on the firmware using a live OS like blackarch?

bluesteal, 2023-05-22:
Also, How does this relate to TinyCore Linux? Does Heads use TinyCore Linux as a dependency?

tlaurion, 2023-05-23:
> Can you explain further what the iomem=relaxed setting is and why it is necessary to include this?

Looking it up would lead you to pages like
https://www.flashrom.org/FAQ#flashrom_doesn't_work_on_my_board,_what_can_I_do?

Which would explain it better then I would do. If you intend to get the image in SPI, you can either try the right way or open up IO for root access as documented there to dump firmware image for static analysis.

Dynamic analysis, like chipsec or BITS will depend on a kernel module that needs to be compiled to work within heads, on top of Python to poke and probe what is needed to test the platform dynamically.

> This setting is necessary for doing dynamic forensic analysis, versus static, on the firmware using a live OS like blackarch?

No. Needed to be able to access the firmware directly only in a unsafe (temporary) manner. "For testing purposes only".

tlaurion, 2023-05-23:
> Also, How does this relate to TinyCore Linux? Does Heads use TinyCore Linux as a dependency?

Tinycore provides Python. Tynycore could be applied on top of Heads if u-root was complementary deployed. There is no other easy way to provide Python there, outside if the firmware image constrained with space, otherwise kexec'ing into another kernel and initrd which would be really similar to booting a live iso at that point, without any benefit and also changing the results. Kexec'ing into a live image can modify the results of the tests.

You can refer to https://www.osfc.io/2018/talks/bits-and-chipsec-as-coreboot-payloads/ into why testing with chipsec as a coreboot payload is more desirable then from a live iso, but as of now it is more a wish then a plan.