linuxboot/heads #1414: Build failure Debian & Fedora

debutifi, 2023-06-09:
## Context of the Build

**1. What board are you trying to build?**
T440p

**2. What repository:branch are you using to build from?**
- [ ] Heads:Master✅
- [ ] Other (please specify)

**3. What version of coreboot are you trying to build**
- [ ] 4.13
- [ ] 4.14
- [ ] 4.15
- [ ] 4.17 ✅
- [ ] Other (please specify)

**4. In building the rom where did you get the blobs?**
- [ ] No blobs required
- [ ] Provided by the company that installed Heads on the device
- [ ] Extracted from a backup rom taken from this device✅
- [ ] Extracted from another backup rom taken from another device (please identify the board model)
- [ ] Extracted from the online bios using the automated tools provided in Heads
- [ ] I don't know

**5. If using the automated tools to get the blobs did you run the relevant scripts in the blobs directory**
- [ ] Yes
- [ ] No

**6. What operating system are you using**
Debian & Fedora
*please specify*

Make fails on both Debian & Fedora for T440p.

make[5]: Leaving directory '/home/user/heads/build/x86/musl-cross-38e52db8358c043ae82b346a2e6e66bc86a53bc1/build/local/x86_64-linux-musl/obj_binutils'
make[4]: *** [Makefile:850: all] Error 2
make[4]: Leaving directory '/home/user/heads/build/x86/musl-cross-38e52db8358c043ae82b346a2e6e66bc86a53bc1/build/local/x86_64-linux-musl/obj_binutils'
make[3]: *** [Makefile:204: obj_binutils/.lc_built] Error 2
make[3]: Leaving directory '/home/user/heads/build/x86/musl-cross-38e52db8358c043ae82b346a2e6e66bc86a53bc1/build/local/x86_64-linux-musl'
make[2]: *** [Makefile:170: install] Error 2
make[2]: Leaving directory '/home/user/heads/build/x86/musl-cross-38e52db8358c043ae82b346a2e6e66bc86a53bc1'
make[1]: *** [Makefile:174: musl-target] Error 2
make[1]: Leaving directory '/home/user/heads/build/x86/musl-cross-38e52db8358c043ae82b346a2e6e66bc86a53bc1'
make: *** [Makefile:444: /home/user/heads/build/x86/musl-cross-38e52db8358c043ae82b346a2e6e66bc86a53bc1/.build] Error 1

What OS you recommend for builds?

tlaurion, 2023-06-10:
@debutifi Debian-11 following instructions which points to install dependencies as CircleCI does.

Soon enough we will resolve this issue using a nix docker once and for all.

But for now only debian-11 is tested.

debutifi, 2023-06-10:
Ok, Debian 11 + CircleCI dependencies fixed that problem. It built two roms but they say dirty. What does this mean?

tlaurion, 2023-06-11:
That your local git tree is unclean