linuxboot/heads #1415: No OS detected after install.

adwall4, 2023-06-11:
I booted Alpine off USB and installed to disk with and without encryption and changed the default boot partition to sdb1.

Unfortunately it still couldn’t find anything to boot. Is Alpine supported under Heads?

adwall4, 2023-06-11:
Last line before error: Failed to parse any boot options

tlaurion, 2023-06-12:
Can you do
find /boot 
And take a screenshot?

adwall5, 2023-06-12:
[image]

this is with just standard alpine install no encryption.

tlaurion, 2023-06-14:
Ho.
I guess you would need to do a customize installation as per instructions on wiki.
At least /boot partition unencrypted and the rest under /

As per screenshot / (root) is under /boot.
This is not good because everything would be hashed and change across reboots.

Same guidelines as for Debian here: https://osresearch.net/InstallingOS/#compatibility

adwall5, 2023-06-14:
Yeah but Alpine puts boot on /dev/sdb1, swap /dev/sdb2, and rest /dev/sdb3. I set default boot to sdb1 but it doesn’t recognize it.

I think this is more issue of Heads not supporting syslinux rather than Alpine.

Unfortunately, I’m not sure how to install grub inside alpine live usb.

tlaurion, 2023-08-01:
> I think this is more issue of Heads not supporting syslinux rather than Alpine.

You are right. syslinux and grub are both supported for example [media-scan](https://github.com/osresearch/heads/blob/06b1b0948d911ab1ee15d08cc4776e41889349cd/initrd/bin/media-scan#L104) (*.cfg), but not anywhere else in [codebase](https://github.com/search?q=repo%3Aosresearch%2Fheads+cfg&type=code)