linuxboot/heads #1421: Legacy boards considered to be deprecated

tlaurion, 2023-06-23:
This is x220/t430/x230 boards mot showing "maximized" on Heads main menu.

How many of you that represents?

merge, 2023-06-25:
I haven't used (or at least changed) a Heads installation in quite a while now. What exactly is the meaning of the issue? - I'm not entierly sure. Skulls users "usually" would unlock IFD, but I consider making this the default behaviour even - as I think ideally we want Heads to enable WP before booting an OS.

tlaurion, 2023-06-25:
> Skulls users "usually" would unlock IFD, this is the default behaviour even - as I think ideally we want Heads to enable WP before booting an OS.

@merge that is exactly my concern. If ifd is locked by default today, those users can only flash Heads legacy ROMs for t430/x230/x220.

And those legacy boards per this ticket are thought to be dropped from main repository because its extremely hard to pack everything under 7mb of normally defined IFD BIOS region.

So my question to you is to how many users out there from skulls user base are having IFD locked, ME non-neutered/not unlocked which won't be able to internally flash to maximized boards.

As for WP protection, io386 and maximized boards are pushing PR0 locking, aka SMM platform locking through SMI chipset locking really recently.

merge, 2023-06-26:
no, IFD is *unlocked* by default when installing Skulls. Most users neuter ME, but it's not yet the default. maximize the "bios" region is not yet done by Skulls, but iiuic it could be done without disrupting the user-workfow. what would it roughly be? coreboot config size-change, new layout-file + ifdtool run on the image? And if the *actual* size of the Skulls image remains <4M, external hardware-flashing is still possible to only the one 4M-chip? patches and testing welcome :)

good to hear WP is progressing

bwachter, 2023-06-26:
I only have boards with neutered ME.

tlaurion, 2023-06-27:
@merge 
> good to hear WP is progressing

This was merged per https://github.com/osresearch/heads/commit/6ec0c814430c5ad0df54f834c89ee72a904466b2 for maximized boards only as of now.

It is to note that further disabling kernel config network options under linux-x230-legacy.config per https://github.com/osresearch/heads/pull/1381/files#diff-2af02345f256b80007d10fb6a49cfaaf8dd4c2d96f044f9976f8f5f1396b3fb6 delays deprecation for legacy boards (which all use linux-x230-legacy.config 5.10.5 from now on).

> maximize the "bios" region is not yet done by Skulls, but iiuic it could be done without disrupting the user-workfow. what would it roughly be? coreboot config size-change, new layout-file + ifdtool [...]

This is the problem we are talking about. In simpler terms, skulls users need to follow https://osresearch.net/Prerequisites#current-legacy-boards **down to this actionable verification** https://osresearch.net/Updating#verify-upgradeability-paths-of-the-firmware

tlaurion, 2023-07-04:
So per #1398 t430-legacy and x230-legacy were tested.

Other untested boards are now renamed UNTESTED.

It is considered that the last working commit for those platforms was [91f65be](https://github.com/osresearch/heads/commit/91f65bed793c0165203b1cbb9ac4f390f15ffa0e) for the following boards:

- UNTESTED_x230-maximized-fhd_edp
- UNTESTED_x230-hotp-maximized-fhd_edp
- UNTESTED_x220-maximized
- UNTESTED_x220-hotp-maximized
- UNTESTED_w530-maximized
- UNTESTED_w530-hotp-maximized
- UNTESTED_w530-dgpu-K2000m-maximized
- UNTESTED_w530-dgpu-K2000m-hotp-maximized
- UNTESTED_w530-dgpu-K1000m-maximized
- UNTESTED_w530-dgpu-K1000m-hotp-maximized
- UNTESTED_t530-maximized
- UNTESTED_t530-hotp-maximized
- UNTESTED_t530-dgpu-maximized
- UNTESTED_t530-dgpu-hotp-maximized
- UNTESTED_t520-maximized
- UNTESTED_t520-hotp-maximized
- UNTESTED_t440p-maximized
- UNTESTED_t440p-hotp-maximized
- UNTESTED_t420-maximized
- UNTESTED_t420-hotp-maximized
- UNTESTED_p8z77-m_pro-tpm1-maximized
- UNTESTED_p8z77-m_pro-tpm1-hotp-maximized

tlaurion, 2023-07-04:
x220-maximized and x220-hotp-maximized salvaged from going untested under #1398 thanks to @srgrint

tlaurion, 2023-07-04:
t440p-maximized and t440p-hotp-maximized salvaged from going untested under https://github.com/osresearch/heads/pull/1398 thanks to @srgrint

tlaurion, 2023-07-05:
t420-maximized and t420-hotp-maximized back to normal, was an error and fixed https://github.com/osresearch/heads/pull/1432

3hhh, 2023-07-15:
I just tested `d4f56bd546c195c8a46b3839813b514c30f2efbceef963c1ed6cb3499eafbaa3  heads-UNTESTED_t530-hotp-maximized-v0.2.0-1705-gedf200e.rom (https://github.com/osresearch/heads/commit/edf200e7913c62975a424cfb9dbd579747d0665c)`, seems to work so far. Apparently it even fixed https://github.com/QubesOS/qubes-issues/issues/8232, thanks!

tlaurion, 2023-07-15:
> I just tested d4f56bd546c195c8a46b3839813b514c30f2efbceef963c1ed6cb3499eafbaa3 heads-UNTESTED_t530-hotp-maximized-v0.2.0-1705-gedf200e.rom (https://github.com/osresearch/heads/commit/edf200e7913c62975a424cfb9dbd579747d0665c), seems to work so far. Apparently it even fixed QubesOS/qubes-issues#8232, thanks!

@3hhh So you get "finalizing chipset" on kexec call?

tlaurion, 2023-07-17:
@3hhh if https://github.com/osresearch/heads/pull/1373#issuecomment-1587752648 doesn't work for you please open issue!

pcm720, 2023-07-19:
@tlaurion You can remove x230-maximized-fhd_edp variations from untested,  edf200e7913c62975a424cfb9dbd579747d0665c works fine on my X230.
Sorry for not being able to test it earlier.

tlaurion, 2023-08-03:
- UNTESTED_x230-hotp-maximized-fhd_edp
- UNTESTED_x230-maximized-fhd_edp
Removed from untested per #1448 being merged

Tickmeister1, 2023-08-04:
KGPE-D16 is working.  (workstation variant tested)

tlaurion, 2023-08-04:
> KGPE-D16 is working.  (workstation variant tested)

That is unexpected. Version? I see you replied on other issue with old version. Is that on master?

Tickmeister1, 2023-08-04:
Yes, it is a fresh build from master, about 2 days old.  
v0.2.0-1713-g06b1b09
Linux 5.10.5-heads

tlaurion, 2023-08-07:
w530-hotp-max 1727 reported working under https://matrix.to/#/!pAlHOfxQNPXOgFGTmo:matrix.org/$XeRUrwx_n3uCjN8UDcYhLS4Ml2wlySGd0N3LmSx9_VU?via=matrix.org&via=nitro.chat&via=fairydust.space

(I suppose without dGPU otherwise would have been specified)

ThePlexus, 2023-08-12:
p8z77-m_pro-tpm1-hotp-maximized
p8z77-m_pro-tpm1-maximized

tested working 
[image]
[image]

tlaurion, 2023-08-16:
Unfortunately @ThePlexus, w530 have not been tested under https://github.com/osresearch/heads/pull/1403 which will be merged soon. Please report testing on top of master or under that branch and report individual board report statuses here so those board flavors can be moved back to tested (normal) so other can flash without worries.

Otherwise this is still considered untested.

tlaurion, 2023-08-16:
Modified OP post to point to the list of boards containing UNTESTED_ from master.
I think this is the most efficient way of dealing with this, and keeping issues open for untested boards

tlaurion, 2023-08-16:
>Can confirm heads-t430-maximized-v0.2.0-1747-g572573f.rom works

_Originally posted by @srgrint in https://github.com/osresearch/heads/issues/1403#issuecomment-1681091500_
-----

t430-hotp-maximized and t430-maximized were removed from untested boards prior of #1403 merge.

tlaurion, 2026-03-05:
Documented under unmaintained_boards/README.md